evcc-io/evcc #22774: evcc doesn't start if one of the Solar invertors is switched off

jpolefs, 2025-08-04:
### Describe the bug

I have just discovered that if one of the SMA invertors is switched off, evcc gives a FATAL ERROR at startup. EVCC requests to perfrom a RESTART but will only restart IF my SMA invertor is active.
This is an issue I didn't have in the past when having the invertors configured in the evcc.yaml file.
I erased all in the evcc.yaml file and configured them via the UI.

For me, this not acceptable: IF one of the SMA invertors e.g. is taken in maintenance, I should need to remove it from the UI configuration and when back operational I need to reconfigure it again.

I suggest to just report that one of the invertors is OUT OF OPERATION but evcc should not refuse to start and give FATAL errors.

When the SMA invertor was switched off and I did the upgrade to 0.207.1, I got the same FATAL error and evcc just hangs...

I don't like this way of evcc operation: if one of the invertors is out of service, evcc should just report the ERROR and  continue working with the other invertors.... 

I've attached screen shots of the ERROR - see log details

[image]
[image]

### Steps to reproduce

1. evcc works fine
2. SMA invertor switched off
3. evcc reports FATAL error and requests Restart
4. clicking restart keeps reporting FATAL error and evcc 'hangs'
5. solution 1: get the invertor operational again and evcc works normal again
6. solution 2: remove the invertor out of the UI configuration and evcc works normal again****

### Configuration details

```yaml
Configuration is done via the UI
I have no idea where to find the evcc database.....sorry

As mentioned above, all invertors are removed from evcc.yaml and recreated in the evcc database
```

### Log details

```text
see attached screenshots
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.0 was restored after upgarde to 0.207.1

mfuchs1984, 2025-08-04:
#14496

humi208, 2025-08-04:
Nachdem ich von yaml auf UI Konfiguration umgestellt hatte, dachte ich auch es wäre gut den STP X als Speedwire WR zu konfigurieren. Problem war dann jedoch, dass der WR nachts ohne Produktion anscheinend auf Speedwire nicht zur erreichen war und man somit evcc zur dunklen Zeit nicht starten konnte.
Ich habe den STP X jetzt einfach wieder auf Modbus geändert und dann kann man evcc auch zu jederzeit neustarten...

closed 2025-08-05 by andig