evcc-io/evcc #22777: HA vehicle range: strconv.ParseInt: parsing "531.52": invalid syntax

Knuppel1983, 2025-08-05:
### Describe the bug

Using [esphome-tesla-ble](https://github.com/PedroKTFC/esphome-tesla-ble) via HA, and latest 0.207.1.
In HA range is a comma value. EVCC seems to expect an integer. Is this something that should be fixed in the code or can i convert this in the yaml?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
- name: tesla_zakelijk
    type: template
    template: homeassistant
    title: # Title, Will be displayed in the user interface (optional)
    capacity: 75 # Battery capacity (kWh), optional
    uri: http://192.168.200.160:8123 # Home Assistant URI, HTTP(S) address
    token: xxx # Home Assistant Long-Lived Access Token
    soc: sensor.tesla_zakelijk_charge_level # State of charge [%]
    range: sensor.tesla_zakelijk_range # Remaining range [km], optional
    status: sensor.tesla_zakelijk_charging_state # Charging status, optional
    limitSoc: sensor.tesla_zakelijk_charge_limit # Target state of charge [%], optional
    odometer: sensor.tesla_zakelijk_odometer # Odometer [km], optional
    climater: binary_sensor.tesla_zakelijk_climate # Climatisation active, optional
    #finishTime: sensor.vehicle_finish_time # Finish time (ISO8601 or Unix), optional
    #start_charging: script.vehicle_start_charge # Service to start charging, optional
    #stop_charging: script.vehicle_stop_charge # Service to stop charging, optional
    #wakeup: script.vehicle_wakeup # Service to wake up vehicle, optional
    icon: car # Icon, Will be displayed in the user interface [car, bike, bus, moped, motorcycle, rocket, scooter, taxi, tractor, rickshaw, shuttle, van, airpurifier, battery, bulb, climate, coffeemaker, compute, cooking, cooler, desktop, device, dishwasher, dryer, floorlamp, generic, heater, heatexchange, heatpump, kettle, laundry, laundry2, machine, meter, microwave, pump, smartconsumer, tool, waterheater] (optional)
    phases: 1 # Maximum number of phases, The maximum number of phases which can be used (optional)
    #mode: # Default charging mode, Used when a vehicle is connected. Possible values are Off, Now, MinPV and PV, or empty if none should be set (optional)
    minCurrent: 5 # Minimum amperage (A), The minimum amperage per connected phase that can be used (optional)
    maxCurrent: 16 # Maximum amperage (A), The maximum amperage per connected phase that can be used (optional)
    #identifiers: # Identification, Mostly this can be added later, see: https://docs.evcc.io/en/docs/features/vehicle (optional)
    priority: 2 # Priority, Priority of the loadpoint or vehicle in relation to other loadpoints or vehicles for allocating pv energy (optional)
```

### Log details

```text
[lp-2  ] ERROR 2025/08/05 10:27:31 vehicle range: strconv.ParseInt: parsing "531.52": invalid syntax
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.1

marq24, 2025-08-05:
seams to be fixed by https://github.com/evcc-io/evcc/commit/8b37ee80bc1a61b0005ee5afe038aec92a211d8c

StefanSchoof, 2025-08-05:
No, see https://github.com/evcc-io/evcc/compare/0.207.1...master

Knuppel1983, 2025-08-05:
Great. Sorry, only just learned about nightly's. Close this one.