evcc-io/evcc #22790: SOC in kW/h

Razer2412, 2025-08-05:
Hallo zusammen, 

dank der PV Prognose kann ich ungefähr abschätzen, wie viel Ertrag ich am jeweiligen Tag noch erwarten kann. 

Ich fände es hilfreich, wenn man die SOCs (Hausakku, Fahrzeug) in "xx kWh/h von xx kW/h" statt Prozent anzeigen könnte. Am besten noch mit "verbleibenden kW/h" in Klammern. 

So wäre direkt ersichtlich, ob der noch zu erwartende Ertrag ausreichend ist, um den Hausakku oder das Fahrzeug noch vollständig laden zu können.

LeonSol, 2025-08-05:
Das finde ich nicht - zu unübersichtlich! Also von meinem 10kW Speicher 32%, 46% oder 80% auszurechnen ist nicht schwer. Und auch bei einem 82kW Akku gelingt einem das mindestens überschlägig auch im Kopf. Ich glaube, es gibt wichtigere Dinge. Man sollte auch nicht gleich einen "issue" aufmachen, sondern dies erst einmal in "discussion" posten.

closed 2025-08-05 by andig

andig, 2025-08-06:
Btw., kW/h ist keine sinnvolle physikalische Einheit: https://www.reddit.com/r/AskPhysics/comments/10iqy0a/can_someone_explain_the_difference_between_kwh/?tl=de