evcc-io/evcc #22798: Set Offered Current/Phases to 0 on disable After tariff charging

stevoboy, 2025-08-06:
### Describe the bug

Dear All

Not sure if this a bug or Feature request.

When having charged with tariffs like Tibber, the offered current and phases stay at 16 and 3 respectively even when the enable attribute goes to disable.

This is problematic for me as I use the offered currents and phases for determination if I can able fast charging or not.

Suggestion:
Either set offered currents and phases to 0 x minutes after enable attribute is set to disabled.

Or implement another attribute in config UI uncaring what is the motivation of evcc for charging: PV, or Min+PV, or Fast

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
…
```

### Log details

```text
…
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

stevoboy, 2025-08-06:
It is relating to the phases1p3p and maxcurrent attributes in configUI.

andig, 2025-08-06:
> This is problematic for me as I use the offered currents and phases for determination if I can able fast charging or not.

How a charger id being disabled depends on the configuration. It's unclear what charger this affects or how this additional logic relates to evcc.

closed 2025-08-06 by andig

stevoboy, 2025-08-06:
Okay understood. However, I have not understood while at least one request for evidence cycle is granted to provide the missing information. Not always, it is clear if the function is intended or not. 

Shall I reopen a bug or continue here?

You can easily see that there is a discrepancy between the values set by HTTP and what the EVCC Adapter on Iobroker shows. For some reason the HTTP does not work always

Screenshots taken taken in the same minute about 4 minutes after last changing the charging mode.
HTTP (maxcurrent is actually offeredcurrent):

[image]

Iobroker EVCC Adapter:

[image]

Custom wallbox (script for offeredcurrent and offeredphases):

maxcurrent: #funktioniert
  source: http
  uri: "http://ip:port/v1/state/0_userdata.0.EVCC.charger.maxcurrent?value={{.maxcurrent}}"
  method: GET
tos: true
phases1p3p: #funktioniert
  source: http
  uri: "http://ip:port/v1/state/0_userdata.0.EVCC.charger.offeredphases?value={{.phases}}"
  method: GET

[evcc-20250806-195621-trace.log](https://github.com/user-attachments/files/21627656/evcc-20250806-195621-trace.log)

andig, 2025-08-06:
> and what the EVCC Adapter on Iobroker shows.

If so, it's not maintained by evcc. If you notice something from with /api/state please let us know.

stevoboy, 2025-08-07:
The Http Post of the Maxcurrent and Offeredphases to iobroker datapoints is only working properly with the PV Mode. You can see that currently the car is charging and the maxcurrent is constantly changing - that's how it should be.
However, when  coming from smartcost based charging enabled to disabled the offeredcurrent and offeredphases stay frozen. 
I can demonstrate this tonight.
I don't know if any logs help here as the Http Post itself works properly within PV mode. I recon the LOG would be just empty when having that smartcost based charging.

Here some pics that PV surplus charging is working flawlessly. I post some pics of the problematic use cases tonight
[image]

[image]

[image]

[image]

[image]

[image]

stevoboy, 2025-08-08:
Here you - set smartcostlimit at 8:05. charging start at 16Ax3P:

[image]

[image]

[image]

On 08:32 turning off the Smartcostcharging returing to PV:

[image]

[image]
>>> Charging stops right away because of the enable status set to disable.

At 8:40 the Maxcurrent and offeredcurrent is still at 16Ax3P (Charger starts charging by his own programm but that does not matter here). So it appears when one drops out of smartcostcharging only the Enable attribute is triggered but not the offeredcurrent and offeredphases. By the way, this happens also with my custom heatpump  where the setmaxpower is stays frozen with deactivation. On the EVCC Adapter for iobroker no issues. It appears that httppost are not working realiably.

[image]

[image]

andig, 2025-08-08:
> So it appears when one drops out of smartcostcharging only the Enable attribute is triggered but not the offeredcurrent and offeredphases. 

That's the expected behavior. 

> It appears that httppost are not working realiably.

There's no evidence of that. `Enable(false)` is correctly called afaikt. If not we'll need a log showing that not being the case.