evcc-io/evcc #22869: UI: Wechsel der Konfigurationsart von Ladepunkten ermöglichen

Tombra1889, 2025-08-10:
**Is your feature request related to a problem? Please describe.**
Wechsel der Konfigurationsart ermöglichen. Wenn ich mich bei einem Gerät einmal auf eine Konfigurationsart festgelegt habe, kann ich diese im Nachgang nicht mehr wechseln.
[image]

Mein aktuelles Gerät wird als lp-2 ausgewiesen. Es existiert auch ein Gerät mit der Kennung lp-3. Daher habe ich jetzt die Befürchtung, dass ich auch zunächst das Gerät mit lp-3 löschen muss um das Gerät mit der Kennung lp-2 neu anlegen zu können.

**Describe the solution you'd like**
Wechsel der Konfigurationsart ermöglichen. Es wäre auch i. O. wenn dann ein Hinweis kommt, dass die Daten der Konfiguration dabei verloren gehen.

**Describe alternatives you've considered**
Manuelle Zuordnung der "lp-xxx". Ist aber wahrscheinlich noch komplizierter.

**Additional context**
Anpassung der Konfiguration in der UI.

andig, 2025-08-10:
Wozu soll das dienen? Welches Problem soll damit gelöst werden?

Tombra1889, 2025-08-10:
Im Endeffekt möchte ich von User-defined heater auf eine andere Einbindung mit sg-ready oder vielleicht auch einer direkten Einbindung der Wärmepumpe wechseln. Ich denke, dass das Szenario auch gut auftreten könnte, wenn mehr und mehr Templates zur Verfügung stehen.

[image]

closed 2025-08-10 by naltatis