evcc-io/evcc #22884: Funktion "Solarprognose anpassen" reduziert die produzierte Leistung um den forecast

LeonSol, 2025-08-10:
### Describe the bug

Moin zusammen,

ich beobachte jetzt eine Weile, dass sich jeden Morgen zwischen 07:00 und 07:30 meine produzierte Solarleistung scheinbar um den Wert der Tagesprognose reduziert. Damit sinkt der scale-Faktor von Tag zu Tag immer weiter und ist nicht nutzbar:

Aug 07 07:15:02 evcc evcc[308933]: [site ] DEBUG 2025/08/07 07:15:02 solar forecast: accumulated 116.988kWh, produced 98.410kWh, scale 0.841
Aug 07 07:15:33 evcc evcc[308933]: [site ] DEBUG 2025/08/07 07:15:33 solar forecast: accumulated 116.994kWh, produced 65.810kWh, scale 0.563

Aug 09 07:02:26 evcc evcc[709]: [site ] DEBUG 2025/08/09 07:02:26 solar forecast: accumulated 196.727kWh, produced 143.548kWh, scale 0.730
Aug 09 07:02:54 evcc evcc[709]: [site ] DEBUG 2025/08/09 07:02:54 solar forecast: accumulated 196.730kWh, produced 104.748kWh, scale 0.532

Aug 10 07:13:26 evcc evcc[709]: [site  ] DEBUG 2025/08/10 07:13:26 solar forecast: accumulated 241.272kWh, produced 149.950kWh, scale 0.621
Aug 10 07:13:54 evcc evcc[709]: [site  ] DEBUG 2025/08/10 07:13:54 solar forecast: accumulated 241.278kWh, produced 104.850kWh, scale 0.435

Die Reduktion erfolgt von EVCC exakt in der Minute, in der die PV-Anlage den ersten Strom liefert -  heißt:

am 07.08 Produktion liefert den 1. Wert zwischen 07:15:02 und 07:15:33 
am 09.08. Produktion liefert den 1. Wert zwischen 07:02:26 und 07:02:54
am 10.08. Produktion liefert den 1. Wert zwischen 07:13:26 und 07:13:54

Das die bis dahin produzierte Leistung im den Wert der jeweiligen Tagesprognose reduziert wird, ist im Moment eine Vermutung, passt aber wunderbar zu den letzten Tagen.

Sonnige Grüße

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
unerheblich
```

### Log details

```text
Bereits im Beitrag
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.1

LeonSol, 2025-08-10:
@andig : es passiert jeden Tag exakt in der Minute, wenn die PV-Anlage den ersten Strom liefert. Dann bleibt der Wert für "accumulated" konstant und von "produced" werden einmalig zig kWh abgezogen (die Höhe entspricht der Tagesprognose!!!). Ich weiß nicht, wie ich es noch verständlicher ausdrücken soll? Wenn jeden Tag von "produced" zig kWh abgezogen werden aber von "accumulated" nicht, dann steigt die Differenz halt immer weiter, der scale-Faktor wird zwangsläufig immer kleiner, ohne dass dies auch nur ansatzweise den Werten entspricht, die an EVCC übermittelt werden. Da ist irgendwo ein Bug, der getriggert wird, wenn der erste Wert ungleich "0" für den aktuellen Tag an PV-Produktion übermittelt wird.

LeonSol, 2025-08-10:
@VolkerK62 Bei dir sind die Änderungen ja minimal und werden auch wieder korrigiert. Des Weiteren erfolgen diese mitten in der Nacht, wenn die PV-Anlage nicht produziert. Bei mir ist der Fall anders und es wird die komplette, kommende Tagesprognose vom bisher produzierten Wert in dem Moment abgezogen, wenn die PV-Anlage zu produzieren anfängt.

LeonSol, 2025-08-10:
Sorry, das kann ich leider nicht! Gibt es irgendwo einen Leitfaden zum Compilieren, dem ich folgen kann?

VolkerK62, 2025-08-10:
@LeonSol  ja, mein Fall ist anders. Ich wollte es trotzdem nicht unerwähnt lassen.
@andig  selber bauen kann ich leider nicht

andig, 2025-08-10:
Ja, in der CONTRIBUTING Doku. Ist in 30min im Nightly.

LeonSol, 2025-08-10:
Ich habe das nigthly installiert. Nachfolgend der log - ich hoffe, dieser enthält alle Informationen für dich @andig 

Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 ----
Aug 10 20:09:08 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:08 charge power: 23W
Aug 10 20:09:08 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:08 charge currents: [1.06 0.868 0.854]A
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 charge power: 0W
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 charge currents: [0 0 0]A
Aug 10 20:09:08 evcc evcc[116693]: [circuit-main] DEBUG 2025/08/10 20:09:08 power: -503W
Aug 10 20:09:08 evcc evcc[116693]: [circuit-main] DEBUG 2025/08/10 20:09:08 current: 0.6A
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 grid power: -503W
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 grid currents: [0.6 0.6 0.6]A
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 battery 1 power: 1286W
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 pv 1 power: 176W
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 accumulated solar yield: pv moved from 151.350 to 151.350
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 battery 1 soc: 95%
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 site power: 883W
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 charge voltages: [0 0 0]V
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 charge total import: 69.051kWh
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 charger status: A
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] INFO 2025/08/10 20:09:08 car disconnected
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 set battery boost: false
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 set session soc limit: 0
Aug 10 20:09:08 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:08 set session energy limit: 0
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 solar forecast: accumulated 0.003 from 2025-08-10 20:08:40.339529233 +0200 CEST m=+58.320763165 to 2025-08-10 20:09:08.875257348 +0200 CEST m=+86.856491336
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 solar forecast: produced 151.350
Aug 10 20:09:08 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:08 solar forecast: accumulated 288.918kWh, produced 151.350kWh, scale 0.524
Aug 10 20:09:08 evcc evcc[116693]: [providerauth] DEBUG 2025/08/10 20:09:08 publishing 0 auth providers
Aug 10 20:09:18 evcc evcc[116693]: [providerauth] DEBUG 2025/08/10 20:09:18 publishing 0 auth providers
Aug 10 20:09:28 evcc evcc[116693]: [providerauth] DEBUG 2025/08/10 20:09:28 publishing 0 auth providers
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 ----
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charge power: 24W
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charge currents: [1.06 0.864 0.854]A
Aug 10 20:09:38 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:38 charge power: 0W
Aug 10 20:09:38 evcc evcc[116693]: [lp-1  ] DEBUG 2025/08/10 20:09:38 charge currents: [0 0 0]A
Aug 10 20:09:38 evcc evcc[116693]: [circuit-main] DEBUG 2025/08/10 20:09:38 power: -0W
Aug 10 20:09:38 evcc evcc[116693]: [circuit-main] DEBUG 2025/08/10 20:09:38 current: 0.6A
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 grid power: -0W
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 grid currents: [0.6 0.6 0.6]A
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 pv 1 power: 221W
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 accumulated solar yield: pv moved from 151.350 to 151.350
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 battery 1 power: 213W
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 battery 1 soc: 95%
Aug 10 20:09:38 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:38 site power: 313W
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charge voltages: [234 233 235]V
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 detected connected phases: 3p
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charge total import: 171.164kWh
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charger status: B
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charge final energy: 0.000kWh
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charge final energy: 0.000kWh
Aug 10 20:09:38 evcc evcc[116693]: [lp-2  ] DEBUG 2025/08/10 20:09:38 charge total import: 171.164kWh
Aug 10 20:09:38 evcc evcc[116693]: [providerauth] DEBUG 2025/08/10 20:09:38 publishing 0 auth providers
Aug 10 20:09:40 evcc evcc[116693]: [script] DEBUG 2025/08/10 20:09:40 /home/pi/EVCC/GET_WPtemp: 53.4
Aug 10 20:09:40 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:40 solar forecast: accumulated 0.003 from 2025-08-10 20:09:08.875440511 +0200 CEST m=+86.856674536 to 2025-08-10 20:09:40.287498112 +0200 CEST m=+118.268732045
Aug 10 20:09:40 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:09:40 solar forecast: produced 151.350
[... cut ...]
Aug 10 20:11:53 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:11:53 solar forecast: accumulated 0.004 from 2025-08-10 20:11:08.821056505 +0200 CEST m=+206.802290530 to 2025-08-10 20:11:53.246470689 +0200 CEST m=+251.227704640
Aug 10 20:11:53 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:11:53 solar forecast: produced 151.450
Aug 10 20:11:53 evcc evcc[116693]: [site  ] DEBUG 2025/08/10 20:11:53 solar forecast: accumulated 288.934kWh, produced 151.450kWh, scale 0.524
Aug 10 20:11:59 evcc evcc[116693]: [providerauth] DEBUG 2025/08/10 20:11:58 publishing 0 auth providers

andig, 2025-08-10:
Ich habs mal formatiert, sehe auf dem Handy aber noch nix

LeonSol, 2025-08-10:
Warte mal morgen ab - ich weiß ja genau, wann es passiert und poste dir den entsprechenden Log. Wird aber erst ca. 14:00 Uhr, da ich früh aus dem Haus muss und das Ereignis erst zwischen 07:00 und 07:30 Uhr eintreten wird.

andig, 2025-08-11:
Ich habs erstmal formatiert und eingekürzt. Wenn jetzt noch der Präfix weg ist kann man die ganzen Zeilen lesen :)

LeonSol, 2025-08-11:
Alles klar! Wird installiert und morgen gibt es mehr Daten - allerdings wieder erst am Nachmittag.

andig, 2025-08-12:
Gönn Deinen Code blocks bitte die

    ```

aussen rum- dann werden sie lesbar formatiert.

Das Problem steckt hier:

    DEBUG 2025/08/12 07:10:46 !! accumulated solar production: pv set 0.100kWh meter total (was: &{clock:0x5f633e0 updated:{wall:13987847351796706142 ext:47004494074309 loc:0x5db84a0} meter:0x40007f8ee0** Accumulated:155.1513612323053})

Die 0.1kWh sind offensichtlich **KEIN** meter total (=Zählerwert) sondern sowas wie ein Tagesenergiezähler. An der Stelle ist das Template (oder Deine Konfiguration) kaputt.

> Configuration details
unerheblich

Dem ist nicht so...

mfuchs1984, 2025-08-12:
> Welcher Zählerstand? 

Der gesamte.

LeonSol, 2025-08-12:
Okay @mfuchs1984 : der gesamte Zählerstand beträgt 16,89 MWh - ja Megawattstunden! Verglichen mit 378,676 kWh. Was soll mir das jetzt bitte bringen?

andig, 2025-08-12:
Bleiben wir doch mal ruhig. Ein Zählerstand wächst monoton. Dein Wert resettet sich auf 0. das führt zu negativer Differenz.

andig, 2025-08-12:
> meine "config" habe ich dir gestern mitgeteilt - du hast den Post auf "hidden" gesetzt! Ich habe mal ein "unhide" gemacht ;-)

Super. Ich hab da gesucht wo ich sie erwartet habe- im Issue. Es hat einen Grund, warum wir danach fragen!

LeonSol, 2025-08-12:
@andig ich dreh mich wirklich im Kreis gerade: ich mach´ den Kram jetzt 48 Jahre (commodore 64 Generation) und frage mich wirklich, warum keiner versteht, dass ich keinen brauchbaren Zählerstand (und die meisten anderen auch nicht!) liefern kann, der ein brauchbares Ergebnis zwischen "accumulated" und "produced" liefert. Wo bitte soll der herkommen? Ich habe jetzt 16,89 MWh (ja Megawattstunden) monoton wachsend!!!!! produziert. Soll ich diesen jetzt unter pv->energy übermitteln? "accumulated" steht bei 378 kWh.

LeonSol, 2025-08-12:
Falsch - nicht das richtige Wort! Missverständlich passt besser. Du bist so tief im Code, das es schwer fällt, deine Posts in Handlungen umzusetzen. Also, für den Normalo:

ich poste in Zukunft "Gesamte produzierte Energie = 16,89 MWh" monoton steigend an evcc und der Drops ist gelutscht?

andig, 2025-08-12:
Bzw.: probiers. Sollte egtl kein Problem sein.

LeonSol, 2025-08-13:
Hat alles wunderbar geklappt. Ich schließe mal diesen langen Issue. @VolkerK62 vielleicht kannst die ja für #22929  einen neuen "issue" eröffnen.

closed 2025-08-13 by LeonSol