evcc-io/evcc #22886: MyPV heaters: Add watchdog for "power timeout" value

docolli, 2025-08-10:
In follow up to this suggestion (https://github.com/evcc-io/evcc/issues/22356#issuecomment-3140536283) and since currently a lot of user problems origin from this non-correct setting, I did take a look at how to get access to this value on my AC THOR 9s.
It is not available from Modbus, but it is available (and can be set) via a URL !

To get the current value for power timeout you need to grab data from here and parse the response for "tout":
```sh
  curl "http://<ip-address>/setup.jsn" -H "User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0" -H "Accept: */*" -H "Accept-Language: de" -H "Accept-Encoding: gzip, deflate" -H "Origin: null" -H "Connection: keep-alive" -H "Priority: u=4"
```
```sh
...
"tout":40,
...
```

To set a new value (here to 40), you can use this URL:

```sh
curl "http://<ip-address>/setup.jsn?tout=40"  -H "User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64; rv:141.0) Gecko/20100101 Firefox/141.0"  -H "Accept: */*"  -H "Accept-Language: de"  -H "Accept-Encoding: gzip, deflate"  -H "Origin: null"  -H "Connection: keep-alive"  -H "Priority: u=0"
```

You get the same response like from above request, so you can check in the response, if `tout` has been set correctly.

I hope this helps. I don't think my skills are enough to implement this by myself. But I can help for testing and minor code changes.

andig, 2025-08-10:
> Add a Log Message to let the user know, what evcc has done.

We can already put that into the yaml file. What needs to happen to keep the watchdog happy?

andig, 2025-08-10:
As discussed before- we can add a watchdog to the yaml, maybe 10s default or user-configurable. it just needs to be implemented...

closed 2025-08-10 by andig