evcc-io/evcc #22893: EVCC 207.2 not starting anymore

thomash73, 2025-08-10:
### Describe the bug

Hi, 
after updating from 207.1 to 207.2 my evcc is not starting anymore.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
my evcc.yaml only contains my token
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/08/10 20:28:33 evcc 0.207.2
[main  ] INFO 2025/08/10 20:28:33 using config file: /config/evcc.yaml
[db    ] INFO 2025/08/10 20:28:33 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/08/10 20:28:34 connecting evcc-1968824852 at tcp://192.168.2.70:1883
[main  ] INFO 2025/08/10 20:28:34 UI listening at :7070
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x30 pc=0x1776674]
goroutine 15 [running]:
github.com/aws/aws-sdk-go-v2/config.addCredentialSource({0x0, 0x0}, 0x14)
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:616 +0x24
github.com/aws/aws-sdk-go-v2/config.resolveCredsFromProfile({0x0?, 0x0?}, 0x4000f0b348, 0x4000e74fc8, 0x4000f0b608, {0x40005dac40, 0x1, 0x1})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:191 +0x294
github.com/aws/aws-sdk-go-v2/config.resolveCredentialChain({0x0, 0x0}, 0x4000f0b348, {0x4000f06f40, 0x3, 0x4})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:123 +0x234
github.com/aws/aws-sdk-go-v2/config.resolveCredentials({0x0, 0x0}, 0x4000f0b348, {0x4000f06f40, 0x3, 0x4})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:70 +0x6c
github.com/aws/aws-sdk-go-v2/config.configs.ResolveAWSConfig({_, _, _}, {_, _}, {_, _, _})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/config.go:160 +0xa8
github.com/aws/aws-sdk-go-v2/config.LoadDefaultConfig({_, _}, {_, _, _})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/config.go:212 +0x170
github.com/evcc-io/evcc/vehicle/fiat.(*Identity).Login(0x4000e18820)
	github.com/evcc-io/evcc/vehicle/fiat/identity.go:164 +0x1708
github.com/evcc-io/evcc/vehicle.NewFiatFromConfig({0x2b5b067?, 0x4000e5bc20?}, 0x4000e5bcb0)
	github.com/evcc-io/evcc/vehicle/fiat.go:53 +0x39c
github.com/evcc-io/evcc/vehicle.NewFromConfig({0x34fc980, 0x4000e5a900}, {0x4000604f5c, 0x4}, 0x4000e5bc20)
	github.com/evcc-io/evcc/vehicle/config.go:47 +0x160
github.com/evcc-io/evcc/vehicle.NewVehicleFromTemplateConfig({0x34fc980, 0x4000e5a900}, 0x40004faea0?)
	github.com/evcc-io/evcc/vehicle/template.go:20 +0x64
github.com/evcc-io/evcc/vehicle.NewFromConfig({0x34fc980, 0x4000e5a900}, {0x400077dd98, 0x8}, 0x4000e5a600)
	github.com/evcc-io/evcc/vehicle/config.go:47 +0x160
github.com/evcc-io/evcc/cmd.vehicleInstance({{0x40003de3a0, 0x4}, {0x400077dd98, 0x8}, 0x4000e5a600})
	github.com/evcc-io/evcc/cmd/setup.go:376 +0xa0
github.com/evcc-io/evcc/cmd.configureVehicles.func2()
	github.com/evcc-io/evcc/cmd/setup.go:449 +0xc0
golang.org/x/sync/errgroup.(*Group).Go.func1()
	golang.org/x/sync@v0.16.0/errgroup/errgroup.go:93 +0x54
created by golang.org/x/sync/errgroup.(*Group).Go in goroutine 1
	golang.org/x/sync@v0.16.0/errgroup/errgroup.go:78 +0x94
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

bierma1966, 2025-08-10:
Same by me, after update evcc didnt Start up. Loge the same AS Post#1

offermannb, 2025-08-11:
Same for me, after updating to 0.207.2 server doesn't start. EV also FIAT

hardoverflow, 2025-08-11:
Same. I reconfigured the template temporary as a `offline` vehicle as workaround:

From:

```yaml
…
vehicles:
  - name: fiat
    type: template
    template: fiat
    title: 'Fiat 500e 3+1'
    user: <redacted>
    password: '<redacted>'
    capacity: 42
…
```

to:

```yaml
…
vehicles:
  - name: fiat
    type: template
    template: offline
    title: 'Fiat 500e 3+1'
    capacity: 42
…
```

hemagi, 2025-08-11:
Same here. Downgrade to 0.207.1 made it running again (`apt install evcc==0.207.1`)

closed 2025-08-11 by andig