evcc-io/evcc #22896: Volvo Integration not auto-detecting connected vehicle

aguilaair, 2025-08-10:
### Describe the bug

[image]

EVCC is not detecting the connected state of the vehicle.

[evcc-20250810-234734-debug.log](https://github.com/user-attachments/files/21707716/evcc-20250810-234734-debug.log)

### Steps to reproduce

1. Set up an API application on the developer console as specified on the EVCC docs
2. Set up the vehicle(s) in EVCC and authenticate wit OAuth
3. Restart EVCC, auto detection will try to run but the state of the vehicle will remain as `not connected`

### Configuration details

```yaml
[main  ] INFO 2025/08/10 23:46:31 using config file: /etc/evcc.yaml
[db    ] INFO 2025/08/10 23:46:31 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/08/10 23:46:31 connecting evcc-1641224088 at tcp://hass.smart:1883
[mqtt  ] DEBUG 2025/08/10 23:46:31 tcp://hass.smart:1883 connected
charger
---
db:7 {Type:template Title: Icon: Product:Shelly} map[channel:1 host:10.3.0.179 integrateddevice:true standbypower: template:shelly]
db:10 {Type:template Title: Icon: Product:Shelly} map[channel:1 host:10.3.0.179 standbypower:15 template:shelly]
db:11 {Type:template Title: Icon: Product:Shelly} map[channel:1 host:10.3.0.179 icon:pump integrateddevice:true standbypower:15 template:shelly]
db:13 {Type:template Title: Icon: Product:Shelly} map[channel:0 heating:true host:10.3.0.179 icon:dryer integrateddevice:true standbypower:50 template:shelly]
db:23 {Type:template Title: Icon: Product:Shelly Pro 1} map[channel:0 heating:true host:10.3.0.77 icon:dishwasher integrateddevice:true standbypower:-550 template:shelly]

vehicle
---
db:19 {Type:template Title: Icon: Product:volvo2mqtt} map[capacity:18.8 icon:car maxCurrent:16 minCurrent:6 mode: phases:1 priority:1 template:volvo2mqtt timeout:720h title:XC60 vin:*****]
db:20 {Type:template Title: Icon: Product:volvo2mqtt} map[capacity:69 icon:car maxCurrent:32 minCurrent:6 phases: template:volvo2mqtt timeout:720h title:XC40 vin:*****]
db:21 {Type:template Title: Icon: Product:Volvo} map[capacity:18.8 clientId:dc-79vial1p3yknyc54rxgqrcc6d clientSecret:fwiydajjpLNAvqPv8FtUfE icon:car maxCurrent:16 minCurrent:6 phases:1 redirectUri:https://evcc.edm.sh/providerauth/callback template:volvo-connected title:Bluey vccapikey:08b26c1cd0c44289b09b26ac6b94477f vin:*****]
db:22 {Type:template Title: Icon: Product:Volvo} map[capacity:69 clientId:dc-79vial1p3yknyc54rxgqrcc6d clientSecret:fwiydajjpLNAvqPv8FtUfE icon:car maxCurrent:32 minCurrent:6 redirectUri:https://evcc.edm.sh/providerauth/callback template:volvo-connected title:Myc vccapikey:08b26c1cd0c44289b09b26ac6b94477f vin:*****]

loadpoint
---
db:12 {Type: Title: Icon: Product:} map[charger:db:11 circuit: defaultMode: limitEnergy:0 maxCurrent:6 meter: minCurrent:3 mode:pv phasesConfigured:1 priority:1 smartCostLimit:<nil> soc:map[estimate:true poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:2e+10 threshold:0]] title:Bomba Piscina vehicle:]
db:14 {Type: Title: Icon: Product:} map[charger:db:13 circuit: defaultMode: maxCurrent:6 meter: mode:pv phasesConfigured:1 priority:0 smartCostLimit:<nil> soc:map[estimate:true poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Bomba de Calor Piscina vehicle:]
db:24 {Type: Title: Icon: Product:} map[charger:db:23 circuit: defaultMode: maxCurrent:3 meter: minCurrent:0.2 mode:pv phasesConfigured:1 priority:2 smartCostLimit:<nil> soc:map[estimate:true poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Bomba de Calor ACS vehicle:]
```

### Log details

```text
[main  ] DEBUG 2025/08/10 23:41:26 vehicle status: A (Bluey)
[main  ] DEBUG 2025/08/10 23:41:26 vehicle status: A (Myc)

But in this case Bluey is plugged in
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.2 (eda1ced3)

aguilaair, 2025-08-10:
I should note that this is not a new issue and has been happening for at least a month now

andig, 2025-08-11:
Please show

    evcc vehicle —log trace

aguilaair, 2025-08-11:
as a side note it would be nice if the dump command removed that automatically (i assumed it did)

aguilaair, 2025-08-12:
testing on the latest nightly (c8a30f24) and seeing the same issue, @lehmanju 

```
db:21
-----
[volvo-connected] TRACE 2025/08/12 07:06:36 GET https://api.volvocars.com/energy/v2/vehicles/***/state
[volvo-connected] TRACE 2025/08/12 07:06:36 {
  "batteryChargeLevel" : {
    "status" : "OK",
    "value" : 45.0,
    "unit" : "percentage",
    "updatedAt" : "2025-08-11T20:52:55Z"
  },
  "electricRange" : {
    "status" : "OK",
    "value" : 29,
    "unit" : "km",
    "updatedAt" : "2025-08-11T20:52:55Z"
  },
  "chargerConnectionStatus" : {
    "status" : "OK",
    "value" : "CONNECTED",
    "updatedAt" : "2025-08-11T20:52:55Z"
  },
  "chargingStatus" : {
    "status" : "OK",
    "value" : "IDLE",
    "updatedAt" : "2025-08-11T20:52:55Z"
  },
  "chargingType" : {
    "status" : "OK",
    "value" : "AC",
    "updatedAt" : "2025-08-11T20:52:55Z"
  },
  "chargerPowerStatus" : {
    "status" : "OK",
    "value" : "NO_POWER_AVAILABLE",
    "updatedAt" : "2025-08-11T20:52:55Z"
  },
  "estimatedChargingTimeToTargetBatteryChargeLevel" : {
    "status" : "OK",
    "value" : 0,
    "unit" : "minutes",
    "updatedAt" : "2025-08-11T20:52:55Z"
  },
  "chargingCurrentLimit" : {
    "status" : "ERROR",
    "code" : "NOT_SUPPORTED",
    "message" : "Resource is not supported for this vehicle"
  },
  "targetBatteryChargeLevel" : {
    "status" : "ERROR",
    "code" : "ERROR_READING_PROPERTY",
    "message" : "Failed to retrieve property."
  },
  "chargingPower" : {
    "status" : "ERROR",
    "code" : "NOT_SUPPORTED",
    "message" : "Resource is not supported for this vehicle"
  }
}
[volvo-connected] TRACE 2025/08/12 07:06:36 GET https://api.volvocars.com/connected-vehicle/v2/vehicles/***/odometer
[volvo-connected] TRACE 2025/08/12 07:06:36 {
  "data" : {
    "odometer" : {
      "value" : 35213,
      "unit" : "km",
      "timestamp" : "2025-08-11T20:42:47.844Z"
    }
  }
}
Soc:           45%
Capacity:      18.8kWh
Charge status: A
Range:         29km
Odometer:      35213km
Finish time:   0000-12-31 23:45:16 -0014 LMT
Limit Soc:     0%
OnIdentified:  MinCurrent:6, MaxCurrent:16

db:22
-----
[volvo-connected] TRACE 2025/08/12 07:06:36 GET https://api.volvocars.com/energy/v2/vehicles/***/state
[volvo-connected] TRACE 2025/08/12 07:06:36 {
  "batteryChargeLevel" : {
    "status" : "OK",
    "value" : 79.0,
    "unit" : "percentage",
    "updatedAt" : "2025-08-12T00:28:57Z"
  },
  "electricRange" : {
    "status" : "OK",
    "value" : 310,
    "unit" : "km",
    "updatedAt" : "2025-08-12T00:28:57Z"
  },
  "chargerConnectionStatus" : {
    "status" : "OK",
    "value" : "DISCONNECTED",
    "updatedAt" : "2025-08-12T00:28:57Z"
  },
  "chargingStatus" : {
    "status" : "OK",
    "value" : "IDLE",
    "updatedAt" : "2025-08-12T00:28:57Z"
  },
  "chargingType" : {
    "status" : "OK",
    "value" : "NONE",
    "updatedAt" : "2025-08-12T00:28:57Z"
  },
  "chargerPowerStatus" : {
    "status" : "OK",
    "value" : "NO_POWER_AVAILABLE",
    "updatedAt" : "2025-08-12T00:28:57Z"
  },
  "estimatedChargingTimeToTargetBatteryChargeLevel" : {
    "status" : "OK",
    "value" : 0,
    "unit" : "minutes",
    "updatedAt" : "2025-08-12T00:28:57Z"
  },
  "chargingCurrentLimit" : {
    "status" : "OK",
    "value" : 32,
    "unit" : "ampere",
    "updatedAt" : "2023-12-25T18:30:05Z"
  },
  "targetBatteryChargeLevel" : {
    "status" : "OK",
    "value" : 100,
    "unit" : "percentage",
    "updatedAt" : "2025-05-26T05:59:01Z"
  },
  "chargingPower" : {
    "status" : "ERROR",
    "code" : "PROPERTY_NOT_FOUND",
    "message" : "No valid value could be found for the requested property"
  }
}
[volvo-connected] TRACE 2025/08/12 07:06:36 GET https://api.volvocars.com/connected-vehicle/v2/vehicles/***/odometer
[volvo-connected] TRACE 2025/08/12 07:06:36 {
  "data" : {
    "odometer" : {
      "value" : 51832,
      "unit" : "km",
      "timestamp" : "2025-08-11T15:55:33.833Z"
    }
  }
}
Soc:           79%
Capacity:      69.0kWh
Charge status: A
Range:         310km
Odometer:      51832km
Finish time:   0000-12-31 23:45:16 -0014 LMT
Limit Soc:     100%
OnIdentified:  MinCurrent:6, MaxCurrent:32
```

lehmanju, 2025-08-12:
It only detects the car if it is actively charging, not if it is just connected and idle.

PR #22930 (2025-08-12): fix(volvo): chargerConnectionStatus field name
Fix #22896 

noch ein typo ...