evcc-io/evcc #22921: evcc startet nicht

Ralf-Th, 2025-08-11:
### Describe the bug

evcc startet nicht
panic bei modbus proxy

### Steps to reproduce

1.reboot
2.
3.
...

### Configuration details

```yaml
wie immer
```

### Log details

```text
main  ] INFO 2025/08/11 18:43:28 evcc 0.207.2
[main  ] INFO 2025/08/11 18:43:28 using config file: /config/evcc.yaml
[db    ] INFO 2025/08/11 18:43:28 using sqlite database: /data/evcc.db
[main  ] INFO 2025/08/11 18:43:28 UI listening at :7070
[proxy-502] DEBUG 2025/08/11 18:43:28 modbus proxy for 172.17.3.85:502 listening at :502
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x30 pc=0x1776674]
goroutine 398 [running]:
github.com/aws/aws-sdk-go-v2/config.addCredentialSource({0x0, 0x0}, 0x14)
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:616 +0x24
github.com/aws/aws-sdk-go-v2/config.resolveCredsFromProfile({0x0?, 0x0?}, 0x4000ef2b08, 0x4000986908, 0x4000ef2dc8, {0x4000355810, 0x1, 0x1})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:191 +0x294
github.com/aws/aws-sdk-go-v2/config.resolveCredentialChain({0x0, 0x0}, 0x4000ef2b08, {0x40006a0b80, 0x3, 0x4})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:123 +0x234
github.com/aws/aws-sdk-go-v2/config.resolveCredentials({0x0, 0x0}, 0x4000ef2b08, {0x40006a0b80, 0x3, 0x4})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/resolve_credentials.go:70 +0x6c
github.com/aws/aws-sdk-go-v2/config.configs.ResolveAWSConfig({_, _, _}, {_, _}, {_, _, _})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/config.go:160 +0xa8
github.com/aws/aws-sdk-go-v2/config.LoadDefaultConfig({_, _}, {_, _, _})
	github.com/aws/aws-sdk-go-v2/config@v1.30.3/config.go:212 +0x170
github.com/evcc-io/evcc/vehicle/fiat.(*Identity).Login(0x4000cc8a50)
	github.com/evcc-io/evcc/vehicle/fiat/identity.go:164 +0x1708
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

neueste

LeonSol, 2025-08-11:
Einfach das aktuelle nightly installieren, das läuft es wieder...war ein Problem mit Fiat!

Ralf-Th, 2025-08-11:
ich habe keinen Plan, was das mit Fiat zu tun hat,
aber evcc Updates fallen für mich erstmal aus.
Das ist Stress pur mit Grosseltern im Haus, wenn man selbst im Urlaub ist.

LeonSol, 2025-08-11:
Dann geh´ mit "sudo apt install evcc=0.206.0" wieder auf die vorherige stabile Version zurück!

closed 2025-08-11 by andig