evcc-io/evcc #22936: WARN  charger logic error: current mismatch

Ministry986, 2025-08-12:
### Describe the bug

Hi everyone, I'm trying to get evccc to work to charge my car with only photovoltaic energy but I always get this error: charger logic error: current mismatch

my setup is:

Shelly EM dual channel ( 1 channel pv , 1 channel grid) (192.168.1.243 )
Wallbox FOX ESS A7300S1-E-2-WO  Firmware 1.17 , ocp firmware 1.03   ( ocpp link ws://192.168.1.249:8887 , profile 0 , no name )
Home Assistant on Raspberry Pi 4 last version of EVCC ( 192.168.1.249:7070)
Car Dacia Spring 2024 26kwh

with trial sponsored It was working, as soon as I put the regular key in it stopped working

I put the logs and settings in the evcc.yaml file

### Steps to reproduce

I tried everything but I can't solve it

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)
modbusproxy:
[... 397 lines cut ...]
[lp-1 ] DEBUG 2025/08/12 13:54:59 charge total import: 0.130kWh
[lp-1 ] DEBUG 2025/08/12 13:55:00 soc estimated: 62.24% (vehicle: 62.00%)
[lp-1 ] DEBUG 2025/08/12 13:55:00 vehicle soc: 62%
[lp-1 ] DEBUG 2025/08/12 13:55:00 vehicle range: 114km
[lp-1 ] WARN 2025/08/12 13:55:00 charger logic error: current mismatch (got 18.9A measured, expected 16.3A) - make sure your interval is at least 30s
[lp-1 ] DEBUG 2025/08/12 13:55:00 pv charge current: 17.8A = 0A + 17.8A (-4083W @ 1p)
[lp-1 ] DEBUG 2025/08/12 13:55:00 set charge current limit: 17.8A
[site ] DEBUG 2025/08/12 13:55:09 ----
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v.0.207.1

Ministry986, 2025-08-12:
Fox Switch settings:

[image]

Evcc page ( without error)   (WARN charger logic error: current mismatch generate orange allert):

[image]

Ministry986, 2025-08-12:
sorry , i have the same problem with 30s

andig, 2025-08-12:
Then I can only suggest contacting the vendor.