evcc-io/evcc #22948: Volvo oauth-Autorisierung schlägt fehl mit "404 page not found" des redirect-Links (http)

JoGroW, 2025-08-12:
### Describe the bug

Nach Update von 0.205.0 auf 0.207.2 schlägt die oauth-Autorisierung (Volvo) fehl.

Bei 0.205.0
Nach Start von evcc wird im Log der Teillink /oauth/login?id=f19cbdc7f.... ausgegeben.
Über manuelle Vervollständigung zu http://192.168.x.y:7070/oauth?id=f19cbdc7f....  erfolgt die Weiterleitung zu https://volvoid.eu.volvocars.com/as/authorization.oauth2?client_id=....

Dort erfolgt der Login mit der Volvo-ID und Weiterleitung auf den Redirect-Link
https://192.168.2.11:7070/oauth/callback?code=tmb.....

Dieser funktioniert aufgrund des https nicht. Nach manuellem entfernen des s in https zu
http://192.168.2.11:7070/oauth/callback?code=tmb.....

funktioniert der redirect-link, die evcc-Startseite wird angezeigt und die Autorisierung ist abgeschlossen. Die aktuellen Daten werden beim Fahrzeug angezeigt (mit Befehl evcc vehicle auf der Konsole, in der UI dauert es manchmal eine Weile).

Bei 0.207.2
Der outh-Teillink wird nicht mehr im Log ausgegeben (habe ihn zumindest in keiner der Log-Varianten gefunden).
Dafür gibt es in der UI nun den Link unter Autorisierungstatus.

Bei klick auf den entsprechenden Link in der UI erfolgt wie zuvor bei Version 0.205.0 die Weiterleitung auf die Volvo-Seite mit dem Login. Nach Login erfolgt wieder die Ausgabe des Redirect-Link mit https. Nach manuellem Abändern auf http:// wird jedoch nicht die evcc-Startseite angezeigt, es kommt die Meldung "404 page not found". Die Daten des Fahrzeuges werden auch nicht abgerufen (versucht über evcc vehicle).

Im Log findet sich der Fehler
[httpd ] ERROR 2025/08/12 22:29:05 http: superfluous response.WriteHeader call from github.com/gorilla/handlers.(*compressResponseWriter).WriteHeader (compress.go:26)

### Steps to reproduce

1. Klick auf Fahrzeug-Link unter Autorisierungstatus
2. Login mit Volvo-ID bei Volvo
3. Manuelle Anpassung des Redirect-Links auf http:// und Aufruf dieses Links führt zu 404

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

log: debug
levels:
  cache: error

# unique installation id
plant: 1234

interval: 10s # control cycle interval

sponsortoken: 1234

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: fronius-solarapi-v1 
  usage: grid  
  host: 192.168.2.10  
  user: customer  
  password: Groos_Westhausen21  
  name: grid1
- type: template
  template: fronius-solarapi-v1 
  usage: pv  
[... 229 lines cut ...]
[site ] DEBUG 2025/08/12 23:09:13 solar forecast: accumulated 1867.171kWh, produced 1795.978kWh, scale 0.962
[providerauth] DEBUG 2025/08/12 23:09:16 login request for provider: C40_einphasig_connected (f19cbdc7)
[httpd ] ERROR 2025/08/12 23:09:16 http: superfluous response.WriteHeader call from github.com/gorilla/handlers.(*compressResponseWriter).WriteHeader (compress.go:26)
[providerauth] DEBUG 2025/08/12 23:09:22 publishing 2 auth providers
[site ] DEBUG 2025/08/12 23:09:22 ----
[lp-1 ] DEBUG 2025/08/12 23:09:22 charge power: 0W
[lp-1 ] DEBUG 2025/08/12 23:09:22 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/12 23:09:23 grid power: -5W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.2 (c8a30f24)

andig, 2025-08-13:
@JoGroW danke für die ausführliche Beschreibung.
@lehmanju hat sich hier evtl. noch ein Fehler eingeschlichen?

lehmanju, 2025-08-13:
Ja, war Absicht. Damit man die UI auch für nicht oauth Provider nutzen kann.

CFenner, 2025-08-22:
> > Habe bei Volvo eine neue Application angelegt und den Redirect angepasst.

Oh je, das ist ja jetzt ein Aufwand, blöd, dass man das nicht anpassen kann.

erniemctoeff, 2025-08-27:
Ich hatte Volvo vor einigen Monaten mal laufen. Jetzt habe ich auf die aktuellste Version von evcc geupdatet. Ich weiß noch, dass man bei Volvo beim Publishen bei Redirect url zwingend https:// mit 's' eingeben muss. Habe ich. Trotzdem kommt dann bei Volvo wenn ich auf 'submit' klicke immer 'something went wrong, please try again later'. Ich probiere das gerade schon seit ner stunde :( irgendwer Tipps?

Meine Redirect url lautet in folgender Form: https://192.168.xxx.xxx:7070/#/evcc/providerauth

Danke!

cwbaer, 2025-09-02:
Hallo,
ich hänge am selben Problem wie erniemctoeff.
Als RedirectURI habe ich "https://192.168.178.84:7070/#/evcc/providerauth" angegeben. 
Leider kann ich meine App im Volvo Dev Account nun auch nicht publishen. 
Weißt jemand weiter?

Viele Grüße
Carsten

cwbaer, 2025-09-02:
> Nehmt mal das /# aus dem redirect-link, dann sollte es gehen.

Kaum macht man es richtig, gehts auch schon ;-) 

Vielen Dank