evcc-io/evcc #22949: Sponsor Token Expiration Prevents Plugin Start and UI Access – No Simple Way to Update

ButtiBBQ, 2025-08-12:
### Describe the bug

I configured the test sponsor token using the experimental UI.
After the token expired, the evcc plugin was no longer able to start.

`starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/08/12 14:23:29 evcc 0.207.2
[main  ] INFO 2025/08/12 14:23:29 using config file: /config/evcc.yaml
[db    ] INFO 2025/08/12 14:23:29 using sqlite database: /data/evcc.db
[main  ] INFO 2025/08/12 14:23:29 UI listening at :7070
[main  ] FATAL 2025/08/12 14:23:30 sponsorship sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2025/08/12 14:23:30 will attempt restart in: 15m0s`

Without the plugin running, there is no access to the UI to update the token.

The documentation states:

"Es ist auch möglich das Token über die UI auf der Konfigurationsseite zu setzen. Beachte: Diese Eingabe wird in der evcc.db gespeichert und hat Vorrang gegenüber der evcc.yaml."
This means that updating the token in evcc.yaml has no effect as long as there is an entry in the database. Since the UI is unavailable, the only remaining option is to enter the Docker container and manually update the entry in the SQLite database (evcc.db).
This is not a good user experience.

Expected Behavior:
There should be an easy way to update the token without the plugin running, for example on of this solutions:

- UI should start even the Token has expired
- Fallback to evcc.yaml when the token has expired
- A dedicated CLI command
- A minimal UI for updating the auth token independent of plugin startup

Environment:

Home Assistant (Suvervisor installation) on Raspberry 3
evcc Plugin 

### Steps to reproduce

1. Configure Test-Sponser-Token in experimental UI
2. wait until token expires
3. try to strart the evcc plugin in home assistant
...

### Configuration details

```yaml
no config details. Sponsor Token used from webpage:

# trial token, valid until 2025-08-12
sponsortoken: eyJhbGciOiJ...

no evcc.yaml
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/08/12 14:23:29 evcc 0.207.2
[main  ] INFO 2025/08/12 14:23:29 using config file: /config/evcc.yaml
[db    ] INFO 2025/08/12 14:23:29 using sqlite database: /data/evcc.db
[main  ] INFO 2025/08/12 14:23:29 UI listening at :7070
[main  ] FATAL 2025/08/12 14:23:30 sponsorship sponsortoken: rpc error: code = Unknown desc = token has invalid claims: token is expired
[main  ] FATAL 2025/08/12 14:23:30 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.2

andig, 2025-08-13:
> Without the plugin running, there is no access to the UI to update the token.

Are you sure?

saessli, 2025-08-13:
same for me

andig, 2025-08-13:
https://github.com/evcc-io/evcc/issues/22949#issuecomment-3182277425 ?