evcc-io/evcc #22963: Allow https mDNS advertising

naltatis, 2025-08-13:
**Is your feature request related to a problem? Please describe.**
Make it possible to mDNS advertise evcc correctly when operated behind a reverse proxy for https. Relevant for instance detection of the iOS/Android app. See https://github.com/evcc-io/app/issues/1

Szenario: evcc run on a dedicated machine (http 7070). For encryption (and basic auth) a reverse proxy (e.g. caddy) is added. App should connect via https://evcc.local/ (https 443).

**Describe the solution you'd like**
Extend or rework the network config section to support this scenario. Current options:

```
network:
  schema: http|https # only used for callback urls
  host: evcc.local # used for callback urls and activating mDNS (when ends with `.local`)
  port: 7070 # port to start http(!) server on, used to generate callback urls
```

The schema definition has no effect on mDNS today. Taking it to account would not help because we've no way to specify the port that should be used in a proxy situation.

We're currently mixing different network related use-cases here

- go network setup (http `port`)
- callback urls (used afaik only by MCP and Tronity)
- mDNS advertising (used by app and eebus?)

I don't see an obvious solution to cleanly support mDNS+HTTPS without some rework of the settings. Maybe we have to introduce dedicated optional "Public Callback" and "mDNS" sections.

Blocks https://github.com/evcc-io/app/pull/74

\cc @Maschga

andig, 2025-08-13:
> Szenario: evcc run on a dedicated machine (http 7070). For encryption (and basic auth) a reverse proxy (e.g. caddy) is added. App should connect via https://evcc.local/ (https 443).

What's the problem? What is the observed vs. the desired behaviour?

naltatis, 2025-08-13:
> What's the problem? What is the observed vs. the desired behaviour?

For the example above evcc advertises as.

```
_http._tcp evcc.local:7070
```

desired behavior would be to get

```
_https._tcp evcc.local:443
```

Currently our mdns implementation ignores `schema` (easy to fix) and we dont have a way to specify a port that should be used for advertising (we can't use `port` here because this is the go http server port). This new "external port", "public port", .. should imho also be used for the callback url generation.

Bonus question: I know that eebus implementation has special requirements for our mdns. I'm unsure if those devices would also be compatible with this or if we need parallel advertising.

andig, 2025-08-13:
> What's the problem?

Mir ist nicht klar wofür wir das brauchen? Solange ich unter lokalem Namen (.local) publishe ist doch :7070 völlig ok?

sdebruyn, 2025-08-14:
Reverse proxy is not per se used for external traffic. It's also easier to centrally manage a correctly signed SSL certificate and then expose all your self-hosted apps through the reverse proxy (Caddy). Evcc might not be reachable locally because of a firewall configuration you might have applied. You could even choose to not expose its ports outside of a Docker network and have evcc and Caddy in the same Docker network and then let Caddy expose the http/https ports.

This is why most self-hosted apps allow you to set the URL at which a service is reachable (e.g. Home Assistant, all the *arrs, ...)

E.g.

EVCC:

```yaml
services:
  evcc:
    network_mode: bridge
    command:
      - evcc
    container_name: evcc
    image: evcc/evcc:latest
    ports:
      - 7070:7070/tcp
      - 8887:8887/tcp
    volumes:
      - ./evcc.yaml:/etc/evcc.yaml
      - ./config:/root/.evcc
    restart: unless-stopped
networks: {}
```

Caddy:

```yaml
services:
  caddy:
    image: iarekylew00t/caddy-cloudflare:latest
    container_name: caddy
    network_mode: bridge
    restart: unless-stopped
    env_file: .env
    cap_add:
      - NET_ADMIN
      - NET_BIND_SERVICE
    ports:
      - 80:80
      - 443:443
      - 443:443/udp
    volumes:
      - ./Caddyfile:/etc/caddy/Caddyfile
      - caddy_data:/data
      - caddy_config:/config
volumes:
  caddy_data: null
  caddy_config: null
networks: {}
```

My Caddyfile:
```
*.domain.tld {
    tls {
        dns cloudflare {env.CF_API_TOKEN}
        resolvers 1.1.1.1 1.1.1.2
        propagation_delay 1m
        propagation_timeout 5m
        dns_ttl 5m
    }

    ... 20+ other apps

    @ev host ev.domain.tld
    handle @ev {
        reverse_proxy hostnameofthemachinehostingevcc:7070
    }
}
```

jmerifjKriwe, 2025-11-12:
Halb on-topic, halb off-topic. 
Wenn du aktuell in der UI die Konfiguration für das Netzwerk klickst und, weil du nen Reverse Proxy vorgeschaltet hast, wie folgt änderst:
- Schema: https
- Hostname: evcc.local
- Port: 443 (weil das ja der Port für https ist)

Dann fährt anschließend evcc nicht mehr hoch. (und man darf anfangen in der DB rumzuändern).

Es steht nirgendwo, wofür der Port ist und das man den eigentlich nicht anpassen sollte.