evcc-io/evcc #22987: Ich stelle schon seit langen fest, das im evcc Gui (und influx) der aktuelle SOC und die aktuelle Reichweite nicht mit der Audi-App übereinstimmt.

mfuchs1984, 2025-08-14:
Ich stelle schon seit langen fest, das im evcc Gui (und influx) der aktuelle SOC und die aktuelle Reichweite nicht mit der Audi-App übereinstimmt.
Beim SOC ist es nicht so schlimm, da evcc sie schätzt und fast genau dem des Autos entspricht.
Die aktuelle Reichweite wird aber nur sehr selten im Gui und in influx aktualisiert, obwohl im evcc Log (Audi trace) schon viel früher aktuelle Werte für SOC und Reichweite stehen. 

Eine Idee woran das liegen kann ? Die neueren Informationen sind ja schon ausgelesen worden.

[image]

[evcc-20250803-145706-trace.log](https://github.com/user-attachments/files/21565918/evcc-20250803-145706-trace.log)

_Originally posted by @Towi68 in https://github.com/evcc-io/evcc/discussions/22720_

andig, 2025-08-14:
Wie bereits in der Discussion geschrieben: bei mir passt es. Es fehlt ein klares Bild, welcher Wert stimmt. Ein bestimmter? Immer der Letzte? Ich mache hier zu bis es dazu eine verifizierte Erkenntnis gibt. Gerne weiter in der Discussion.

mfuchs1984, 2025-08-22:
Bisher sieht es so aus, als wäre charging immer der aktuellste, aber die Datenbasis ist noch gering. Sicherer - aber wahrscheinlich unnötig - wäre wohl, den timestamp auszuwerten.

premultiply, 2025-08-22:
Ist das eine vielleicht der "UserSoC" (charging) und die anderen beiden der "RealSoC"?

andig, 2025-08-22:
Closed in 5167700c7069e4bcba4b5c1c5a3c45c9df201203