evcc-io/evcc #22997: Trigger automatic backup download via curl

michis0806, 2025-08-15:
**Is your feature request related to a problem? Please describe.**

I am very forgetful and therefore want to automate as many processes as possible. This also applies to backups, of course. I have therefore automated all backups for my home server. Until recently, I was still running evcc via proxmox and backed up evcc.yaml and .db there via scp. However, I have now switched from proxmox to a HomeAssistant add-on. However, I don't want to rely solely on HomeAssistant for backups.

**Describe the solution you'd like**

For my application, it would be helpful if I could trigger the new backup with a simple curl call and download the file. Something like this, for example:

```
curl -i http://evcc.local:7070/backup \
  -H "Authorization: Bearer mypassword"
```

there is a similar possibility for deconz which I found very helpful when I was using deconz some time ago

`curl -k -S -v -H "Content-Type: application/json" http://192.168.188.203:9090/ -X "POST /api/"NEWUSERNAME"/config/export"
`

**Describe alternatives you've considered**

My alternatives are trying to write a script that connects to HomeAssistant via ssh and fetch the files in connecting to the Add-On Container thru the console.

Thanks for your great work!
Michael

andig, 2025-08-15:
You would expect the response to contain the actual file disposition, right?

naltatis, 2025-08-15:
Hi @michis0806, it's currently possible to do this. You can use the same endpoint the UI uses to download the backup.

The endpoint is `POST /system/backup`. You have to provide your admin password as payload since the backup contains sensible data. 

See https://github.com/evcc-io/evcc/blob/master/server/http.go

You also need to provide a valid cookie which you can get from `/auth/login`. In https://github.com/evcc-io/evcc/issues/22099 we're planning to introduce an alternative that will make this step easier.

So you can implement your backup routine right now.

michis0806, 2025-08-15:
@andig: that's right ;)
@naltatis: thanks for the input. I'll check that out 

Michael

closed 2025-08-15 by andig

michis0806, 2025-08-15:
Meine Lösung:

```
#!/bin/bash
set -euo pipefail

# === Konfiguration ===
EVCC_URL="${EVCC_URL:-http://evcc.local:7070}"   # ggf. http://evcc.local:7070
PASSWORD="${PASSWORD:-ADMIN_PASSWORD}"
BACKUP_DIR="${BACKUP_DIR:-/tmp}"
BACKUP_FILE="${BACKUP_DIR}/$(date +%Y-%m-%d).evcc.db"

COOKIE_JAR="$(mktemp)"
trap 'rm -f "$COOKIE_JAR"' EXIT

# Optional: Self-signed TLS zulassen (EVCC_INSECURE=1 setzen)
CURL_OPTS=(--silent --show-error --fail-with-body)
[[ "${EVCC_INSECURE:-0}" = "1" ]] && CURL_OPTS+=(-k)

# === 1) Login: Auth-Cookie holen ===
curl "${CURL_OPTS[@]}" -c "$COOKIE_JAR" \
  -H "Content-Type: application/json" \
  -X POST "$EVCC_URL/api/auth/login" \
  -d "{\"password\":\"$PASSWORD\"}" >/dev/null

# Sanity-Check: eingeloggt?
LOGIN_STATUS="$(curl "${CURL_OPTS[@]}" -b "$COOKIE_JAR" "$EVCC_URL/api/auth/status" || true)"
if [[ "$LOGIN_STATUS" != "true" ]]; then
  echo "Login fehlgeschlagen (auth/status=$LOGIN_STATUS). Passwort/URL prüfen."
  exit 1
fi

HTTP_CODE="$(curl "${CURL_OPTS[@]}" -w "%{http_code}" -b "$COOKIE_JAR" \
-H 'Content-Type: application/json' -X POST \
-d "{\"password\":\"$PASSWORD\"}" \
-o "$BACKUP_FILE" "$EVCC_URL/api/system/backup" || true)"

if [[ "$HTTP_CODE" != "200" || ! -s "$BACKUP_FILE" ]]; then
  echo "Backup fehlgeschlagen (HTTP $HTTP_CODE). Endpoint ist evcc-versionsabhängig."
  echo "Workaround: /etc/evcc.yaml und /var/lib/evcc/evcc.db direkt sichern (siehe Doku)."
  exit 1
fi

echo "Backup gespeichert: $BACKUP_FILE"
```

michis0806, 2025-08-15:
```
# 1) Login (Cookie speichern)
curl -c cookie.txt -H "Content-Type: application/json" \
  -X POST http://evcc.local/api/auth/login \
  -d '{"password":"DEIN_PASSWORT"}'

# 2) Status prüfen -> sollte 'true' zurückgeben
curl -b cookie.txt http://evcc.local/api/auth/status

# 3) Backup holen (Variante A: GET)
curl -b cookie.txt -o evcc-backup.tar.gz http://evcc.local/api/system/backup

# 3b) Falls A scheitert: Variante B: POST mit Passwort im Body
curl -b cookie.txt -H "Content-Type: application/json" \
  -X POST -d '{"password":"DEIN_PASSWORT"}' \
  -o evcc-backup.tar.gz http://evcc.local/api/system/backup

```

andig, 2025-08-15:
getBackup prozessiert den selbst, sollte also gehen.

michis0806, 2025-08-15:
Ohne den Cookie-Zwischenschritt über die /api/auth/login bekomm ich immer ein 'unauthorized' zurück.