evcc-io/evcc #23010: Wattsonic - improvements to the config UI and data

frankb-CZ, 2025-08-15:
### Describe the bug

Hi, I checked the new Wattsonic config and wanted to check active battery control you added based on my comment, thanks for that! I found out some issues with the implementation.

1. Config UI suggests MODBUS ID 247 but I know that only 255 works. You can check in my manual config I used before.
2. SOC is not being read out correctly. It should be visible on the register 33000. 
3. You can read also PV1 and PV2 separately. Check my config as well.

Manual config I used before is below in config delays.

### Steps to reproduce

1.Start with fresh EVCC installation
2.Add Wattsonic device (either battery, PV or grid meter)
3.You see wrong MODBUS ID
4.When you add battery you can see that SOC cannot be read out.
...

### Configuration details

```yaml
Previous working manual config: 

meters:
- name: meter-pv1
  type: custom
  power:
   source: modbus
   delay: 100ms
   uri: 192.168.1.71:502
   id: 255
   rtu: false
   timeout: 10s
   register:
    address: 11062
    type: holding
    decode: int32
  currents:
   - source: modbus
     delay: 100ms
     uri: 192.168.1.71:502
     id: 255
     rtu: false
     timeout: 10s
     register:
      address: 11039
      type: holding
      decode: uint16
     scale: 0.1
   - source: const
     value: 0
   - source: const
     value: 0
- name: meter-pv2
  type: custom
  power:
   source: modbus
   delay: 100ms
   uri: 192.168.1.71:502
   id: 255
   rtu: false
   timeout: 10s
   register:
    address: 11064
    type: holding
    decode: int32
  currents:
   - source: modbus
     delay: 100ms
     uri: 192.168.1.71:502
     id: 255
     rtu: false
     timeout: 10s
     register:
      address: 11041
      type: holding
      decode: uint16
     scale: 0.1
   - source: const
     value: 0
   - source: const
     value: 0
- name: meter-grid
  type: custom
  power:
   source: modbus
   delay: 100ms
   uri: 192.168.1.71:502
   id: 255
   rtu: false
   timeout: 10s
   register:
    address: 11000
    type: holding
    decode: int32
   scale: -1
  powers:
   - source: modbus
     delay: 100ms
     uri: 192.168.1.71:502
     id: 255
     rtu: false
     timeout: 10s
     register:
      address: 10994
      type: holding
      decode: int32
     scale: -1
   - source: modbus
     delay: 100ms
     uri: 192.168.1.71:502
     id: 255
     rtu: false
     timeout: 10s
     register:
      address: 10996
      type: holding
      decode: int32
     scale: -1
   - source: modbus
     delay: 100ms
     uri: 192.168.1.71:502
     id: 255
     rtu: false
     timeout: 10s
     register:
      address: 10998
      type: holding
      decode: int32
     scale: -1
- name: meter-battery
  type: custom
  power:
   source: modbus
   delay: 100ms
   uri: 192.168.1.71:502
   id: 255
   rtu: false
   timeout: 10s
   register:
    address: 30258
    type: holding
    decode: int32
  soc:
   source: modbus
   delay: 100ms
   uri: 192.168.1.71:502
   id: 255
   rtu: false
   timeout: 10s
   register:
    address: 33000
    type: holding
    decode: uint16
   scale: 0.01
  currents:
   - source: modbus
     delay: 100ms
     uri: 192.168.1.71:502
     id: 255
     rtu: false
     timeout: 10s
     register:
      address: 30255
      type: holding
      decode: int16
     scale: 0.1
   - source: const
     value: 0
   - source: const
     value: 0
  capacity: 11.7
```

### Log details

```text
I am sorry, I do not know how to take it when I am running on Home Assistant. Please pardon me.
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.3

andig, 2025-08-15:
Do you want to make the required changes to the template and open a PR?

frankb-CZ, 2025-08-16:
@andig I can try. To be honest I am not well educated in coding and programming but I know now pretty good the MODBUS protocol of Wattsonic so I can help with telling you from where you should get which data and how to make the control. But is this PR? I thought that this is more like a bug fixing but of course ready to help as much as I can.

andig, 2025-08-16:
For pv1 and pv2 I guess the template should just add their values using the `calc` plugin?