evcc-io/evcc #23011: Experimental UI glitches

ButtiBBQ, 2025-08-15:
### Describe the bug

Wenn ich PV Anlage, Netzanschluss und Batterie über UI anlege, dann sind mir folgende Dinge aufgefallen:

[image]

1.) Optionale Einstellung "Cache" Description sagt Daten werden von Fahrzeug geladen. Das kann verwirrend sein weil die Daten ja von der PV Anlage geladen werden. => Wort Fahrzeug entfernen, dann ist es für alle "input Devices" OK

2.) wäre aus meiner Sicht "Refresh Intervall" ein treffender Titel für die Einstellung

3.) wenn ich prüfen sage erhalte ich **immer** diesen Fehler:

[image]

danach kann ich aber Speicher und der Datenabruf klappt ohne Fehler im Log. Versteh also nicht warum das Prüfen nicht funktioniert.

4.) nach dem Speicher der Config erhalte ich folgendes:

[image]

Allerdings wird das Plugin in Home Assistant gestoppt aber nicht neu gestartet. Ich habe kein Watchdog an, würde aber erwarten, wenn ich auf eine Schaltfläche "Neustart" klicke das auch ein Neustart und kein Shutdown passiert. (Bin gerade im Nightly Build)

5.) die UI in der Config zeigt manchmal Daten an und manchmal einfach nicht:

[image]
Ich konnte nicht heraus finden, wann Daten angezeigt werden und wann nicht.
Fehler im Log sind nicht zu finden.

Ansonsten sind die Anpassung vom letzten Nighy super mit https und Mutex. Danke @andig 

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Enphase Config über die UI genutzt
```

### Log details

```text
[main ] INFO 2025/08/15 19:42:25 new version available: 0.207.3
[lp-1 ] DEBUG 2025/08/15 19:42:26 charge power: 0W
[lp-1 ] DEBUG 2025/08/15 19:42:26 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/15 19:42:29 grid power: 25W
[site ] DEBUG 2025/08/15 19:42:36 pv 1 power: 233W
[site ] DEBUG 2025/08/15 19:42:43 !! solar production: accumulate set db:4 472.828kWh meter total (was: Accumulated: 0.000kWh updated: 0001-01-01 00:00:00 +0000 UTC)
[site ] DEBUG 2025/08/15 19:42:43 !! solar production: accumulate moved db:4 from 0.000 to 0.000
[site ] DEBUG 2025/08/15 19:42:49 grid currents: [1.95 -1.22 -0.971]A
[site ] DEBUG 2025/08/15 19:42:49 site power: 25W
[lp-1 ] DEBUG 2025/08/15 19:42:49 charge voltages: [234 235 235]V
[lp-1 ] DEBUG 2025/08/15 19:42:49 charge total import: 3.970kWh
[lp-1 ] DEBUG 2025/08/15 19:42:49 charger status: A
[lp-1 ] INFO 2025/08/15 19:42:49 car disconnected
[lp-1 ] DEBUG 2025/08/15 19:42:49 set battery boost: false
[lp-1 ] DEBUG 2025/08/15 19:42:49 set session soc limit: 0
[lp-1 ] DEBUG 2025/08/15 19:42:49 set session energy limit: 0
[site ] DEBUG 2025/08/15 19:43:49 ----
[lp-1 ] DEBUG 2025/08/15 19:43:49 charge power: 0W
[lp-1 ] DEBUG 2025/08/15 19:43:49 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/15 19:43:51 grid power: 17W
[site ] DEBUG 2025/08/15 19:44:01 pv 1 power: 237W
[site ] DEBUG 2025/08/15 19:44:06 !! solar production: accumulate set db:4 472.834kWh meter total (was: Accumulated: 0.000kWh updated: 2025-08-15 19:42:43 +0200 CEST meter: 472.828kWh)
[site ] DEBUG 2025/08/15 19:44:06 !! solar production: accumulate moved db:4 from 0.000 to 0.005
[site ] DEBUG 2025/08/15 19:44:14 grid currents: [1.96 -1.27 -1.02]A
[site ] DEBUG 2025/08/15 19:44:14 site power: 17W
[lp-1 ] DEBUG 2025/08/15 19:44:14 charge voltages: [234 235 236]V
[lp-1 ] DEBUG 2025/08/15 19:44:14 charge total import: 3.970kWh
[lp-1 ] DEBUG 2025/08/15 19:44:14 charger status: A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

``` evcc 0.207.2 (6a29b415) ```

thedoctor586, 2025-08-15:
[image]

Kann das icon für die wallbox /ladepunkt hinzugefügt werden im dashboard so wie es in der konfigurationsübersicht schon vorhanden ist?

andig, 2025-08-16:
Cache siehe 38d5441ed642d1db086b644a8a92a8f684de8b19

andig, 2025-08-16:
> Danach kann ich aber Speicher und der Datenabruf klappt ohne Fehler im Log. Versteh also nicht warum das Prüfen nicht funktioniert.

Mangels Info nicht bewertbar.

> Allerdings wird beim klicken auf den Button das Plugin in Home Assistant nur gestoppt aber nicht neu gestartet. 

Bitte in den HA Discussions hier posten- wir haben keinen Support für HA /cc @thecem 

> 5.) Konsistenz die UI in der Config zeigt manchmal Daten an und manchmal einfach nicht:

/cc @naltatis

naltatis, 2025-08-16:
Das Verhalten, dass keine Werte da sind. Aber ohne weitere Infos können wir hier nur Raten. Meine Vermutung ist auch, dass das hier (siehe Timeout) etwas Device-spezifisches ist und weniger ein allgemeines Config UI Problem. @ButtiBBQ sonst mach gerne dafür ein separates Issue mit den nötigen Details auf.

closed 2025-08-16 by andig