evcc-io/evcc #23017: Probleme ab 0.207.0

Peach1975, 2025-08-16:
### Describe the bug

Hallo

Meine Wallbox steht etwas weg vom Haus. Darum, damit Zähler korrekt zählt diese Anpassung im Config:

meters:
- name: grid_pur
  type: template
  template: goodwe-wifi
  usage: grid
  uri: 10.0.0.3

- name: grid
  type: custom
  power:
    source: http
    uri: http://10.0.0.92:7070/api/state
    jq: .result.ext[0].power + .result.loadpoints[0].chargePower

site:
  meters:
    grid: grid
    ext: grid_pur

Seit 0.207.0 geht aber EVCC nicht mehr - sprich startet nicht.

Protokoll:

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/08/16 07:56:12 evcc 0.207.3
[main  ] INFO 2025/08/16 07:56:12 using config file: /config/evcc.yaml
[db    ] INFO 2025/08/16 07:56:12 using sqlite database: /data/evcc.db
[main  ] INFO 2025/08/16 07:56:12 UI listening at :7070
[site  ] INFO 2025/08/16 07:56:16 site config:
[site  ] INFO 2025/08/16 07:56:16   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/08/16 07:56:16     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/08/16 07:56:16     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/08/16 07:56:16     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/08/16 07:56:16   vehicles:
[site  ] INFO 2025/08/16 07:56:16     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/08/16 07:56:16   tariffs:
[site  ] INFO 2025/08/16 07:56:16     grid:      pricestatic
[site  ] INFO 2025/08/16 07:56:16     feed-in:   pricestatic
[site  ] INFO 2025/08/16 07:56:16     co2:       ✗
[site  ] INFO 2025/08/16 07:56:16     solar:     ✓
[lp-1  ] INFO 2025/08/16 07:56:16 loadpoint 1:
[lp-1  ] INFO 2025/08/16 07:56:16   mode:        pv
[lp-1  ] INFO 2025/08/16 07:56:16   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/08/16 07:56:16   meters:      charge ✓
[lp-1  ] INFO 2025/08/16 07:56:16     charge:    power ✓ energy ✓ currents ✓
[site  ] WARN 2025/08/16 07:56:16 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] ERROR 2025/08/16 07:56:25 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2025/08/16 07:56:45 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2025/08/16 07:56:54 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2025/08/16 07:57:03 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax

Hat jemand eine Idee?

Backup nach 0.206.0 - alles o.k.

Danke & Grüsse
Peach

### Steps to reproduce

2.
3.
...

### Configuration details

```yaml
1.
meters:
- name: grid_pur
  type: template
  template: goodwe-wifi
  usage: grid
  uri: 10.0.0.3

- name: grid
  type: custom
  power:
    source: http
    uri: http://10.0.0.92:7070/api/state
    jq: .result.ext[0].power + .result.loadpoints[0].chargePower

site:
  meters:
    grid: grid
    ext: grid_pur
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/08/16 07:56:12 evcc 0.207.3
[main  ] INFO 2025/08/16 07:56:12 using config file: /config/evcc.yaml
[db    ] INFO 2025/08/16 07:56:12 using sqlite database: /data/evcc.db
[main  ] INFO 2025/08/16 07:56:12 UI listening at :7070
[site  ] INFO 2025/08/16 07:56:16 site config:
[site  ] INFO 2025/08/16 07:56:16   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/08/16 07:56:16     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/08/16 07:56:16     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/08/16 07:56:16     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2025/08/16 07:56:16   vehicles:
[site  ] INFO 2025/08/16 07:56:16     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/08/16 07:56:16   tariffs:
[site  ] INFO 2025/08/16 07:56:16     grid:      pricestatic
[site  ] INFO 2025/08/16 07:56:16     feed-in:   pricestatic
[site  ] INFO 2025/08/16 07:56:16     co2:       ✗
[site  ] INFO 2025/08/16 07:56:16     solar:     ✓
[lp-1  ] INFO 2025/08/16 07:56:16 loadpoint 1:
[lp-1  ] INFO 2025/08/16 07:56:16   mode:        pv
[lp-1  ] INFO 2025/08/16 07:56:16   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/08/16 07:56:16   meters:      charge ✓
[lp-1  ] INFO 2025/08/16 07:56:16     charge:    power ✓ energy ✓ currents ✓
[site  ] WARN 2025/08/16 07:56:16 interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval
[site  ] ERROR 2025/08/16 07:56:25 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2025/08/16 07:56:45 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2025/08/16 07:56:54 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax
[site  ] ERROR 2025/08/16 07:57:03 grid power: strconv.ParseFloat: parsing "<nil>": invalid syntax
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.3

VolkerK62, 2025-08-16:
/api/state wurde geändert.  https://github.com/evcc-io/evcc/releases/tag/0.207.0
Lösche ` .result ` dann geht es wieder

Peach1975, 2025-08-16:
danke für die Hilfe

VolkerK62, 2025-08-16:
zumachen nicht vergessen

Peach1975, 2025-08-16:
Hmm

.result raus.. dafür jetzt fragezeichen und Stom vom Netz..
Wie könnte man das beheben? Hausverbauch ist weg..

[image]