evcc-io/evcc #23042: Improve experimental optimizer

andig, 2025-08-17:
With XX merged, we're using this as tracking issue for more optimizer capabilities

### Model capabilities:

- [x] EV min charge power (if charging)
- [x] EV max charge power
- [x] EV fixed charge demand step (model Min+PV and Fast charging) https://github.com/andig/evopt/pull/4
- [ ] EV charge goal (model charging plan goals)
- [x] Battery grid charging (https://github.com/andig/evopt/pull/3)
- [ ] PV AC limitation
- [ ] Grid Feed-in limitation/ peak shaving
- [ ] Load management restrictions
- [ ] End of forecast financial value
- [ ] Forced grid charging

### Blocked (missing parameters):

- [ ] Battery max charge/discharge power (not available in evcc)
- [ ] Battery charge goal

**NOTE**: at this time, the optimizer is purely information-only ("what would happen if we actually used this"). It is not used to make actual decisions.

djfanatix, 2025-08-28:
Another wishlist item, longterm:

Get from vehicle api: destination, SoC when arriving at destination and ETA destination.
If destination is the location of EVCC (='home'), use the SoC in the optimizer, to prepare for the EV to arrive. (maybe already prepare the planner)

rafdev18, 2025-09-13:
Excellent project !

Optimizer should focus on the overall energy management for the battery in the house. In this model the car is just another dependency to take into account when planing for the house battery charge. We have plenty of functionality to plan and optimise the car charging already in evcc but we do lack the energy management for the house battery.

The battery in the house is the often the bottleneck because most installations do have a battery capacity of 10 kWh which normally is exceeded by the consumption in the winter time if you do operate a heat pump. Your daily consumption on a cold day can vary from 10 - 25 kWh depending on the efficiency.

Heat pump will be added into the daily house forecast and then the need is to charge the battery from grid and solar before the grid price peaks during the day. This can easily result in multiple (2-3) charging cycles during the day in order to drive needed savings.

I would be good to calculate the average kWh price of the current battery load to understand at what price level it is cheaper to depend on the battery and when we have a favourable price level for the battery grid reloading. I saw that you already have an efficiency of 90% planed for the battery consumed energy which reflects the loss of the system.

From an algorithm side in the winter it is less about achieving a certain SOC on the house battery but more about tapping into the lowest grid prices and leverage the stored energy to offset peak pricing. You need to stretch the battery load over the peak price hours efficiently.

Kiechlus, 2025-09-15:
In summer, "PV AC limitation" would be a very useful feature for new systems. We have to reduce the feed-in power to 60%. This means the battery should only be charging in the noon hours with the surplus that cannot be fed in.

SMA has such a feature and I think other inverters as well. But in the case of SMA, if "prognosebasiertes Laden" is enabled, forced grid charging in `evcc` is always interupted, then restarting and so on. So it's better if there is just one brain.

This is also discussed here: https://github.com/evcc-io/evcc/discussions/18510

Kiechlus, 2025-09-15:
Thanks. In this case both of them would be very useful in my case.

dsgrafiniert, 2025-10-06:
Hi @andig, thanks for your outstanding work regarding the Optimizer. I would like to give it a try and hand over control to the optimizer. When can we expect that this possibility is given and Optimizer is not "information only"?

From my point of view, the results and plans, the optimizer creates are better than any other optimization method, I have at hand.

andig, 2025-11-12:
Pls open new issue

redzioch, 2025-11-12:
> Pls open new issue

It's OK, I understand why evcc behaves like that. My goal was to encourage you to give evopt a try.