evcc-io/evcc #23044: UI: Anzahl der Phasen wird nach Dienst Neustart im Ladepunkt-Konfiguration nicht angezeigt

StefanR71, 2025-08-17:
### Describe the bug

Neuen Ladepunkt erstellt und so eingestellt:

[image]

Wenn man dann den Dialog wieder öffnet, ist es immer noch so ausgewählt, startet man dann den Dienst neu, sieht es so aus:

[image]

Ob wohl in der DB steht:

meter: phasesConfigured:3

Ich denke ein Anzeigefehler, da immer wieder reproduzierbar.

### Steps to reproduce

1. Dienst ist frisch gestartet, zuvor einen Ladepunkt neu angelegt.
2. Konfiguration in der UI öffnen, einen Ladepunkt bearbeiten.
3. 1 oder 3 Phasen ist nicht ausgewählt.
4. z.B. 3 Phasen anwählen und speichern.
5. Geht man erneu in den Konfigurations-Dialog ist ist die Anzahl der Phasen noch ausgewählt.
6. Dienst Neustarten
7. Konfiguration in der UI öffnen, einen Ladepunkt bearbeiten.
8. Das zuvor ausgewählte und gespeicherte ist nicht mehr ausgewählt.

### Configuration details

```yaml
root@srv46:/# evcc config
[main  ] INFO 2025/08/17 12:14:25 using config file: /etc/evcc.yaml
[eebus ] ERROR 2025/08/17 12:14:25 websocket server error: listen tcp :4712: bind: address already in use
charger
---
db:6 {Type:template Title: Icon: Product:Cupra Charger Connect 2} map[ip:192.168.77.71 ski:***** template:elli-2]
db:8 {Type:template Title: Icon: Product:Cupra Charger Connect 2} map[ip:192.168.77.72 ski:***** template:elli-2]

meter
---
db:3 {Type:template Title:sonnenBatterie Icon: Product:Sonnen sonnenBatterie} map[capacity:40 defaultmode:self-consumption host:192.168.77.100 maxchargepower:8000 port:80 template:sonnenbatterie token:***** usage:battery]
db:4 {Type:template Title:PV Anlage West Icon: Product:Kostal Piko} map[host:192.168.77.54 template:kostal-piko-pv usage:pv]
db:5 {Type:template Title:PV Anlage Süd Icon: Product:Kostal Piko} map[host:192.168.77.53 template:kostal-piko-pv usage:pv]

loadpoint
---
db:7 {Type: Title: Icon: Product:} map[charger:db:6 circuit:wallboxen defaultMode:pv meter: mode:pv phasesConfigured:3 soc:map[estimate:true poll:map[interval:6e+11 mode:always]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Garage links vehicle:ev1]
db:9 {Type: Title: Icon: Product:} map[charger:db:8 circuit:wallboxen defaultMode:pv meter: mode:pv phasesConfigured:3 soc:map[estimate:false poll:map[interval:6e+10 mode:always]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Garage rechts vehicle:ev2]

root@srv46:/#
root@srv46:/# cat /etc/evcc.yaml
# open evcc at http://evcc.xxxx.local
network:
  schema: http
  host: evcc.xxxx.local
  port: 80

site:
  title: HOME
  meters:
    grid: grid1
    #pv:
    #  - kostal15
    #  - kostal17
    #battery: battery1

log: error
levels:
  site: error
  lp-1: error
  lp-2: error
  cache: error
  cupra: debug
  mqtt: error
  eebus: warn

plant: *****
sponsortoken: *****
interval: 15s

database:
  type: sqlite
  dsn: /var/lib/evcc/evcc.db

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2455
  feedin:
    type: fixed
    price: 0.10

eebus:
  shipid: EVCC-Home
  interfaces:
    - ens33
  certificate:
    public: |
      -----BEGIN CERTIFICATE-----
      -----END CERTIFICATE-----
    private: |
      -----BEGIN EC PRIVATE KEY-----
      -----END EC PRIVATE KEY-----

mqtt:
  broker: localhost:1883
  topic: ""

circuits:
  - name: main
    title: Hauptsicherung
    maxCurrent: 35
    meter: grid1
  - name: wallboxen
    title: Wallboxen
    maxCurrent: 32
    parent: main

meters:
- name: grid1
  type: template
  template: cg-em24_e1
  usage: grid
  modbus: tcpip
  id: 1
  host: 192.168.77.57
  port: 502

vehicles:
- name: ev1
  type: template
  template: cupra
  user: *****
  password: *****
  vin: *****
  title: Cupra Tavascan
  capacity: 77
  icon: car
  phases: 3
  mode: PV
  minCurrent: 6
  maxCurrent: 16
  cache: 5m
- name: ev2
  type: custom
  title: Seat Mii
  icon: car
  capacity: 32
  phases: 2
  soc:
    source: mqtt
    topic: mii/soc
```

### Log details

```text
Habe keins, da vermutlich nur Anzeigeproblem, sollte dennoch was benötigt werden, bitte schreiben.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.3

StefanR71, 2025-08-17:
Was ist das? Die Mail habe ich gerade bekommen: 

[evcc-io/evcc] Claude Issue Triage workflow run 
Claude Issue Triage: All jobs have failed

Und der Grund soll sein: Credit balance is too low

naltatis, 2025-08-19:
Can reproduce. I'll check.

closed 2025-08-22 by andig