evcc-io/evcc #23113: Tariff panic

andig, 2025-08-19:
Tried adding the `interval` parameter to the grid tariff which does not seem to be supported though. evcc even crashes when doing so:
```
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x1d05ce0]

goroutine 1 [running]:
github.com/evcc-io/evcc/tariff.(*Tariffs).Get(0x4000764880?, 0x2462780?)
        github.com/evcc-io/evcc/tariff/tariffs.go:58 +0x60
github.com/evcc-io/evcc/core.(*Site).GetTariff(0x4000cbf8f0?, 0x5?)
        github.com/evcc-io/evcc/core/site_api.go:289 +0xa8
github.com/evcc-io/evcc/core.(*Site).Boot(0x4000adf208, 0x40000db4d0, {0x4000cbea40, 0x2, 0x4000f4f078?}, 0x2020fec?)
        github.com/evcc-io/evcc/core/site.go:167 +0x354
github.com/evcc-io/evcc/cmd.configureSite(0x0?, {0x4000cbea40, 0x2, 0x2}, 0x0)
        github.com/evcc-io/evcc/cmd/setup.go:1051 +0x50
github.com/evcc-io/evcc/cmd.configureSiteAndLoadpoints(0x5f2d080)
        github.com/evcc-io/evcc/cmd/setup.go:993 +0x3ac
github.com/evcc-io/evcc/cmd.runRoot(0x5f2b140, {0x29a457b?, 0x7?, 0x2997be4?})
        github.com/evcc-io/evcc/cmd/root.go:206 +0x928
github.com/spf13/cobra.(*Command).execute(0x5f2b140, {0x400004c090, 0x0, 0x0})
        github.com/spf13/cobra@v1.9.1/command.go:1019 +0x7bc
github.com/spf13/cobra.(*Command).ExecuteC(0x5f2b140)
        github.com/spf13/cobra@v1.9.1/command.go:1148 +0x350
github.com/spf13/cobra.(*Command).Execute(...)
        github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
        github.com/evcc-io/evcc/cmd/root.go:123 +0x24
main.main()
        github.com/evcc-io/evcc/main.go:44 +0x34
```

_Originally posted by @CiNcH83 in https://github.com/evcc-io/evcc/discussions/23069#discussioncomment-14138818_

CiNcH83, 2025-08-19:
Crashing config:
```
tariffs:
  grid:
    type: template
    template: energy-charts-api # Fraunhofer ISE
    bzn: DE-LU
    interval: 1h
```

andig, 2025-08-19:
Closed in 5bb51c7481e685e9cdb2be59e2ce37cfe7b221d0