evcc-io/evcc #23120: Promote sponsoring more

Sonnenwerk, 2025-08-20:
**Is your feature request related to a problem? Please describe.**
I would like to see the sponsoring promoted in the UI more than now.

**Describe the solution you'd like**
Show status of token in the main UI --> no token/trial/sponsor. 
Show the name of the sponsor with a text and nice "confetti"-icon "Thank you xyz for supporting this project!"

**Additional context**
Add on: Maybe extend this functionality with a info "you are supporting xy month" / "one time supporter"

hybe2022, 2025-08-20:
Bitte keine solche unnötige Information auf dem Hauptscreen. Ich weiß selbst, dass ich Sponsor bin.
OK wäre allerdings ein Banner zu bringen, wenn jemand KEIN Sponsor ist...

Sonnenwerk, 2025-08-21:
:-)