evcc-io/evcc #23133: EM2GO Home - multiple Modbus/TCP connections crash modbus stack on wallbox

hifiberry, 2025-08-20:
### Describe the bug

I had some trouble getting the EM2GO Home configured at all. After lots of debugging I found out that evcc seems to open multiple parallel TCP connections to the modbos port 502 on the wallbox. The software on the wallbox can't handle this and stops responding at any modbus requests.

Workaround is a 2nd EVCC instance with only Modbus proxy activated and configuring the "main" EVCC instance to not use the wallbox directly, but via the Modbus proxy on the 2nd instance.

### Steps to reproduce

1. Add EM2Go Wallbox via the configuration API
2. Validate -> works
3. Save -> stops working
...

### Configuration details

```yaml
/app # evcc config
[main  ] INFO 2025/08/20 21:24:47 using config file: /app/evcc.yaml
[db    ] INFO 2025/08/20 21:24:47 using sqlite database: /root/.evcc/evcc.db
```

### Log details

```text
I don't have any logs when it doesn't work. 

[em2go ] TRACE 2025/08/20 21:26:23 modbus: send 00 01 00 00 00 09 ff 10 00 5b 00 01 02 00 3d
[em2go ] TRACE 2025/08/20 21:26:23 modbus: recv 00 01 00 00 00 06 ff 10 00 5b 00 01
[em2go ] TRACE 2025/08/20 21:26:23 modbus: send 00 02 00 00 00 06 ff 03 00 5b 00 01
[em2go ] TRACE 2025/08/20 21:26:23 modbus: recv 00 02 00 00 00 05 ff 03 02 00 3d
[em2go ] TRACE 2025/08/20 21:26:23 modbus: send 00 03 00 00 00 06 ff 03 00 c8 00 01
[em2go ] TRACE 2025/08/20 21:26:24 modbus: recv 00 03 00 00 00 05 ff 03 02 00 01
[em2go ] TRACE 2025/08/20 21:26:25 modbus: send 00 04 00 00 00 06 ff 03 00 1c 00 02
[em2go ] TRACE 2025/08/20 21:26:25 modbus: recv 00 04 00 00 00 07 ff 03 04 00 00 00 02
[em2go ] TRACE 2025/08/20 21:26:25 modbus: send 00 05 00 00 00 06 ff 03 00 5f 00 01
[em2go ] TRACE 2025/08/20 21:26:25 modbus: recv 00 05 00 00 00 05 ff 03 02 00 01
[em2go ] TRACE 2025/08/20 21:26:25 modbus: send 00 06 00 00 00 09 ff 10 00 5b 00 01 02 00 3c
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.3

killamilla0815, 2025-09-28:
Same for me, how to resolve this ?

gschelter, 2026-02-23:
@killamilla0815 @aveseli 
I reponed the issue here: https://github.com/evcc-io/evcc/issues/27625
Please add logs/info if available