evcc-io/evcc #23137: UI-Config: keine Fahrzeugauswahl bei `integrateddevice: true`

VolkerK62, 2025-08-21:
In der Ladepunkt-Konfiguration bei denen im Charger "integriertes Gerät = ja" konfiguriert ist, wird seit kurzem die Fahrzeugauswahl ausgeblendet.
Wenn man die Konfiguration aufruft, dann dauert es einen Moment, bis alles (Ladepunkte, Fahrzeuge) aktualisiert ist.
Während der kurzem Zeit wird der Fahrzeugteil im Ladepunkt angezeigt, wenn die "Aktualisierung" abgeschlossen ist, ist es verschwunden (ausgeblendet?).

Bug or feature?

Die Funktionlität, also die Anzeige von Soc bzw. Temparatur mit Hilfe des Fahrzeugs, ist nicht betroffen.

naltatis, 2025-08-22:
> Bug or feature?

Feature. Felder die nicht relevant sind (bspw. Fahrzeugauswahl bei Wärmepumpe oder Phasenumschaltung bei einphasigen Geräten) werden ausgeblendet.

Ich weiß, dass du einen Use-case hattest wo du trotz Integrated Device noch ein Fahrzeug zuordnest. Ich bekomm den aber nicht mehr zusammen. Was war hier der Grund? Charger ist fertiges Template (non-custom) und die Zusatzinfo Temperatur kommt über ein Custom Fahrzeug?

VolkerK62, 2025-08-22:
>Charger ist fertiges Template (non-custom) und die Zusatzinfo Temperatur kommt über ein Custom Fahrzeug?

Genau das.
Aber wenn das so sein soll, auch kein Problem.