evcc-io/evcc #23147: Since Kia API update: vehicle soc: timeout

Andes1984, 2025-08-21:
### Describe the bug

Since the KIA API changed and evcc got updated, the SoC of my Kia Niro EV is not getting reported to the wallbox. In the evcc log, I have this message:

```
[lp-1 ] ERROR 2025/08/21 16:53:45 vehicle soc: timeout
```
Charging works fine but the SOC is not getting shown.

### Steps to reproduce

1. Connect car
2. Look at log
3. See the error
...

### Configuration details

```yaml
vehicles:
- type: template
  template: kia
  title: Kia Niro EV
  user: xxxxxxxxxxxxxxxxxxxxxxx
  password: xxxxxxxxxxxxxxx
  vin: KNACRxxxxxxxxxxxx
  capacity: 64
  language: en
  name: niroev
```

### Log details

```text
[site ] DEBUG 2025/08/21 16:55:15 battery 1 soc: 97%
[site ] DEBUG 2025/08/21 16:55:15 site power: -724W
[lp-1 ] DEBUG 2025/08/21 16:55:15 charger status: C
[lp-1 ] ERROR 2025/08/21 16:55:15 vehicle soc: timeout
[lp-1 ] DEBUG 2025/08/21 16:55:15 pv charge current: 8.22A = 7.17A + 1.05A (-724W @ 3p)
[lp-1 ] DEBUG 2025/08/21 16:55:15 set charge current limit: 8.22A
[site ] DEBUG 2025/08/21 16:55:15 solar forecast: accumulated 0.012Wh from 2025-08-21 16:55:05 +0200 CEST to 2025-08-21 16:55:15 +0200 CEST
[site ] DEBUG 2025/08/21 16:55:15 solar forecast: produced 1985.132
[site ] DEBUG 2025/08/21 16:55:15 solar forecast: accumulated 1779.505kWh, produced 1985.132kWh, scale 1.116
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.207.3

CaptainChaos09, 2025-08-21:
Habe das selbe Problem mit der selben Fehlermeldung. Ich hatte das auch mit der HomeAssistant Integration vor ein paar Wochen, KIA hat wohl was an der Login Methode geändert.

fklapro, 2025-08-22:
yes, same to me. 
unfortunately did this read too late... so after update and restart evcc - my kia ev3 couldn't connect ...
hope there will be a chance kia uses a friendly captcha API?!

Robert28021962, 2025-08-22:
Thank you for your response. I already understood that this is the result of an other problem.  
The question now is of  Kia or EVCC kan work around this.  
lets see whats happens ....

AndiH, 2025-08-24:
Here's the solution they came up with: https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/pull/871/files (plus, consenting to a banner when logging in to the German Kia website)

wjasperse, 2025-08-24:
> Where do you consent to the Banner? Can you please share a Link or explain
> where to Login?
> 
> Andreas Herten ***@***.***> schrieb am So., 24. Aug. 2025,
> 14:10:
> […](#)

Log in to https://www.kia.com/de and grant access to the account

0x3d13f, 2025-08-24:
kia_uvo obviously has a solution with  v2.43.9: https://github.com/Hyundai-Kia-Connect/kia_uvo/releases/tag/v2.43.9
See also https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/commit/4cb75697485f846063b5a658118c651b40f0ecfe for what needs to be changed.

TheBrankman, 2025-08-25:
Is someone trying to solve this problem?

plieven, 2025-08-25:
I wonder where they got the value for the "state" field from. It does not look very stable. Looking at the discussion I found no clue where this is coming from.

https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/commit/4cb75697485f846063b5a658118c651b40f0ecfe#diff-40fbbfd31a6bd44b4df09375bde429a6c71446dcf7ff8a08f8e55a1e6b560763R1117

musikus24, 2025-08-26:
Hallo, habe immer noch das Problem, dass das template nicht erzeugt werde kann. Fehler auf dem Screenshot 

[image]

plieven, 2025-08-26:
Der aktuell favorisierte Weg, scheint zu sein, das Refresh Token irgendwie extern zu besorgen und dann in die Config einfach änderbar einzutragen.

Unabhängig davon, selbst wenn es noch einen Weg mit Username+Passwort geben sollte, dann sollte der Refresh Token so in EVCC gespeichert werden, dass er einen Neustart übersteht.

Bis dahin, bei wem es läuft, nicht neustarten!

Getiem, 2025-08-26:
Discussion about that captcha and/or refresh token  https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/issues/855

Getiem, 2025-08-26:
Maybe reopen issue? Since the release of today did not fix it (at least for some posters in this issue).  @andig

plieven, 2025-08-26:
@Getiem it fixed it for me because I seem to have started evcc nightly before they inserted the capatcha. I guess the way with generating a capatcha token and suppling user+pass+recapatcha token in the evcc config seems the best solution for now.

This would require evcc to persistenly store the refresh token and use the repatcha token only once (delete it from the config once used) and in case a relogin is necessary a new repatcha token has to be generated and supplied. In case there is no (new) repatcha token in the config evcc should not try to relogin to avoid being blocked.

In the original thread https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/issues/855 they discussed how the BMW Api implementation solved this issue: https://bimmer-connected.readthedocs.io/en/stable/captcha.html#getting-a-captcha-token

Getiem, 2025-08-26:
> I would suggest that you- as customers of Kia- approach them and ask how you can get programmatic access to _your_ data. There is very little we can do, but there is a lot of market power you have. Take this feature into consideration when purchasing your next car. Really the only vehicle manufacturer that has any usable approach is- whatever you may think of them otherwise- is Tesla.

Good point.   I asked the support of kia to provide an answer.

https://cloud.eu.connect.kia.com/DesktopWebform_KiaConnectDataProtection?l=&c=de

`The account login on kia.com/de through https://idpconnect-eu.kia.com is changed recently. 
It works with the actual apps KIA from Appstore, but we also need it for apps like EVCC and HomeAssistant to read the current state of the vehicle (like the current state of charge of the battery, range, connected to charger).  The EVCC and HomeAssistant are used to control the relation between a charger, solarpanels and the Kia vehicle.

Can you provide a working solution to programmatically get the required vehicle status information. By preference providing a working method to programmatically login to the KIA API  (via IDP).   A good working practice would be to generate an individual API token on a personal account (for example Samsung is doing that for their SmartThings platform). 

Currently the Recaptcha and cookie during login are blocking login from the apps like EVCC and HomeAssistant. Recaptcha is used for human interface and is unusable for programmatically accessing the api.   

An issue is reported at EVCC https://github.com/evcc-io/evcc/issues/23147,  the library to connect to the API   https://github.com/Hyundai-Kia-Connect/hyundai_kia_connect_api/issues/855 that is also used in HomeAssistant.

Please provide an answer from a software engineer at KIA. The login issue is affecting a lot of users of the EVCC and HomeAssistant in Europe right now. `

polskafan, 2025-08-26:
There is developers.kia.com for Korean devs (to register you need a Korean phone number or an iPIN, whatever that is). They have an Open API to access vehicle data.

Unfortunately this is from their FAQ:
> Is the Open API provided by KIA Developers limited to Korea?
> Yes. Since the data currently being collected is from connected cars in Korea, you can only access domestic vehicle data using this API.
> - https://developers.kia.com/web/v1/kia/faqs

I guess it might be worth it to contact developers@kia.com. I'll try shooting them an email and see what their response is.

In the meantime there [1] is now a solution that extracts the refresh_token via Python script and the browser. Not ideal, but might be worth exploring.

[1] = https://github.com/Hyundai-Kia-Connect/kia_uvo/issues/1277#issuecomment-3223019018

dMopp, 2025-08-27:
Fazit: FU KIA, einmal und nie wieder. Was ein Drecksladen das geworden ist.

plieven, 2025-08-27:
kann man den log level zur laufzeit hochsetzen?

pencoe, 2025-08-28:
ich habe jetzt mal an info@kia-connect.eu geschrieben:

> Sehr geehrte Damen und Herren,
> 
> mit wachsendem Unmut stelle ich fest, dass Kia nicht etwa einen Weg zur Öffnung für Smarthome- oder Energiemanagement-Systeme beschreitet (was man aktiv bewerben und vermarkten könnte: "erster KFZ-Hersteller mit works with Homeassistant" z.B.) sondern ganz im Gegenteil in letzter Zeit zunehmend Wege beschreitet, um alle Zugriffe auf die Daten meines(!) Fahrzeugs durch andere Anwendungen als Kia Connect unmöglich zu machen.
> 
> Dazu kommt der offensichtliche Versuch, mich auch "after Sales" noch zu monetarisieren: mit Werbung in der App (die es in der alten App nicht gab), mit ständiger Aufforderung, doch unbedingt ein völlig überteuertes OTA-Abo abzuschließen, mit dem Versuch, alle meine(!) Daten exklusiv zu vermarkten.
> 
> Dazu kommt der Bruch einer durchaus mit kaufentscheidenden Aussage (2 Kartenupdates pro Jahr, und selbst das ist eigentlich erschreckend wenig).
> 
> Sollte sich diese Entwicklung so stabilisieren ist dies mein erster und letzter Kia.
>

TheBrankman, 2025-08-28:
> I was a few days not @home . As far as I understand there is still no working solution. Is that correct ?

Here at this EVCC topic it is a bit more quite, but maybe it is an idea to cooperate with the guys in the hyundai/kia integration topic.
There they are quite busy with this with several people.

They also have contnacted Kia about this, with an answer we were not hoping for...

plieven, 2025-08-28:
Ich habe mir die Frage gestellt, ob wir hier mit Art. 20 DSGVO weiter kommen? Immerhin sammeln sie auch Standortdaten des Fahrzeugs. Da das ganze mit Nutzerprofilen etc. verknüpft ist, gehe ich hier stark davon aus, dass die DGSVO einschlägig ist. Man könnte KIA also auffordern, einen Zugang bereitzustellen, indem die Personenbezogenen Daten maschinell abrufbar bereitgestellt werden.

Das sollte man aber vielleicht im Zusammenschluß von allen Betroffenen mit jemand machen, der sich damit auskennt.

markusjnagel, 2025-08-28:
> Geht ohne die Schnittstelle dann überhaupt noch gescheit PV Überschuss laden?

Ja klar - im Prinzip alles, was mit nem "Gastfahrzeug" auch geht. Überschussladen heisst ja erstmal das alles was Du nicht verbrauchst (verbrauchen möchtest mit Hausbatterie Batterie, etc) in die Wallbox geht.

Was ohne API nicht (mehr) geht bei uns Kia Fahrern:

1) Automatische Erkennung des Autos (ohne Konfiguration der Wallbox zwecks Erkennung mit RFID Karte)
2) Damit automatisches Setzen des Lademodus sobald Du einsteckst.
--> die beiden Punkte kannste aber manuell machen, indem Du in evcc das Fahrzeug selbst auswählst. Ist halt leider ein Schritt mehr und nicht ganz so bequem.
2.1) Dazu gehört dann auch, dass Du evcc nicht sagen kannst - nur bis x% laden. Das musst Du dann per Kia App einstellen. Oder im Auto selbst.

3) Was kritischer ist: Ladeplanung - ich möchte gerne morgen früh voll aufgeladen losfahren. Ohne SoC kann er nicht bestimmen, wann er mit dem Laden anfangen soll und bis wieviel % SoC.

naja, und was Du sonst noch so alles mit den Daten anstellen möchtest - falls Du HA nutzt, die Farbe der Garageninnenbeleuchtung anhand des SoC regeln, Dashboards mit gefahrenen Kilometeren, Verbrauch, etc.... aber was evcc UseCases angeht, sind das die drei die mir einfallen. Also, PV Überschussladen geht.

dennisN75, 2025-08-28:
Ich hoffe da wird es noch eine Lösung geben. Es war alles so schön eingerichtet

Starwolf74, 2025-08-29:
Ich warte leider noch auf Antwort von Kia (jetzt 2 Tage) - ich habe speziell mit Hinweis auf den EU Data Act (final gültig ab 15.9.2025) angefragt. Der schreibt verpflichten vor, dass gesammelte daten von "connected devices" elektronisch verarbeitbar zugängig gemacht werden müssen. Ich bin gespannt.

Mike-F362, 2025-09-01:
> Some serious workarounds in progress, not finished, but hopefull
> 
> [Hyundai-Kia-Connect/kia_uvo#1277](https://github.com/Hyundai-Kia-Connect/kia_uvo/issues/1277)

In fact, [the temporary fix](https://github.com/Hyundai-Kia-Connect/kia_uvo/issues/1277#issuecomment-3223019018) works in conjunction with the [EVCC Homeassistant Vehicle](https://github.com/evcc-io/evcc/discussions/22676).

pijay01, 2025-09-01:
Habe ebenfalls ein Support-Case bei Kia geöffnet. Nicht nur ist die neue App der letzte Dreck, jetzt auch noch aktiv gute Implementierungen verhindern 👎 . Leasing läuft nächstes Jahr aus, dann wird es kein KIA / Hyundai mehr.

pencoe, 2025-09-02:
> In fact, [the temporary fix](https://github.com/Hyundai-Kia-Connect/kia_uvo/issues/1277#issuecomment-3223019018) works in conjunction with the [EVCC Homeassistant [...]

@Mike-F362 could you please be so kind to post your Config in HA as an example? Thank you.

chriz77, 2025-09-03:
Ich habe KIA auch geschrieben und meinen Frust zum Ausdruck gebracht. Nach meiner Erfahrung ist der KIA Kundensupport aber leider unterirdisch. Es werden nur ein paar vordefinierte Textskonserven als Antwort zurückkommen. Die neue App ist auch eine Vollkatastrophe! Sie ist schlechter als die sehr schlechte App zuvor. Das muss man erstmal hinbekommen. Wieso bitte ist die im Apple Appstore mit 4 Sternen bewertet? Bots? Fast täglich muss ich mich neu einloggen. Was machen die da in Korea? Das war auf jeden Fall mein letzer KIA/Hundai. An einer Lösung wäre mir auch sehr gelegen.

Derektar, 2025-09-05:
> > [@Snited](https://github.com/Snited) wow... ich habe noch keine Antwort, aber die wird sicherlich ganz ähnlich aussehen.
> > "Aus Sicherheitsgründen" _lach_ Das ist Security by obscurity, das hat noch nie funktioniert. Aber ok, nie wieder Kia...
> 
> Vielleicht an dieser Stelle eine kleine Anekdote, die hier niemanden verteidigen sondern nur aufklären soll: Wir kämpfen bei unserem Soul EV seit gut 2 Jahren mit sehr sporadischem vollständigen Leersaugen der 12V-Batterie durch simples Herumstehen. Oft steht das Auto dabei an eine Ladesäule (steht er, dann lädt er..), wobei er nachts nicht lädt, und am nächsten Tag ist alles tot. Kann man googlen: ist ein bekanntes Problem für das bis dato keine wirkliche Erklärung gefunden wurde. Auch wir haben unzählige Werkstattbesuche hinter uns, die wir letztendlich bezahlen mussten - Diagnose: Auto heile, irgendein Fremdverbraucher entlädt das Auto im Stehen. (EDIT: Mal ehrlich: Jedes E-Auto, dass die 12V-Batterie überhaupt leerstehen kann ist ein einziger Konstruktionsfehler)
> 
> Trotzdem hatte ich wirklich einen bemühten Kontakt bei meinem Autohaus, der sich da auch mit KIA drüber gefetzt hat. Letztendlich will KIA herausgefunden haben, dass zu viele "Anfragen von Drittanbietern" bei meinem Auto für dieses Problem verantwortlich sein könnten. Kann natürlich durchaus sein - und seitdem ich die Anfragen durch EVCC deutlich eingedämmt habe, tritt das Problem auch deutlich seltener bis gar nicht mehr auf.
> 
> Insofern kann ich KIA in dieser Hinsicht auch ein wenig verstehen. ABER was wäre sicherlich viel besser?! Klar: Eine offizielle API zu programmieren, die eine bestimmte Anzahl Anfragen zulässt, die das Auto auch aushält, und fertig is. Aber dafür müsste man ja gute Devs haben - und jeder der einen KIA hat weiß dass das nicht der Fall ist.

Ich hatte dieses sporadische Leerziehen der 12V auch vor Einzug in unser Haus schon bei meinem E-Niro recht häufig und in der damaligen Mietwohnung hatte ich weder Homeassistant noch EVCC - in drei Jahren Verwendung beider Tools ist das Problem auch nicht häufiger aufgetreten und es GAR NICHT MEHR aufgetreten, nachdem KIA ein Softwareupdate verschlampt hat und ich deswegen auf die Live-Services über ein Jahr keinen Zugang mehr hatte (Kundensupport ist bei denen absolut daneben!) - in der Zeit habe ich HA und EVCC aber normal weitergenutzt. Daher würde ich eher vermuten, dass im Auto selbst gerne so ein Kommunikationsmodul einfach hängen bleibt und das KIA das ganz alleine versaut hat und dass es einfach Glück/Pech ist, wann das Problem auftritt.

Dass sich KIA einen Dreck um Kunden kümmert, zeigt ja auch der Umgang mit kritischen Sicherheitslücken, durch welche man ein Fahrzeug einfach mitnehmen kann ... es wurde in mehrfachen Kontakten immer wieder betont, wie sehr sich KIA für die Verbesserung der Sicherheit einsetzen würde, um mir dann zu raten, ich möge mir eine Lenkradkralle zulegen. Ein von mir unterbreiteter Vorschlag zur Lösung des Problems wurde zwei Mal mit leeren Texthülsen beantwortet, die gezeigt haben, dass man meine Mail nicht mal gelesen hat. 
Ich würde von KIA also nichts sinnvolles in Richtung der API erwarten.

Die Sache mit der 12V ist für mich auch schleierhaft. Die Karre kriegt ja sogar mit, dass die 12V aufgibt und gibt eine Warnung aus ... nur um dann in der Garage stehend stumpf zu verrecken, während der Fahrakku 80% hat. Wäre absolut vermeidbar gewesen, dass ich mal wieder mit meinem Powerpack im Motorraum hantieren muss (und mit Gehörschutz, weil ja jedes Mal die Alarmanlage loslegt, während man gerade mit dem Gesicht direkt über der Hupe hängt)

Ein KIA wird mir auch nie wieder auf den Hof kommen, wenn mein E-Niro durch damit ist.

Zur Sache EVCC: Ich habe auch ein Ticket bei KIA eröffnet. Wird aber sicherlich auch nur wieder leeres BlaBla kommen

flandste, 2025-09-05:
KIA Niro EV / 3.1.20
Bei der Installation werden folgende Meldungen angezeigt - möglicherweise hilft das weiter:

$ iobroker upgrade bluelink@3.1.20 --debug
Update bluelink from @3.1.3 to @3.1.20
NPM version: 10.9.3
Installing iobroker.bluelink@3.1.20... (System call)
npm warn deprecated npmlog@4.1.2: This package is no longer supported.
npm warn deprecated are-we-there-yet@1.1.7: This package is no longer supported.
npm warn deprecated glob@7.2.3: Glob versions prior to v9 are no longer supported
npm warn deprecated gauge@2.7.4: This package is no longer supported.
> bluelinky@9.1.0 prepare> tsc
> usocket@0.3.0 install> node-gyp rebuild
gyp info it worked if it ends with ok
gyp info using node@22.19.0 | linux | arm64
(node:122369) [DEP0060] DeprecationWarning: The `util._extend` API is deprecated. Please use Object.assign() instead.(Use `node --trace-deprecation ...` to show where the warning was created)
gyp info find Python using Python version 3.11.2 found at "/usr/bin/python3"
gyp ERR! UNCAUGHT EXCEPTION
gyp ERR! stack at FSReqCallback.oncomplete (node:fs:187:23)gyp ERR! System Linux 6.12.34+rpt-rpi-2712gyp ERR! command "/usr/bin/node" "/opt/iobroker/node_modules/usocket/node_modules/.bin/node-gyp" "rebuild"gyp ERR! cwd /opt/iobroker/node_modules/usocketgyp ERR! node -v v22.19.0gyp ERR! node-gyp -v v7.1.2gyp ERR! Node-gyp failed to build your package.gyp ERR! Try to update npm and/or node-gyp and if it does not help file an issue with the package author.
changed 2 packages in 29s
240 packages are looking for funding run `npm fund` for details
Updating objects from io-package.json for adapter "bluelink" with version "3.1.20"
Update "system.adapter.bluelink.0"
upload [12] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/bluelink.png bluelink.png image/png
upload [11] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/i18n/de/translations.json i18n/de/translations.json application/json
upload [7] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/i18n/it/translations.json i18n/it/translations.json application/json
upload [6] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/i18n/nl/translations.json i18n/nl/translations.json application/json
upload [5] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/i18n/pl/translations.json i18n/pl/translations.json application/json
upload [4] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/i18n/pt/translations.json i18n/pt/translations.json application/json
upload [3] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/i18n/ru/translations.json i18n/ru/translations.json application/json
upload [1] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/i18n/zh-cn/translations.json i18n/zh-cn/translations.json application/json
upload [0] bluelink.admin /opt/iobroker/node_modules/iobroker.bluelink/admin/jsonConfig.json jsonConfig.json application/json
Process exited with code 0

andig, 2025-09-05:
Falsches repo…

Mike-F362, 2025-09-07:
@Magnus8513 , @TheMr-S Bitte mal den Thread aufmerksam durcharbeiten. Hier: https://github.com/evcc-io/evcc/issues/23147#issuecomment-3250132784 ist der Weg zur quasi schlüsselfertigen Lösung beschrieben.

Magnus8513, 2025-09-07:
Does this require a home assistant instance - which I don't run on the raspi in question at the moment.

Mike-F362, 2025-09-07:
> Does this require a home assistant instance - which I don't run on the raspi in question at the moment?

Yes, this handles the docker compose setup. On raspi you should move the docker image dir away from the sd-card towards to a mounted usb-ssd.

Getiem, 2025-09-08:
> Antwort von den EU-Wettbewerbshütern:
> 
> ```
> Danke für Ihre Nachricht.
> Wir betrachten es als Marktinformation und haben es an die Kollegen, die sich mit diesen Themen befassen, weitergeleitet.
> Sollten diese noch Fragen haben werden wir Sie kontaktieren
> DG COMP Response Team
> ```
> 
> ... zumindest haben sie es nicht gleich geschreddert. 😉
> 
> [@pencoe](https://github.com/pencoe): Ich habe das [Repo](https://github.com/Mike-F362/EVCC-Homeassistant-Config.git) für EVCC-HA inkl. Patch fertig gestellt (Siehe [ReadMe](https://github.com/Mike-F362/EVCC-Homeassistant-Config/blob/main/Readme.md)).

Thanks, working.

Alternative Instructions for if you have HomeAssistant already running and having more dev-skills a little instruction from my side:  

1. follow the fixes for the HomeAssistant HACS integration for KIA found here:  https://github.com/Hyundai-Kia-Connect/kia_uvo/issues/1277#issuecomment-3223019018

Used this thread to solve some sub-puzzles https://github.com/Hyundai-Kia-Connect/kia_uvo/discussions/1285: 
a) Remind: if you want to replace the KiaUvoApiEU.py, you will find it normally in the specific subfolders  \config\deps\, but if that one is empty, you should look INTO the homeassistant container. Going there is a little bit tricky. You have to use the ADVANCED ssh & webterminal  (and not the normal one), and have to switch off Safety Mode that is on that Add-one. 
You will then have to go into the specific container:   https://community.home-assistant.io/t/execute-in-home-assistant-container-context/415031/4?u=get_em
Replace file  (using command move file (mv))  in folder /usr/local/lib/python3.13/site-packages/hyundai_kia_connect_api/KiaUvoApiEU.py.

RESTART Homeassistant to take effect.

b) Token fetching: You can use firefox with the instructions when running the KiaFetchApiTokens.py   (i used this one https://gist.github.com/marvinwankersteen/af92c571881ac76579a037fac4f3a63a#file-kiafetchapitokens-py ), but remind to copy the EXACT user agent string and set in the Firefox Addon "User-Agent Switcher and Manager"

c) when you found your Refreshtoken using the script, the instructions from point 1 say you replace the value at the password configuration. But alternatively you can best remove your Kia from HA, and then re-ad the integration, now with Password field filled with the Refreshtoken.  That is much easier.

2. Alternative instructions for adding the HomeAssistant Kia integration sensors to your EVCC:

a) generate a long-lived token  in HomeAssistant   http://homeassistant.local:8123/profile/security
b) Add a extra vehicle in EVCC using the interface configuration, choose option: HomeAssistant Vehicle, deep down in the dropdown choice box
c) fill in all the values for the sensors and the long-lived token. Simply use the HomeAssistant Entity-id copy function to copy it. 
d) validate input, save and restart EVCC

In general, you can just have a second configured vehicle next to your primary vehicle. I named my configured vehicle Kia HA.

JHPWBN, 2025-09-09:
@Getiem: > Long Story Short: Once you have a working KIA connection in HomeAssistant, It is really easy to configure the Kia via the EVCC interface configuration. Use simply HomeAssistant Vehicle.

Maybe I am missing something here. I do not have a working KIA connection (using EVCC in HomeAssistant).
If it would work then I do not need the workaround you  describe.

What do I not understand ?

plieven, 2025-09-09:
Wouldn't it be feasible to implement a way to store the refresh token in evcc as soon as the new approach turns out to be stable? Install HA just to read from the Kia API seems overkill.

I can look at it as soon as need to restart evcc or my refresh token expires. The implementation in evcc seems to be totally stable if it has a valid refresh token.

Getiem, 2025-09-09:
> [@Getiem](https://github.com/Getiem): > Long Story Short: Once you have a working KIA connection in HomeAssistant, It is really easy to configure the Kia via the EVCC interface configuration. Use simply HomeAssistant Vehicle.
> 
> Maybe I am missing something here. I do not have a working KIA connection (using EVCC in HomeAssistant). If it would work then I do not need the workaround you describe.
> 
> I do no have any dev skills either.
> 
> What do I not understand ?

Ah, well. You can have your KIA connected two ways. 

Information flow:
1. . KIA --> EVCC
2.  KIA --> HomeAssistant --> EVCC

First is de default with the KIA integration in EVCC. This one is broken at this moment, thats this issue about.

Second is a workaround, by using a HACS integration for connecting your KIA to HomeAssistant. And then connecting the information-sensors from HomeAssistant into EVCC as a virtual Vehicle. 

Mind you, this has nothing to do with where EVCC runs. If it runs in a container/add-on within HomeAssistant or as a separate VM or RaspberryPi, does not matter. 

For the second solution you need that working connection between KIA and HomeAssistant. Since this issue in EVCC with KIA is due to Changes at the KIA servers, also the connection from KIA with the HomeAssistant was broken. My first part describes the fix of the Hyundai-Kia HACS integration to connect KIA with HomeAssistant.

So in short: EVCC direct connection to KIA is still not fixed. HomeAssistant integration to KIA is also not fixed. But for the HomeAssistant integration there is a scripted workaround (hence the developer knowledge needed). If that workaround works for you, you can then easily create a virtual vehicle in EVCC with the information from the HomeAssistant Kia sensors.  
If you do not have Developer Knowledge, the workaround is probably difficult, so the advise is waiting until the developers of EVCC and the KIA-Hyundai libary fix things that makes it easier for non-developers.

Getiem, 2025-09-17:
> Die evcc nightly von gestern hat bereits den Fix integriert

Thanks, it wasnt in the release notes, therefore I missed it, but i found it in the commits for this release  
https://github.com/evcc-io/evcc/compare/0.207.6...master

Robert28021962, 2025-09-17:
> 0.207.6...master

Because this is all more or less new stuff for me!
I understand that:
1) get the evcc version v.0.207.6
2)  I have to get somewhere a refresh token from kia
3)  use that token in the evcc yaml instead of my password

it looks like this :

vehicles:
  - name: my_car
    type: kia
    user: my email    # Username, e.g. email address, user id, etc.
    password: 'REFRESH TOKEN'     # Password, Use single quotes for leading
zeros
    vin: 'KNFAD......'
    title: Kia EV3            # Title, Will be displayed in the user
interface (optional)
    capacity: 81              # Battery capacity (kWh), optional
    language: en              # Language, [en, de] (optional)
    cache: 15m

is that correct ?

Robert28021962, 2025-09-17:
> That's *mostly* correct, the major thing that's not correct is
>
>
>    1. get the evcc version v.0.207.6
>
> because the necessary changes to evcc aren't in that version. Right now
> you have to use the *nightly* version
> <https://docs.evcc.io/docs/installation/linux> (Switch to the "Nightly"
> tabs).

thanks!

dieter114, 2025-09-24:
Ich habe nach dem Post von Getiem einen Token erzeugt und die Nightly installiert.
Mein KIA PHEV ist wieder einwandfrei integriert und auch nach einigen Neustarts immer noch erreichbar.
Hat Jemand von Euch Erfahrungen wie lange so ein Token hält?
Also wenn alle 6-8 Wochen ein neuer erzeugt werden muss könnte ich damit erstmal leben
aber wenn es nur eine Woche ist ??

JHPWBN, 2025-09-27:
Good Morning (yes it is!)
I installed the regular update for Home Assistant and changed the password in yaml to the refresh token
My KIA is recognised again.
Thank you so much!

Robert28021962, 2025-09-27:
Hi great that there is a working solution fot the Kia problem,  but it is still not clear for me where and how to find the refresh token.  Is there anybody who can explain that ?
I have HA in a raspberry pi 5 and the latest version of EVCC  (V0208)

Tenker1, 2025-09-27:
Because this is all more or less new stuff for me!
I understand that:
1)  get the evcc version v.0.208.0 + Nightly (curl -1sLf 'https://dl.evcc.io/public/evcc/unstable/setup.deb.sh' | sudo -E bash)
2)  I have to get somewhere a refresh token from kia
3)  use that token in the evcc yaml instead of my password
     password: 'REFRESH TOKEN'     # Password, Use single quotes for leading
4)  sudo systemctl restart evcc

**Unfortunately, my EV6 is still not recognized.**

```
[main ] INFO 2025/09/27 15:18:02 evcc 0.208.0 (4a4027c7)
[main ] INFO 2025/09/27 15:18:02 using config file: /etc/evcc.yaml
[db ] INFO 2025/09/27 15:18:02 using sqlite database: /var/lib/evcc/evcc.db
[mqtt ] INFO 2025/09/27 15:18:02 connecting evcc-931171498 at tcp://192.168.0.13:1883
[main ] INFO 2025/09/27 15:18:02 UI listening at :7070
[mqtt ] DEBUG 2025/09/27 15:18:02 tcp://192.168.0.13:1883 connected
[main ] ERROR 2025/09/27 15:18:03 creating vehicle ev6 failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': cannot find vehicle, got: [KNAC481AFP5154653]
[tariff] DEBUG 2025/09/27 15:18:03 using cache: grünstromindex-d226acc33428df984379abdcd649118b33008b361800e6cba5deab9270ea6911 (start: 2025-09-27 14:00:00 +0000 UTC, end: 2025-10-01 23:00:00 +0000 UTC)
[lp-1 ] DEBUG 2025/09/27 15:18:03 charge total import: 5420.000kWh
[site ] WARN 2025/09/27 15:18:04 vehicle 'Kia EV6' does not support automatic detection
```

**Thank you for your help.**

RustyDust, 2025-09-27:
@Robert28021962 
Try [this Wiki entry](https://github.com/evcc-io/evcc/wiki/Kia:-Refresh%E2%80%90Token-ermitteln)

@Tenker1 
Since 0.208.0 there should be no need to use a nightly. As for your problem: did you specify the correct VIN in your config/the UI?