evcc-io/evcc #23149: Peblar charger with firmware 1.8.0 fails to set charge limit

dutchiechris, 2025-08-21:
### Describe the bug

Updated Peblar Home charger firmware from to 1.8.0 (previously 1.7.1) and EVCC control of charger is no longer possible throwing the error:

```
[lp-1 ] ERROR 2025/08/21 19:25:58 set charge current limit 6A: modbus: exception '4' (server device failure), function '16'
```

Verified smart charging settings are set to `Default` and modbus is enabled for read/write.

### Steps to reproduce

1. Peblar Home with firmware 1.8.0
2.
3.
...

### Configuration details

```yaml
chargers:
  - name: peblar
    type: template
    template: peblar
    modbus: tcpip
    id: 1
    host: 192.168.0.6
    port: 502
```

### Log details

```text
More info shows reading via modbus is fine:

 evcc charger  --diagnose
[main  ] INFO 2025/08/21 19:11:07 using config file: /etc/evcc.yaml
[db    ] INFO 2025/08/21 19:11:07 using sqlite database: /root/.evcc/evcc.db
Power:          1196W
Energy:         434.2kWh
Current L1..L3: 5.37A 0A 0A
Voltage L1..L3: 226V 0V 0V
Charge status:  C
Enabled:        true
Charged:        1.5kWh
Diagnostic dump:
        SN:     <REDACTED>
        PN:     <REDACTED>
        Firmware:       1.8.0+1+WL-1

But trying to set a charging level fails:

/app # evcc charger  -i 1400 --log trace 
[main  ] INFO 2025/08/21 19:13:37 using config file: /etc/evcc.yaml
[db    ] INFO 2025/08/21 19:13:37 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/08/21 19:13:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="sessions" -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "sessions" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `sessions` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `sessions` 68 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="meters" -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "meters" AND sql IS NOT NULL order by type = "table" desc 2 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `meters` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT count(*) FROM sqlite_master WHERE type = "index" AND tbl_name = "meters" AND name = "meter_ts" -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `settings` 14 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="caches" -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "caches" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `caches` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/08/21 19:13:37 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[main  ] ERROR 2025/08/21 19:13:37 set current: modbus: exception '4' (server device failure), function '16'
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.3 (e4d6cd58)

andig, 2025-08-21:
Open issue with peblar? What has changed on their side?

JeppeKnockaert, 2025-08-22:
I was in contact with Peblar two days ago and they let met know that they will fix the issue in the next patch release. For now, you can either:

- Downgrade to 1.7.1
- Configure Peblar to 3 phases and then use modbus to force 1 phase

closed 2025-08-22 by andig