evcc-io/evcc #23155: Add support for Netcontrol 2302 switch

growers12, 2025-08-22:
**Is your feature request related to a problem? Please describe.**
Gude Netcontrol 2302 seems to be popular with some installers to control water heating. It should be integrated as a Switch. As there is no power monitoring available with this, power has to be assigned for each output via config file or UI.

**Describe the solution you'd like**
Integrate a switch devices similar to the "MyStrom" switch integration
[Netcontrol 2302](https://gude-systems.com/en/products/expert-net-control-2302/)

**Describe alternatives you've considered**
none

dingoduffel, 2025-08-22:
maybe you could also integrate the switch in HA, and by doing so make it available in evcc?
see: https://community.home-assistant.io/t/gude-expert-net-control-2301-smart-relais-power-sensors/374245

dingoduffel, 2025-08-22:
@growers12 , it's actually the other way around: once your switch is defined in HA, you can make it available in evcc. Please see https://docs.evcc.io/en/docs/devices/smartswitches#home-assistant-switch if this woud work for you. I had before a problem to integrate Legrand Netatmo switches, and opened an issue for it, but then was redirected by @andig to this solution, which works as a charm.

closed 2025-09-04 by github-actions[bot]