evcc-io/evcc #23156: Add generic SoC HTTP/BLE proxy

growers12, 2025-08-22:
**Is your feature request related to a problem? Please describe.**
SOX Monitroing with many modern cars works only if mobile network is available. That's not always the case where the charger is located. Alternative method to read the SoC should be considered

**Describe the solution you'd like**
There are many cheap OBD dongles on the market that can be use to access SoC data (not standardized). EVCC could either directly pull data via BLE (if close to the car) or via some simple proxy host (e.g. an ESP32Home device).

**Describe alternatives you've considered**
There are services available to monitor via OBD, but they often require additional hardware (e.g. Android devices)

StefanSchoof, 2025-08-22:
There are plenty of options there: https://docs.evcc.io/en/docs/devices/vehicles#generic-support

What exactly is missing for you?

growers12, 2025-08-22:
But this works only for Tesla.

andig, 2025-08-22:
evcc does not have a BLE plugin. You can still integrate via HA, ioBroker and any other platform using the `custom` vehicle.