evcc-io/evcc #23158: setting constant battery capacity fails expected type 'float64', got unconvertible type 'map[string]interface {}'

macrl2000, 2025-08-22:
### Describe the bug

When adding a Battery (custom) using constant capacity fails

```yaml
power: # current power
  source: const
  value: 1000 # W

soc: # state of charge in % (for battery)
  source: const
  value: 75

capacity: # capacity in kWh (for battery)
  source: const
  value: 10.5
```

cannot create meter type 'custom': decoding failed due to the following error(s): 'Capacity' expected type 'float64', got unconvertible type 'map[string]interface {}'

without capacity working fine - what do i miss ?

### Steps to reproduce

1.In database mode, try to add a home battery, using "custom battery" using yaml 
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

version 0.207.4 (e4d6cd58)

andig, 2025-08-22:
Error message says it all:

    capacity: 10.5

closed 2025-08-22 by andig

LeonSol, 2025-08-22:
capacity ist bereits vom typ "const"! Der Eintrag lautet also capacity: 10.5

andig, 2025-08-22:
/cc @naltatis