evcc-io/evcc #23181: Add Support For Danish Energi Data Service Grid co2 Forecast

gllmlbrt, 2025-08-22:
**Is your feature request related to a problem? Please describe.**

https://www.energidataservice.dk/tso-electricity/CO2EmisProg provides a near term prognosis of grid CO2 intensity, unecessarily with 5min interval, for free and without any authentication requirement for both DK1 and DK2 grid zones.

The API allows to draw a lot more data, but the need and opportunity here is about CO2 forecast as electricity cost is already covered in EVCC with energinet template.

Here is an example of API request:
https://api.energidataservice.dk/dataset/CO2EmisProg?filter={%22PriceArea%22:[%22DK2%22]}

Here is the format of the query response with limit to 10 for this post (default is 100):
`{
  "total": 909204,
  "filters": "{\"PriceArea\":[\"DK2\"]}",
  "limit": 10,
  "dataset": "CO2EmisProg",
  "records": [
    {
      "Minutes5UTC": "2025-08-23T21:55:00",
      "Minutes5DK": "2025-08-23T23:55:00",
      "PriceArea": "DK2",
      "CO2Emission": 26
    },
    {
      "Minutes5UTC": "2025-08-23T21:50:00",
      "Minutes5DK": "2025-08-23T23:50:00",
      "PriceArea": "DK2",
      "CO2Emission": 26
    },
    {
      "Minutes5UTC": "2025-08-23T21:45:00",
      "Minutes5DK": "2025-08-23T23:45:00",
      "PriceArea": "DK2",
      "CO2Emission": 26
    },
    {
      "Minutes5UTC": "2025-08-23T21:40:00",
      "Minutes5DK": "2025-08-23T23:40:00",
      "PriceArea": "DK2",
      "CO2Emission": 26
    },
    {
      "Minutes5UTC": "2025-08-23T21:35:00",
      "Minutes5DK": "2025-08-23T23:35:00",
      "PriceArea": "DK2",
      "CO2Emission": 26
    },
    {
      "Minutes5UTC": "2025-08-23T21:30:00",
      "Minutes5DK": "2025-08-23T23:30:00",
      "PriceArea": "DK2",
      "CO2Emission": 26
    },
    {
      "Minutes5UTC": "2025-08-23T21:25:00",
      "Minutes5DK": "2025-08-23T23:25:00",
      "PriceArea": "DK2",
      "CO2Emission": 25
    },
    {
      "Minutes5UTC": "2025-08-23T21:20:00",
      "Minutes5DK": "2025-08-23T23:20:00",
      "PriceArea": "DK2",
      "CO2Emission": 25
    },
    {
      "Minutes5UTC": "2025-08-23T21:15:00",
      "Minutes5DK": "2025-08-23T23:15:00",
      "PriceArea": "DK2",
      "CO2Emission": 25
    },
    {
      "Minutes5UTC": "2025-08-23T21:10:00",
      "Minutes5DK": "2025-08-23T23:10:00",
      "PriceArea": "DK2",
      "CO2Emission": 25
    }
  ]
}`

**Describe the solution you'd like**

Ideally a template for the tariff function in EVCC, allowing to add from the UI as follows:
```
tariffs:
  co2:
    type: template
    template: energidataservice
    uri: https://api.energidataservice.dk/dataset/CO2EmisProg # URI, HTTP(S) address
    token: # Token
    zone: DK2 # Zone DK1 or DK2
```

**Describe alternatives you've considered**
There is no alternative available in EVCC for Denmark. Perhaps it could be possible to import Home Assistant sensor values to bring into the forecast and Clean Grid method of EVCC ?

**Additional context**

andig, 2025-08-22:
API has data backwards (that could be fixed by jq) but it will also need aggregation of the 5min intervals. It would be helpful if we had a plugin to do that that takes from/to slots and either aggregates (plugin 1) or splits (plugin 2). Operation depends on the type of data. Price/ Co2 would be averaged, Solar production summed.

thecem, 2025-08-22:
```yaml
template: dk-co2-forecast
deprecated: false
products:
  - brand: EnergiDataService DK2/DK1 CO2 API
    description:
      de: CO₂-Intensitätsdaten für Dänemark (DK1 und DK2) stündlich aggregiert
      en: CO2 intensity data for Denmark (DK1 and DK2) aggregated hourly

requirements:
  description:
    de: Stündliche CO₂-Intensitätsdaten aus der EnergiDataService API für Dänemark.
    en: Hourly CO2 intensity data from EnergiDataService API for Denmark.
  evcc: ["skiptest"]

group: co2

params:
  - name: priceArea
    type: string
    default: DK2
    description:
      de: Auswahl zwischen DK1 und DK2
      en: Select between DK1 and DK2
  - name: interval
    default: 1h
    advanced: true

render: |
  type: custom
  tariff: co2
  forecast:
    source: http
    uri: "https://api.energidataservice.dk/dataset/CO2EmisProg?filter={\"PriceArea\":[\"{{ .priceArea }}\"]}"
    jq: |
      [.records[]
        | {
          start: (.Minutes5UTC[0:13] + \":00:00Z\"),
          end: (.Minutes5UTC[0:13] 
                | strptime(\"%Y-%m-%dT%H\") 
                | mktime + 3600 
                | strftime(\"%Y-%m-%dT%H:%M:%SZ\")),
          value: .CO2Emission
        }
      ]
      | group_by(.start)
      | map({
          start: .[0].start,
          end: ..end,
          value: (map(.value) | add / length | floor)
        })
      | tostring
  interval: {{ .interval }}
```

andig, 2025-09-20:
@gllmlbrt please feel free to open PR.