evcc-io/evcc #23211: Optimizer connection refused on HA Addon

lukastripp, 2025-08-24:
### Describe the bug

Neues POC Feature Optimizer lädt nicht
optimizer: Post \"http://localhost:7050/optimize/charge-schedule\": dial tcp [::1]:7050: connect: connection refused

Der Port ist im Addon für evopt TCP 7050 aktiviert

Der Punkt Optimizer taucht dann einfach nicht auf.

### Steps to reproduce

Port offen im Addon
Nach jedem Neustart die gleich „Connection refused“ Meldung

### Configuration details

```yaml
Port offen im HA Addon
```

### Log details

```text
2 ] DEBUG 2025/08/24 10:14:22 detected connected phases: 1p
[lp-2 ] DEBUG 2025/08/24 10:14:22 charge total import: 557.307kWh
[lp-2 ] DEBUG 2025/08/24 10:14:22 charger status: B
[site ] DEBUG 2025/08/24 10:14:22 optimizer: optimizing 65 slots until 2025-08-27 02:00:00 +0200 CEST: grid=159, feedIn=159, solar=65, first slot: 45m37s
[site ] ERROR 2025/08/24 10:14:22 optimizer: Post "http://localhost:7050/optimize/charge-schedule": dial tcp [::1]:7050: connect: connection refused
[site ] DEBUG 2025/08/24 10:14:22 solar forecast: accumulated 0.016Wh from 2025-08-24 10:14:02 +0200 CEST to 2025-08-24 10:14:22 +0200 CEST
[site ] DEBUG 2025/08/24 10:14:22 solar forecast: produced 374.118
[site ] DEBUG 2025/08/24 10:14:22 solar forecast: accumulated 423.807kWh, produced 374.118kWh, scale 0.883
[site ] DEBUG 2025/08/24 10:14:42 ----
[lp-3 ] DEBUG 2025/08/24 10:14:42 charge power: 0W
[lp-3 ] DEBUG 2025/08/24 10:14:42 charge currents: [0 0 0]A
[lp-2 ] DEBUG 2025/08/24 10:14:42 charge power: 0W
[lp-2 ] DEBUG 2025/08/24 10:14:42 charge currents: [0.012 0 0]A
[lp-1 ] DEBUG 2025/08/24 10:14:42 charge power: 0W
[lp-1 ] DEBUG 2025/08/24 10:14:42 charge currents: [0 0 0]A
[site ] DEBUG 2025/08/24 10:14:42 grid power: 0W
[site ] DEBUG 2025/08/24 10:14:42 grid powers: [280 -490 210]W
[site ] DEBUG 2025/08/24 10:14:42 grid currents: [2.64 -2.62 1.14]A
[site ] DEBUG 2025/08/24 10:14:42 pv 1 power: 1641W
[site ] DEBUG 2025/08/24 10:14:42 !! solar production: accumulate set my_pv 9777.417kWh meter total (was: Accumulated: 374.118kWh updated: 2025-08-24 10:14:22 +0200 CEST meter: 9777.409kWh)
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.4

andig, 2025-08-24:
See HA discussion

closed 2025-08-24 by andig

andig, 2025-08-24:
All HA topics belong into https://github.com/evcc-io/evcc/discussions/categories/homeassistant. We can‘t give HA support.