evcc-io/evcc #23219: SMA Smart Energy Hybrid Inverter zeigt nicht ersichtliche Konfigurationsoption

Tombra1889, 2025-08-24:
### Describe the bug

Wenn ich einen SMA Smart Energy Hybrid Inverter als Battery konfigurieren möchte, gibt es eine nicht ersichtliche    Konfigurationsoption. "[]" mit optional:

[image]

### Steps to reproduce

Neues Gerät als SMA Smart Energy Hybrid Inverter über die UI konfigurieren.

### Configuration details

```yaml
nicht relevant
```

### Log details

```text
nicht relevant
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.207.4

andig, 2025-08-24:
@naltatis das hatten wir schonmal?

spuky, 2025-08-24:
kleine Korrektur... bei mir kommt es nur das erste mal nach einem Neustart richtig...

spuky, 2025-08-24:
Ja ist nur beim ersten mal nach dem Start von EVCC richtig... jedes weitermal ist es falsch... meine Vermutung da wird ein JS objekt nicht richtig weggeräumt... und dann ein zweites mal gefüllt...

andig, 2025-08-24:
Ahh:

```
					<PropertyEntry
						v-for="param in normalParams"
						:id="`meterParam${param.Name}`"
						:key="param.Name"
						v-bind="param"
						v-model="values[param.Name]"
					/>
```

naltatis, 2025-08-25:
[image]

Die API liefert hier ein Empty Attribut zurück. Ich schau mal woran das liegt.

naltatis, 2025-08-25:
@andig du hattest doch kürzlich die Deprecated Props rausgefiltert:

https://github.com/evcc-io/evcc/blob/6132a0b815b0c2b45158abe4607a71182b81623e/templates/definition/meter/sma-hybrid.yaml#L29-L30

spuky, 2025-08-25:
und auch erst beim zweiten Aufruf der Api im ersten aufruf... ist es nicht dabei...

spuky, 2025-08-25:
kurz zwei andere gecheckt die auch 

deprecated: true haben

Sonnenbatterie

Fronius Symo GEN24 Plus

passiert das gleiche beim zweiten Aufruf...

andig, 2025-08-25:
Es gibt da einen Zombieparameter, keine Ahnung wo der her kommt:

[image]

closed 2025-08-25 by andig