evcc-io/evcc #23226: charger enable trotz Modus "Aus"

Robinhio83, 2025-08-24:
### Describe the bug

Wird ein Fahrzeug an die Wallbox angesteckt das permanent Status C über den CP zurück meldet startet evcc die Ladung obwohl keine Ladefreigabe besteht(Modus "Aus" oder kein PV-Überschuss in "PV") nur um die Ladung dann direkt wieder zu unterbrechen.

Dadurch zieht die HeidelbergEC wohl das Netzschütz an, Fahrzeug beginnt zu laden und die HeidelbergEC geht in Fehler(und öffnet damit das Netzschütz). Ich vermute sie wartet auf Status B, was aber nie passiert. Nach dem Fehler startet die Wallbox neu und der Spaß geht von vorne los.

Ich hoffe die configuration details passen so, `evcc dump --cfg --database /var/lib/evcc/evcc.db` wirft mir nur ein paar Fehlermeldungen ala `[lp-2  ] ERROR 2025/08/24 20:12:21 attempt to write a readonly database (8)`

### Steps to reproduce

1. Modus auf "Aus" oder "PV" wenn nicht genug Überschuss
2. Fahrzeug anstecken das permanent Status C zurück meldet (und lädt sobald Spannung anliegt.)
3. Klick-Klack, An-Aus-An-Aus

### Configuration details

```yaml
charger
---
db:9 {Type:template Title: Icon: Product:Walther Werke Basic Evo Pro} map[baudrate:19200 comset:8E1 device:/dev/ttyAMA0 modbus:rs485serial template:heidelberg]
db:11 {Type:template Title: Icon: Product:Walther Werke Basic Evo Pro} map[baudrate:19200 comset:8E1 device:/dev/ttyAMA0 id:2 modbus:rs485serial template:heidelberg]

meter
---
db:6 {Type:template Title: Icon: Product:Fronius Symo GEN24 Plus} map[host:***.***.***.*** id:200 maxacpower:0 maxchargerate:100 port:502 template:fronius-gen24 usage:grid]
db:7 {Type:template Title:Fronius Icon: Product:Fronius Symo GEN24 Plus} map[host:***.***.***.*** id:200 maxacpower:8000 maxchargerate:100 port:502 template:fronius-gen24 usage:pv]
db:8 {Type:template Title:BYD Icon: Product:Fronius Symo GEN24 Plus} map[capacity:7.68 host:***.***.***.*** id:200 maxacpower:0 maxchargerate:25 port:502 template:fronius-gen24 usage:battery]

vehicle
---
db:5 {Type:template Title: Icon: Product:Generisches Fahrzeug (ohne API)} map[capacity:5.5 icon:motorcycle maxCurrent:16 minCurrent:15 mode: phases:1 priority:1 template:offline title:Wildfire]
db:13 {Type:template Title: Icon: Product:Renault} map[cache:15m capacity:48 icon:car minCurrent: password:***** phases: template:renault title:Zoe user:***** vin:***** wakeupmode:default]
db:14 {Type:template Title: Icon: Product:Generisches Fahrzeug (ohne API)} map[icon:car template:offline title:Gastfahrzeug]

loadpoint
---
db:10 {Type: Title: Icon: Product:} map[charger:db:9 circuit:garage defaultMode: limitEnergy:0 maxCurrent:16 meter: minCurrent:6 mode:off phasesConfigured:3 soc:map[estimate:true poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Stirnseite vehicle:db:5]
db:12 {Type: Title: Icon: Product:} map[charger:db:11 circuit:garage defaultMode:pv limitEnergy:0 limitSoc:0 meter: mode:pv phasesConfigured:3 soc:map[estimate:true poll:map[interval:3.6e+12 mode:charging]] thresholds:map[disable:map[delay:1.8e+11 threshold:0] enable:map[delay:6e+10 threshold:0]] title:Tor vehicle:]
```

### Log details

```text
[heidel] TRACE 2025/08/24 19:34:53 modbus: send 01 04 00 05 00 01 21 cb
[heidel] TRACE 2025/08/24 19:34:53 modbus: recv 01 04 02 00 06 39 32
[lp-1  ] DEBUG 2025/08/24 19:34:53 charger status: C
[heidel] TRACE 2025/08/24 19:34:54 modbus: send 01 04 00 11 00 02 21 ce
[heidel] TRACE 2025/08/24 19:34:54 modbus: recv 01 04 04 00 00 0a b6 7c 92
[lp-1  ] DEBUG 2025/08/24 19:34:54 charge start energy: 2.742kWh
[lp-1  ] INFO 2025/08/24 19:34:54 car connected
[lp-1  ] DEBUG 2025/08/24 19:34:54 pv timer elapse
[lp-1  ] DEBUG 2025/08/24 19:34:54 pv timer inactive
[heidel] TRACE 2025/08/24 19:34:54 modbus: send 01 04 00 11 00 02 21 ce
[heidel] TRACE 2025/08/24 19:34:54 modbus: recv 01 04 04 00 00 0a b6 7c 92
[lp-1  ] DEBUG 2025/08/24 19:34:54 charge total import: 2.742kWh
[heidel] TRACE 2025/08/24 19:34:54 modbus: send 01 04 00 11 00 02 21 ce
[heidel] TRACE 2025/08/24 19:34:54 modbus: recv 01 04 04 00 00 0a b6 7c 92
[lp-1  ] DEBUG 2025/08/24 19:34:54 charge start energy: 2.742kWh
[lp-1  ] INFO 2025/08/24 19:34:54 start charging ->
[lp-1  ] DEBUG 2025/08/24 19:34:54 wake-up timer: stop
[heidel] TRACE 2025/08/24 19:34:54 modbus: send 01 03 01 05 00 01 95 f7
[heidel] TRACE 2025/08/24 19:34:54 modbus: recv 01 03 02 00 00 b8 44
[lp-1  ] WARN 2025/08/24 19:34:54 charger logic error: disabled but charging
[heidel] TRACE 2025/08/24 19:34:54 modbus: send 01 10 01 05 00 01 02 00 3c b6 d4
[heidel] TRACE 2025/08/24 19:34:54 modbus: recv 01 10 01 05 00 01 10 34
[lp-1  ] DEBUG 2025/08/24 19:34:54 charger enable
[heidel] TRACE 2025/08/24 19:34:54 modbus: send 01 10 01 05 00 01 02 00 00 b6 c5
[heidel] TRACE 2025/08/24 19:34:54 modbus: recv 01 10 01 05 00 01 10 34
[lp-1  ] DEBUG 2025/08/24 19:34:54 charger disable
[lp-1  ] DEBUG 2025/08/24 19:34:54 wake-up timer: stop
[... 47 lines cut ...]
[heidel] TRACE 2025/08/24 19:35:22 modbus: recv 02 04 02 00 02 7c f1
[heidel] TRACE 2025/08/24 19:35:22 modbus: send 01 04 00 05 00 01 21 cb
[heidel] TRACE 2025/08/24 19:35:22 modbus: recv 01 04 02 00 06 39 32
[heidel] TRACE 2025/08/24 19:35:25 modbus: send 01 04 00 05 00 01 21 cb
[heidel] TRACE 2025/08/24 19:35:25 modbus: recv 01 04 02 00 0a 39 37
[heidel] TRACE 2025/08/24 19:35:25 modbus: send 01 03 01 03 00 01 75 f6
[heidel] TRACE 2025/08/24 19:35:26 modbus: recv 01 03 02 00 01 79 84
[heidel] ERROR 2025/08/24 19:35:26 heartbeat: invalid status: 10
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.4

andig, 2025-08-24:
Ich fürchte da können wir wenig machen. Die Status-Sync Logik brauchen wir für Charger mit Eigenleben.

Robinhio83, 2025-08-24:
> Das ist auch eine Frage wie man Status C1 ganz konkret interpretiert.
> 
> Jedenfalls darf das Fahrzeug im Status B1 ohne PWM nicht anfangen zu laden bzw. C1 signalisieren. Da wird das IEC-State-Modell vom OBC missachtet.
> 
> evcc sieht hier diesen Status und versucht nur das Ganze irgendwie einzufangen/zu syncen.
> 
> So gesehen ist es auch kein Wunder dass dieser OBC überall massive Probleme machen dürfte.

In der Modbustabelle der Walther-Werke basic evo pro wird genau der problematische Zustand  vermerkt, deswegen wundert mich das Verhalten von evcc. Ich ging zuerst davon aus das die Wallbox sich auf den zurück gemeldeten Status verlässt um das Netzschütz zu schließen.

[image]

Ich geb euch recht das die Lösung für die Probleme mit dem Fahrzeug nur durch das Fahrzeug gelöst werden können, Ladung beenden(Übergang C auf B) hätte man ja durch ein "korrektes" Verhalten von evcc nicht gelöst.
Aber durch das aktuelle Verhalten wäre ganz ganz theoretisch Stromdiebstahl möglich....

Robinhio83, 2025-09-04:
Das ergibt Sinn(auch für mich). Dann mach ich hier wohl mal zu.

closed 2025-09-04 by Robinhio83