evcc-io/evcc #23228: Health probe unavailable until OCPP connection established

andresp, 2025-08-24:
### Describe the bug

It appears that the issue pointed out in #4821 is still present in the latest 0.207.4 release. The UI won't listen on port 7070 until after the OCPP chargers in my configuration have connected to the OCPP service, or the timeout for connectiontimeout is reached. Because Kubernetes routes traffic based on health of a pod (container), it won't permit OCPP connections to reach a new (or upgraded) instance of evcc because it is unable to ping the container on /api/health. As you can see, this creates a circular dependency that cannot be resolved: K8s won't route traffic to the new container because the health probe is unreachable, and the health probe is contingent to the OCPP chargers checking in or timing out before it will be responsive. 

### Steps to reproduce

1. Set up a pod for EVCC with a liveness probe on port 7070, path /api/health in a configuration with an OCPP charger and a K8s service for port 7070 and 8887
2. Start pod
3. Observe pod health

Expected: Pod becomes healthy as soon as "UI listening at :7070" appears in container
Actual: Pod recycles due to liveness probe never coming online

### Configuration details

```yaml
site:  
    title: A site
    meters:
      grid: grid
      pv:
        - pv

  loadpoints:
    - title: A charger
      charger: charger
      vehicle: my_car # default vehicle

  chargers:
    - name: charger
      type: template
      template: ocpp-abb
```

### Log details

```text
[main  ] INFO 2025/08/24 23:46:27 evcc 0.207.4
[main  ] INFO 2025/08/24 23:46:27 using config file: /etc/evcc.yaml
[main  ] INFO 2025/08/24 23:46:27 UI listening at :7070
[charger-1] DEBUG 2025/08/24 23:46:27 waiting for chargepoint: 5m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.4

spuky, 2025-08-25:
couldn't you script your probe to 
just check for the occp server on port 8887 for the first 5 minutes after startup and then switch to... /api/health

spuky, 2025-08-25:
Claude meint:

Hier ist das Liveness-Probe-Script für EVCC. Das Script funktioniert folgendermaßen:
Funktionsweise:

Erste 5 Minuten: Prüft ob Port 8887 (OCPP) erreichbar ist
Nach 5 Minuten: Wechselt zur Überprüfung der Health-API auf Port 7070
```
#!/bin/bash

# EVCC Liveness Probe
# Prüft die ersten 5 Minuten OCPP Port 8887, danach /api/health auf Port 7070

STARTUP_TIME=300  # 5 Minuten in Sekunden

# Container-Startzeit ermitteln (PID 1 ist der Hauptprozess im Container)
START_TIME=$(stat -c %Y /proc/1)

# Funktion für OCPP Port Check
check_ocpp_port() {
    nc -z localhost 8887 >/dev/null 2>&1
    return $?
}

# Funktion für Health API Check
check_health_api() {
    response=$(curl -s -o /dev/null -w "%{http_code}" http://localhost:7070/api/health)
    if [ "$response" = "200" ]; then
        return 0
    else
        return 1
    fi
}

# Hauptlogik
current_time=$(date +%s)
elapsed_time=$((current_time - START_TIME))

if [ $elapsed_time -lt $STARTUP_TIME ]; then
    # Erste 5 Minuten: OCPP Port prüfen
    echo "Startup phase: Checking OCPP port 8887 (elapsed: ${elapsed_time}s)"
    if check_ocpp_port; then
        echo "OCPP port 8887 is accessible"
        exit 0
    else
        echo "OCPP port 8887 is not accessible"
        exit 1
    fi
else
    # Nach 5 Minuten: Health API prüfen
    echo "Runtime phase: Checking health API on port 7070"
    if check_health_api; then
        echo "Health API returned OK"
        exit 0
    else
        echo "Health API check failed"
        exit 1
    fi
fi
```
Verwendung:

Als ausführbare Datei speichern:
`bashchmod +x evcc-liveness-probe.sh`

In Kubernetes/Docker als Liveness Probe:
```
livenessProbe:
  exec:
    command:
      - /bin/bash
      - /path/to/evcc-liveness-probe.sh
  initialDelaySeconds: 30
  periodSeconds: 30
  timeoutSeconds: 10
  failureThreshold: 3
```

Manual testen:
bash./evcc-liveness-probe.sh

Anpassungen möglich:

STARTUP_TIME=300 - Ändern Sie die 300 Sekunden (5 Minuten) nach Bedarf
Ports 8887 und 7070 können angepasst werden
Timeout-Werte für curl/nc können hinzugefügt werden

Das Script verwendet nc (netcat) für die Portprüfung und curl für den HTTP-Check. Stellen Sie sicher, dass beide Tools in Ihrem Container/System verfügbar sind.