evcc-io/evcc #23269: Alfen charger is not recognised to have separate phase current measurements

tomvanswam, 2025-08-26:
### Describe the bug

When configuring load management using phases like this:
[image]

I get the following error:
[image]
As far as I can tell this means the load management component thinks the Alfen charger does not have 3 phase current values to calculate the over current on the circuit with.

However when I see the charger settings, it does display 3 currents. And even in the source code I found reference to 3 modbus registers beeing mentioned to read the currents (alfen.go line 47) 
[image]

I'd rather use the current appoach to limit charging instead of the power approach as limiting power could still result in overcurrent on one of the grid breakers if the load on the 3 phases is not identical

### Steps to reproduce

1. Create config with alfen charger, grid meter and load management configured with MaxCurrent and grid meter
2. Restart EVCC
3. Get the error shown in the describe the bug first screenshot.

### Configuration details

```yaml
app # evcc dump --database /data/evcc.db 
[main  ] INFO 2025/08/26 21:35:06 config file not found, database-only mode
[db    ] INFO 2025/08/26 21:35:06 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/08/26 21:35:07 connecting evcc-2050884725 at tcp://a0d7b954-emqx:1883
config
======

grid: db:4
----------

pv 1: db:5
----------

vehicle: Ruimteschip
--------------------
Soc:           41%
Capacity:      80.0kWh
Charge status: B
Range:         220km
Odometer:      9447km
Limit Soc:     70%
OnIdentified:  Mode:pv, MinCurrent:5, MaxCurrent:16
Features:      [CoarseCurrent]

loadpoint 1
===========

charger: db:1
-------------

/app # evcc dump --database /data/evcc.db --cfg
[main  ] INFO 2025/08/26 21:36:07 config file not found, database-only mode
[db    ] INFO 2025/08/26 21:36:07 using sqlite database: /data/evcc.db
[mqtt  ] INFO 2025/08/26 21:36:07 connecting evcc-1927402219 at tcp://a0d7b954-emqx:1883

Version: `0.207.5`
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.5

mfuchs1984, 2025-08-26:
What exactly is your goal? Maybe there are other ways to achieve it by adding additional circuits.

tomvanswam, 2025-08-27:
I'd like to keep the current on each of the 3 phases on my grid meter under 25A, as my grid circuit breakers are 25A.
So when total power on any of the 3 phases exeeds 25A, charging current should be dialed back. (E.g. when an oven starts on a phase drawing 16A, together with the 16A of the charger this will overload the grid breaker on that phase).

I would not mind dialing back all 3 phases on the charger when only one of the phases is exceeding 25A, to keep the load for the charger equal on all phases, however as I interpret the error, this seems not to be possible.

I was just able to take a screenshot while charging:

[image]

This also shows that the per phase current is available for the Alfen charger, so I do not understand why this is not working.

mfuchs1984, 2025-08-27:
If your want to keep the currents at your main breakers lower than 25, your grid meter needs to support phase currents. The alfen meter does not help there.