evcc-io/evcc #23272: panic: runtime error: invalid memory address or nil pointer dereference

gllmlbrt, 2025-08-27:
### Describe the bug

panic: runtime error: invalid memory address or nil pointer dereference

### Steps to reproduce

1. Changed the tariff parameters in the UI (Using Energinet-Price template)
2. Restart add-on
3.no access to UI to revert tariff
4. Add does not start.

Tried with nightly, but there I got an invalid sponsor token error and did not get anywhere.

### Configuration details

```yaml
#No access to the UI so I cannot copy-paste the exact config. but it is something like that:
  currency: DKK
  grid:
    type: template
    template: energinet-price
    region: dk2 # or dk1
    charges: 0.7673 # 0.7610 'elafgift' + 0.0063 surchage on spot price
    # elafgift see https://findenergi.dk/guides/elafgift/#elementor-toc__heading-anchor-3
    # surcharge Vindstød see https://vindstoed.dk/se-hvad-du-kan-spare-og-tilmeld
    tax: 0.25 # 25% VAT
    # dynamic gridfees defined by local distribution grid operator;
    # below fees are taken from https://radiuselnet.dk/elnetkunder/tariffer-og-netabonnement/
    formula: |
      // Variable declaration
      gridfee := 0.0

      // Determine if it's summer (April - September)
      isSummer := ts.Month() >= 4 && ts.Month() <= 9

      // Set grid fee based on season and time of day
      if isSummer {
        if ts.Hour() < 6 {
          gridfee = 0.0976  // low load summer
        } else if ts.Hour() >= 17 && ts.Hour() < 21 {
          gridfee = 0.3808  // peak load summer
        } else {
          gridfee = 0.1467  // high load summer
        }
      } else {
        if ts.Hour() < 6 {
          gridfee = 0.0976  // low load winter
        } else if ts.Hour() >= 17 && ts.Hour() < 21 {
          gridfee = 0.8788  // peak load winter
        } else {
          gridfee = 0.2929  // high load winter
        }
      }

      (price + charges + gridfee) * (1 + tax)

  feedin:
    type: template
    template: energiet-price
    region: dk2
    formula: |
      // specification of surcharges per kWh
      energinet_indfoedningstarif := -0.0050  // https://energinet.dk/el/elmarkedet/tariffer/aktuelle-tariffer/
      energinet_balancetarif      := -0.0065  // https://energinet.dk/el/elmarkedet/tariffer/aktuelle-tariffer/
      radius_indfoedningstarif    := -0.0059  // https://radiuselnet.dk/elnetkunder/tariffer-og-netabonnement/, incl. VAT
      vindstoed_balancetarif      := -0.0100  // https://www.vindstoed.dk/tilmelding-solcelle

      (price + energinet_indfoedningstarif + energinet_balancetarif + radius_indfoedningstarif + vindstoed_balancetarif)
```

### Log details

```text
Using config file: /config/evcc/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/config/evcc/evcc.db evcc --config /config/evcc/evcc.yaml'
[main  ] INFO 2025/08/27 07:38:06 evcc 0.207.5
[main  ] INFO 2025/08/27 07:38:06 using config file: /config/evcc/evcc.yaml
[db    ] INFO 2025/08/27 07:38:06 using sqlite database: /config/evcc/evcc.db
[mqtt  ] INFO 2025/08/27 07:38:06 connecting evccmqtt at tcp://192.168.68.120:1883
[main  ] INFO 2025/08/27 07:38:06 UI listening at :7070
panic: runtime error: invalid memory address or nil pointer dereference
[signal SIGSEGV: segmentation violation code=0x1 addr=0x38 pc=0x1d55b60]
goroutine 1 [running]:
github.com/evcc-io/evcc/tariff.(*Tariffs).Get(0x40000aaa40?, 0x24c94a0?)
	github.com/evcc-io/evcc/tariff/tariffs.go:58 +0x60
github.com/evcc-io/evcc/core.(*Site).GetTariff(0x40010837a0?, 0x5?)
	github.com/evcc-io/evcc/core/site_api.go:289 +0xa8
github.com/evcc-io/evcc/core.(*Site).Boot(0x4000bf2fc8, 0x40006403c0, {0x4000c83c00, 0x7, 0x400043c180?}, 0x4000e41050?)
	github.com/evcc-io/evcc/core/site.go:167 +0x354
github.com/evcc-io/evcc/cmd.configureSite(0x0?, {0x4000c83c00, 0x7, 0x7}, 0x0)
	github.com/evcc-io/evcc/cmd/setup.go:1054 +0x50
github.com/evcc-io/evcc/cmd.configureSiteAndLoadpoints(0x5d4d140)
	github.com/evcc-io/evcc/cmd/setup.go:996 +0x3ac
github.com/evcc-io/evcc/cmd.runRoot(0x5d4b200, {0x2a20fcd?, 0x7?, 0x2a14630?})
	github.com/evcc-io/evcc/cmd/root.go:201 +0x88c
github.com/spf13/cobra.(*Command).execute(0x5d4b200, {0x400011e280, 0x2, 0x2})
	github.com/spf13/cobra@v1.9.1/command.go:1019 +0x7bc
github.com/spf13/cobra.(*Command).ExecuteC(0x5d4b200)
	github.com/spf13/cobra@v1.9.1/command.go:1148 +0x350
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:123 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x34
	github.com/spf13/cobra@v1.9.1/command.go:1148 +0x350
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:123 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x34
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.5

andig, 2025-08-27:
There is no `energiet-price` template. You're missing an n. With that change tariffs are shown. But even without, I cannot repro a crash. Can you provide more details how to reproduce the crash?

gllmlbrt, 2025-08-27:
That's jsut me typing in poorly. I am sure I ahve energinet without the typo inside the 

The crash occured as I was modifying the indentation around the formula for the grid tariff and feed-in tariff. Before that it woudl not work either and return only the price value and not the add on charges since the change in 0.207.5 (trying to resolve as in https://github.com/evcc-io/evcc/issues/23264)

So I was testing things around with the formula. I am not able to reset the tariff in the UI to bring it back.
Then it crahsed I cannot bring it back up.
I suppose I coudl use a back up from a few days ago... but I am not aware how to restore without the UI up.

closed 2025-08-27 by andig