evcc-io/evcc #23296: Beim Wechsel von Lademodus "Schnell" auf "Min+PV" wird nicht von 3 Phasen auf 1 Phase zurück gestellt

viper-666, 2025-08-28:
### Describe the bug

Ich habe das Problem, dass beim Wechsel vom Lademodus "Schnell" auf "Min+PV" nicht wieder zurück auf einer Phase geschaltet wird.
Statt 1x6A wird weiterhin mit 3x6A weiter geladen. Nach meinem Verständnis von "Min" müsste auf 1x6A zurück geschaltet werden.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

# mqtt message broker
mqtt:
  broker: 192.168.178.xxx:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  clientid: evcc
  user: xxx
  password: xxx

log: debug
levels:
  cache: error
  # unique installation id
plant: xxx

interval: 30s # control cycle interval

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.2753 # EUR/kWh
  feedin:
    type: fixed
    price: 0.082 # EUR/kWh

meters:
- type: template
  template: victron-energy
  usage: grid
  host: 192.168.178.9xx
  port: 502
  minsoc: 25
  maxsoc: 80
  name: grid1
- type: template
  template: victron-energy
  usage: pv
  host: 192.168.178.xxx
  port: 502
  minsoc: 25
  maxsoc: 80
  name: pv2
- type: template
  template: victron-energy
  usage: battery
    host: 192.168.178.xxx
  port: 502
  minsoc: 25
  maxsoc: 80
[... 163 lines cut ...]
[sungrow] TRACE 2025/08/28 16:57:13 modbus: recv 87 89 00 00 00 07 f8 04 04 10 55 00 00
[lp-1 ] DEBUG 2025/08/28 16:57:13 charge power: 4181W
[sungrow] TRACE 2025/08/28 16:57:13 modbus: send 87 8a 00 00 00 06 f8 04 53 36 00 01
[sungrow] TRACE 2025/08/28 16:57:13 modbus: recv 87 8a 00 00 00 05 f8 04 02 00 3b
[sungrow] TRACE 2025/08/28 16:57:13 modbus: send 87 8b 00 00 00 06 f8 04 53 38 00 01
[sungrow] TRACE 2025/08/28 16:57:13 modbus: recv 87 8b 00 00 00 05 f8 04 02 00 3c
[sungrow] TRACE 2025/08/28 16:57:13 modbus: send 87 8c 00 00 00 06 f8 04 53 3a 00 01
[sungrow] TRACE 2025/08/28 16:57:13 modbus: recv 87 8c 00 00 00 05 f8 04 02 00 3b
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.4

mfuchs1984, 2025-08-28:
Behebe das mal und probier's nochmal. Der Loadpoint muss natürlich auf automatische Umschaltung eingestellt sein.

viper-666, 2025-08-28:
So ich habe evcc auf Version 0.207.5 aktualisiert und dabei dann auch neu gestartet.
Der Fehler bleibt leider, siehe LOG:

````
[lp-1 ] INFO 2025/08/28 17:38:31 start charging ->
[lp-1 ] DEBUG 2025/08/28 17:38:31 wake-up timer: stop
[lp-1 ] DEBUG 2025/08/28 17:38:31 soc estimated: 51.00% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:38:31 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:38:31 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:38:31 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:38:45 charge power: 8210W
[lp-1 ] DEBUG 2025/08/28 17:38:45 charge currents: [11.8 12 11.7]A
[lp-1 ] DEBUG 2025/08/28 17:38:46 charge voltages: [232 228 232]V
[lp-1 ] DEBUG 2025/08/28 17:38:46 charge total import: 199.054kWh
[lp-1 ] DEBUG 2025/08/28 17:38:46 charger status: C
[lp-1 ] DEBUG 2025/08/28 17:38:46 soc estimated: 51.00% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:38:46 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:38:46 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:38:46 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:39:00 charge power: 8215W
[lp-1 ] DEBUG 2025/08/28 17:39:00 charge currents: [11.9 12 11.7]A
[lp-1 ] DEBUG 2025/08/28 17:39:01 charge voltages: [232 229 232]V
[lp-1 ] DEBUG 2025/08/28 17:39:01 charge total import: 199.054kWh
[lp-1 ] DEBUG 2025/08/28 17:39:01 charger status: C
[lp-1 ] DEBUG 2025/08/28 17:39:01 soc estimated: 51.00% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:39:01 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:39:01 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:39:01 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:39:01 set charge mode: minpv
[lp-1 ] DEBUG 2025/08/28 17:39:01 charge power: 8209W
[lp-1 ] DEBUG 2025/08/28 17:39:02 charge currents: [11.8 11.9 11.7]A
[lp-1 ] DEBUG 2025/08/28 17:39:02 charge voltages: [231 229 232]V
[lp-1 ] DEBUG 2025/08/28 17:39:02 charge total import: 199.054kWh
[lp-1 ] DEBUG 2025/08/28 17:39:02 charger status: C
[lp-1 ] DEBUG 2025/08/28 17:39:02 soc estimated: 51.00% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:39:02 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:39:02 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:39:02 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:39:02 pv charge current: min 6A > 0A (8790W @ 3p, battery: false)
[lp-1 ] DEBUG 2025/08/28 17:39:02 set charge current limit: 6A
[lp-1 ] DEBUG 2025/08/28 17:39:15 charge power: 4170W
[lp-1 ] DEBUG 2025/08/28 17:39:15 charge currents: [5.9 5.9 6]A
[lp-1 ] DEBUG 2025/08/28 17:39:16 charge voltages: [235 231 235]V
[lp-1 ] DEBUG 2025/08/28 17:39:16 charge total import: 199.054kWh
[lp-1 ] DEBUG 2025/08/28 17:39:16 charger status: C
[lp-1 ] DEBUG 2025/08/28 17:39:16 soc estimated: 51.00% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:39:16 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:39:16 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:39:16 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:39:16 pv charge current: min 6A > 0A (4638W @ 3p, battery: false)
[lp-1 ] DEBUG 2025/08/28 17:39:30 charge power: 4179W
[lp-1 ] DEBUG 2025/08/28 17:39:30 charge currents: [5.9 5.9 6]A
[lp-1 ] DEBUG 2025/08/28 17:39:31 charge voltages: [235 232 234]V
[lp-1 ] DEBUG 2025/08/28 17:39:31 detected active phases: 3p
[lp-1 ] DEBUG 2025/08/28 17:39:31 session energy: 0.093kWh
[lp-1 ] DEBUG 2025/08/28 17:39:31 charge total import: 199.147kWh
[lp-1 ] DEBUG 2025/08/28 17:39:31 charger status: C
[lp-1 ] DEBUG 2025/08/28 17:39:31 soc estimated: 51.09% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:39:31 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:39:31 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:39:31 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:39:31 available power -456W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/08/28 17:39:31 start phase scale1p timer
[lp-1 ] DEBUG 2025/08/28 17:39:31 phase scale1p in 3m0s
[lp-1 ] DEBUG 2025/08/28 17:39:31 pv charge current: min 6A > 0A (4635W @ 3p, battery: false)
[lp-1 ] DEBUG 2025/08/28 17:39:45 charge power: 4173W
[lp-1 ] DEBUG 2025/08/28 17:39:45 charge currents: [5.9 5.9 6]A
[lp-1 ] DEBUG 2025/08/28 17:39:46 charge voltages: [235 232 234]V
[lp-1 ] DEBUG 2025/08/28 17:39:46 detected active phases: 3p
[lp-1 ] DEBUG 2025/08/28 17:39:46 charge total import: 199.147kWh
[lp-1 ] DEBUG 2025/08/28 17:39:46 charger status: C
[lp-1 ] DEBUG 2025/08/28 17:39:46 soc estimated: 51.09% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:39:46 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:39:46 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:39:46 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:39:46 available power -478W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/08/28 17:39:46 phase scale1p in 2m45s
[lp-1 ] DEBUG 2025/08/28 17:39:46 pv charge current: min 6A > 0A (4651W @ 3p, battery: false)
[lp-1 ] DEBUG 2025/08/28 17:40:00 charge power: 4166W
[lp-1 ] DEBUG 2025/08/28 17:40:00 charge currents: [5.9 5.9 6]A
[lp-1 ] DEBUG 2025/08/28 17:40:01 charge voltages: [234 233 234]V
[lp-1 ] DEBUG 2025/08/28 17:40:01 detected active phases: 3p
[lp-1 ] DEBUG 2025/08/28 17:40:01 charge total import: 199.147kWh
[lp-1 ] DEBUG 2025/08/28 17:40:01 charger status: C
[lp-1 ] DEBUG 2025/08/28 17:40:01 soc estimated: 51.09% (vehicle: 51.00%)
[lp-1 ] DEBUG 2025/08/28 17:40:01 vehicle soc: 51%
[lp-1 ] DEBUG 2025/08/28 17:40:01 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/08/28 17:40:01 vehicle range: 199km
[lp-1 ] DEBUG 2025/08/28 17:40:01 available power -481W < 4140W min 3p threshold
[lp-1 ] DEBUG 2025/08/28 17:40:01 phase scale1p in 2m30s
[lp-1 ] DEBUG 2025/08/28 17:40:01 pv charge current: min 6A > 0A (4647W @ 3p, battery: false)
````

closed 2025-08-28 by viper-666