evcc-io/evcc #23297: Store the used RFID from charger

enboer, 2025-08-28:
To unlock the charger I use an RFID tag, I am using the Peblar charger. I would like to have the RFID number stored in the charging sessions table. I assume the field ID can be used for that?
I do not now if this will be charger specific but for the Peblar charge I know the commands to get the RFID of the last session which is the session currently running.

As a workaround I add this RFID to the database when EVCC starts charging, but it seems that when charging is finished it rewrites the whole chargesession record? at least the value stored when the charging started is gone.

enboer, 2025-09-03:
The charges is not connected to any OCPP. I will ask Peblar if there is another solution.
Are other charger integrations able to get this number into the EVCC database directly?
For the moment I am running this script when the car gets disconnected and put it in the EVCC database.

premultiply, 2025-09-03:
It should be able to connect via OCPP protocol. RFID support on OCPP is mandatory. So that would be a simple and quick solution. Did you try connecting the charger to evcc using OCPP?

Other solution would be the vendor to fix the firmware.
RFID tag readout support should be added to the Modbus API.

enboer, 2025-09-08:
Currently both options are not available. Getting RFID via modbus is planned, but when is unknown. Adding EVCC as a custom OCPP is not that easy. As a customer you can not add via the GUI. By firmware update there is a separate file with all the supported OCPP providers.

premultiply, 2025-09-10:
If you do not want to wait or modify the list of OCPP providers the whole charger implementation needs to be switched to die REST API https://developer.peblar.com/local-rest-api