evcc-io/evcc #23300: Growatt battery control not working

baraoresmungao, 2025-08-28:
### Describe the bug

I am using a Growatt TL-X(H) Hybrid Inverter and battery and I am connecting to EVCC thru aMODBUS adaptor. All seams good, I can see the battery levels and the power exported from the battery correctly.

I want to keep the battery for my home consumption and not use it for car charging.
I configured all to the experimental GUI.

Also I defined the following, so it should never allow for car charging thru battery.

[image]

The problem is when charging starts the system gives periodically the error:

```
[site ] ERROR 2025/08/29 00:16:02 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:05 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:11 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:16 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:23 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:27 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:33 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:40 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:45 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:51 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:57 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:16:59 battery mode: modbus: exception '0' (unknown), function '16'
[site ] ERROR 2025/08/29 00:17:03 battery mode: modbus: exception '0' (unknown), function '16'
```

### Steps to reproduce

1. Start charging the car
2. Config is thru experimental GUI

[image]

3. Config UI as follows

[image]

[image]

[image]

### Configuration details

```yaml
I am using EVCC thru Home Assistant add-on so I cannot (or I don't know how) to export the Config UI.
```

### Log details

```text
[site ] DEBUG 2025/08/29 00:16:23 grid power: 6535W
[site ] DEBUG 2025/08/29 00:16:23 grid powers: [2201 2260 2075]W
[site ] DEBUG 2025/08/29 00:16:23 grid currents: [9.23 9.6 8.69]A
[httpd ] TRACE 2025/08/29 00:16:23 GET /api/config/devices/charger/db:1/status
[etrel ] TRACE 2025/08/29 00:16:23 modbus: send 00 58 00 00 00 06 01 04 00 1a 00 02
[etrel ] TRACE 2025/08/29 00:16:23 modbus: recv 00 58 00 00 00 07 01 04 04 41 25 e3 54
[etrel ] TRACE 2025/08/29 00:16:23 modbus: send 00 59 00 00 00 06 01 04 00 0e 00 06
[etrel ] TRACE 2025/08/29 00:16:23 modbus: recv 00 59 00 00 00 0f 01 04 0c 41 72 2d 0e 41 6d 0e 56 41 6e 14 7b
[etrel ] TRACE 2025/08/29 00:16:23 modbus: send 00 5a 00 00 00 06 01 04 00 00 00 01
[etrel ] TRACE 2025/08/29 00:16:23 modbus: recv 00 5a 00 00 00 05 01 04 02 00 04
[etrel ] TRACE 2025/08/29 00:16:23 modbus: send 00 5b 00 00 00 06 01 04 00 04 00 02
[db:11 ] TRACE 2025/08/29 00:16:23 modbus: send 00 bd 00 00 00 06 01 04 0b bd 00 02
[etrel ] TRACE 2025/08/29 00:16:23 modbus: recv 00 5b 00 00 00 07 01 04 04 41 80 00 00
[httpd ] TRACE 2025/08/29 00:16:23 GET /api/config/devices/charger/db:6/status
[db:11 ] TRACE 2025/08/29 00:16:23 modbus: recv 00 bd 00 00 00 07 01 04 04 00 00 00 00
[db:11 ] TRACE 2025/08/29 00:16:23 modbus: send 00 be 00 00 00 06 01 04 0b c1 00 02
[db:11 ] TRACE 2025/08/29 00:16:23 modbus: recv 00 be 00 00 00 07 01 04 04 00 00 00 00
[... 374 lines cut ...]
[shelly] TRACE 2025/08/29 00:16:48 POST http://192.168.1.51/rpc/Switch.GetStatus
[shelly] TRACE 2025/08/29 00:16:48 {"id":0,"src":"evcc","method":"Switch.GetStatus"} -- {"id":0, "source":"WS_in", "output":true, "apower":0.0, "voltage":239.8, "freq":50.0, "current":0.000, "aenergy":{"total":53901.000,"by_minute":[0.000,0.000,0.000],"minute_ts":1756422960}, "ret_aenergy":{"total":0.000,"by_minute":[0.000,0.000,0.000],"minute_ts":1756422960},"temperature":{"tC":45.9, "tF":114.5}}
[httpd ] TRACE 2025/08/29 00:16:48 GET /api/config/products/meter
[httpd ] TRACE 2025/08/29 00:16:48 GET /api/config/devices/meter/14
[httpd ] TRACE 2025/08/29 00:16:48 GET /api/config/templates/meter
[site ] DEBUG 2025/08/29 00:16:51 ----
[shelly] TRACE 2025/08/29 00:16:51 POST http://192.168.1.51/rpc/Switch.GetStatus
[etrel ] TRACE 2025/08/29 00:16:51 modbus: send 00 74 00 00 00 06 01 04 00 1a 00 02
[etrel ] TRACE 2025/08/29 00:16:51 modbus: recv 00 74 00 00 00 07 01 04 04 41 25 e3 54
[lp-1 ] DEBUG 2025/08/29 00:16:51 charge power: 10368W
[etrel ] TRACE 2025/08/29 00:16:51 modbus: send 00 75 00 00 00 06 01 04 00 0e 00 06
[etrel ] TRACE 2025/08/29 00:16:51 modbus: recv 00 75 00 00 00 0f 01 04 0c 41 73 33 33 41 6c cc cd 41 6e 14 7b
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

baraoresmungao, 2025-08-30:
Will try it during the weekend. I am having troubles running Modbus-cli, it is giving me errors.

closed 2025-09-18 by github-actions[bot]

baraoresmungao, 2025-09-18:
Nobody can help me on this?

sanger-sketch, 2026-09-25:
Confirmed working on real hardware: Growatt MOD 10KTL3-XH (BP) with a Growatt APX 5.0P-B1 battery, connected via a USR-TCP232-410S RS485-to-TCP gateway (set to "Modbus TCP" mode) on RS485A1/B1 (pins 3/4 of the comm connector).

Manually verified with mbpoll before configuring evcc:
- Standalone FC16 write to 1100/1101/1102 (the `growatt-hybrid` registers) → rejected with modbus exception 0, as expected for this inverter family
- Atomic FC16 write to 3038+3039 as a single uint32 (0xA000173B) + separate write to 3049 → succeeded, read-back confirmed correct values, no exception

This confirms the fix in this PR works as intended on MOD-XH hardware.