evcc-io/evcc #23306: Manual State of Charge Selection for Residential Battery When Charging from the Grid

marknisk, 2025-08-29:
**Is your feature request related to a problem? Please describe.**
Currently, the state of charge (SOC) target for the residential battery when charging from the grid cannot be set manually (if current price driver condition is met, battery will charge to 100%). This limits the value of this feature especially when planning for forecasted grid conditions e.g. I need 10kWh charge in the night to survive expensive early hours of the morning when there isn't enought PV production. However, I don't want to chage to 100% since there will be lots of PV production during the day, and the price will be low (making selling the surplus not lucrative).

**Describe the solution you'd like**
Add a pull-down menu or similar UI element in the EVCC application that allows users to manually select the desired SOC target for their residential battery when charging from the grid. Other limits would stay the same (e.g. starting whent the electricity price goes below the set treshold).

**Describe alternatives you’ve considered**
Setting SOC target manually (modifying the values in the inververter), but this is complicated, less transparent and does not allow for quick, manual overrides when needed.

Relying solely on automatic logic, but this removes essential user control for specific scenarios and is complicated.

**Additional context**
This feature would help users better manage battery longevity and adapt to dynamic electricity pricing, maintenance schedules, or personal needs.

marknisk, 2025-08-29:
Thanks for the comment. I took a look on #17424, but it was closed and I don't know how to add this to anywhere. I'm not that good with GitHub either so it could be a bigger competence issue on my side.

marknisk, 2025-08-29:
Sorry, my bad: "locked and limited to the contributors". However, I don't know how to add it over there so I will check the optimiser next and maybe it will deliver what I'm looking for. Thanks for reading the request.

andig, 2025-08-29:
Ah, sorry. Issue is just for visibility, please use the discussion linked.