evcc-io/evcc #23309: evcc startet nicht

man10to, 2025-08-29:
### Describe the bug

Hallo,
seit heute startet EVCC nicht mit der folgenden Meldung

```
[lp-3 ] ERROR 2025/08/29 13:20:40 charge total import: Get "http://192.168.78.134/status": dial tcp 192.168.78.134:80: connect: no route to host
[main ] FATAL 2025/08/29 13:20:40 loadpoint [db:9] charger does not support phase switching
```

ist es wirklich so das bei einer falschen einstellung eines ladepunktes kein Zugriff merh möglich ist ?

wie finde ich raus an welchen dateien ich das beheben kann

mfg
Egbert

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
egbert@iobroker:~$ cat \etc\evcc.yaml
cat: etcevcc.yaml: No such file or directory
egbert@iobroker:~$ cat /etc/evcc.yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: d61656b264b9453d99f52b759600ec3e5a6bd4224b10ed151b26351bb9b9b5ca

interval: 10s # control cycle interval

sponsortoken: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJleHAiOjE3ODUwMDI5NTcsImlhdCI6MTY5MDM5NDk1NywiaXNzIjoiZXZjYy5pbyIsInN1YiI6Im1hbjEwdG8ifQ.amCX9PP17PLYRrUfvCqsBQ30uCwx6KaGCrr1EI5iZvw

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:

chargers:

vehicles:

loadpoints:

site:
  title: Wesseln
  meters:
    grid:
    pv:

    battery:
```

### Log details

```text
[main ] INFO 2025/08/29 13:20:19 evcc 0.207.5
[main ] INFO 2025/08/29 13:20:19 using config file: /etc/evcc.yaml
[db ] INFO 2025/08/29 13:20:19 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/08/29 13:20:19 UI listening at :7070
[main ] ERROR 2025/08/29 13:20:30 creating vehicle db:17 failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: Post "https://idpconnect-eu.kia.com/auth/account/signin": dial tcp: lookup idpconnect-eu.kia.com: i/o timeout
[main ] ERROR 2025/08/29 13:20:30 creating vehicle db:16 failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: Post "https://idpconnect-eu.kia.com/auth/account/signin": dial tcp: lookup idpconnect-eu.kia.com: i/o timeout
[main ] ERROR 2025/08/29 13:20:34 creating vehicle db:18 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': Get "https://identity.vwgroup.io/oidc/v1/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&scope=openid+profile+badge+cars+dealers+vin&response_type=code+id_token+token&redirect_uri=weconnect%3A%2F%2Fauthenticated&state=97d17c37b3b9bfbce5ca52c39a9f693887cb6e66faeb76b4d4b4e7cc77b3bf25&nonce=EQPzHCyQCyecutIexbnCzFwWgQXbcNBoswwtPfckTNt": dial tcp: lookup identity.vwgroup.io: i/o timeout
[lp-1 ] DEBUG 2025/08/29 13:20:34 set title: goe136
[lp-1 ] DEBUG 2025/08/29 13:20:34 set priority: 0
[lp-1 ] DEBUG 2025/08/29 13:20:34 set smart cost limit: <nil>
[lp-1 ] DEBUG 2025/08/29 13:20:34 set smart feed-in limit: <nil>
[lp-1 ] DEBUG 2025/08/29 13:20:34 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-1 ] DEBUG 2025/08/29 13:20:34 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1 ] DEBUG 2025/08/29 13:20:34 set session energy limit: 0
[lp-1 ] DEBUG 2025/08/29 13:20:34 set session soc limit: 0
[lp-1 ] DEBUG 2025/08/29 13:20:34 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000cd940c}
[lp-1 ] DEBUG 2025/08/29 13:20:34 set default mode: pv
[lp-1 ] DEBUG 2025/08/29 13:20:34 set phases: 0
[lp-1 ] DEBUG 2025/08/29 13:20:34 set max current: 14
[lp-2 ] DEBUG 2025/08/29 13:20:34 set title: goe135
[lp-2 ] DEBUG 2025/08/29 13:20:34 set priority: 0
[lp-2 ] DEBUG 2025/08/29 13:20:34 set smart cost limit: <nil>
[lp-2 ] DEBUG 2025/08/29 13:20:34 set smart feed-in limit: <nil>
[lp-2 ] DEBUG 2025/08/29 13:20:34 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-2 ] DEBUG 2025/08/29 13:20:34 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2 ] DEBUG 2025/08/29 13:20:34 set session energy limit: 0
[lp-2 ] DEBUG 2025/08/29 13:20:34 set session soc limit: 0
[lp-2 ] DEBUG 2025/08/29 13:20:34 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000db5bec}
[lp-2 ] DEBUG 2025/08/29 13:20:34 set default mode: pv
[lp-2 ] DEBUG 2025/08/29 13:20:34 set phases: 0
[lp-2 ] DEBUG 2025/08/29 13:20:34 set min current: 6
[lp-2 ] DEBUG 2025/08/29 13:20:34 set max current: 13
[lp-3 ] DEBUG 2025/08/29 13:20:37 set title: goe134
[lp-3 ] DEBUG 2025/08/29 13:20:37 set priority: 0
[lp-3 ] DEBUG 2025/08/29 13:20:37 set smart cost limit: <nil>
[lp-3 ] DEBUG 2025/08/29 13:20:37 set smart feed-in limit: <nil>
[lp-3 ] DEBUG 2025/08/29 13:20:37 set thresholds: {Enable:{Delay:1m0s Threshold:0} Disable:{Delay:3m0s Threshold:0}}
[lp-3 ] DEBUG 2025/08/29 13:20:37 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-3 ] DEBUG 2025/08/29 13:20:37 set session energy limit: 0
[lp-3 ] DEBUG 2025/08/29 13:20:37 set session soc limit: 0
[lp-3 ] DEBUG 2025/08/29 13:20:37 set soc config: {Poll:{Mode:charging Interval:1h0m0s} Estimate:0xc000b78ffc}
[lp-3 ] DEBUG 2025/08/29 13:20:37 set default mode: pv
[lp-1 ] DEBUG 2025/08/29 13:20:37 charge total import: 5462.687kWh
[lp-2 ] DEBUG 2025/08/29 13:20:37 charge total import: 6798.180kWh
[lp-3 ] ERROR 2025/08/29 13:20:40 charge total import: Get "http://192.168.78.134/status": dial tcp 192.168.78.134:80: connect: no route to host
[main ] FATAL 2025/08/29 13:20:40 loadpoint [db:9] charger does not support phase switching
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.5

andig, 2025-08-29:
@naltatis der echte Fehler ist:

> loadpoint [db:9] charger does not support phase switching

das sieht aus wie https://github.com/evcc-io/evcc/pull/23170

man10to, 2025-08-29:
ok

ButtiBBQ, 2025-08-29:
Kurzes Update:

Ich hatte heute genau die gleiche Meldung. Es stellte sich heraus, das der FI raus war und die Wallbox somit offline (stromlos)

=> also in meinem Fall ist die Meldung irreführend. 
Anstelle der Meldung, loadpoint [db:9] charger does not support phase switching"
sollte die Korrekte Meldung
**"Wallbox nicht erreichbar Meldung kommen"**

andig, 2025-08-29:
> Saubere Fehlermeldungen reduzieren den Supportaufwand ;)

Das ist ein sehr wertvoller Hinweis.

> Es stellte sich heraus, das der FI raus war und die Wallbox somit offline (stromlos)

Das hat mit der Meldung vmtl. überhaupt nichts zu tun.

Saubere Issues reduzieren den Supportaufwand- hättest Du dazu ein Trace log? Mit der Information lässt sich so leider nix anfangen.