evcc-io/evcc #23312: Experimental UI configuration - Home Assistant Switch Error 404

Bloodhoundje, 2025-08-29:
### Describe the bug

When trying to add a home assistant switch through the experimental UI configuration it keeps failing to validate with the mesasage: chargeStatus: unexpected status: 404 (Not Found)

This could normally be either a missing '/' after api in the url or a missing entity in Homeassistant.
I can do an API call through curl to retrieve the information so it seems isolated to the new UI configuration.

All other non-homeassistant integrations are running flawlessly (Cupra Born connect / Solar panels / P1 Smart Meter)

### Steps to reproduce

EVCC configuration:
Add charger -> Home Assistant Switch
Settings:
[Base URL:](http://192.168.1.108:8123/) http://192.168.1.108:8123/ or http://127.0.0.1:8123/ (I run EVCC on my Home assistance instance).
[Long-Lived Access Token:](http://192.168.1.108:8123/) Generated through HomeAssistant.
**Entity ID of the switch device:** switch.op_ev_lader_smartsocket
**Standby power:** 15

In Home Assistant add a new device with device entity name "op_ev_lader_smartsocket"

### Configuration details

```yaml
➜  ~ docker exec -it addon_49686a9f_evcc /bin/sh
/app # evcc config
[main  ] INFO 2025/08/29 16:52:51 config file not found, database-only mode
[db    ] INFO 2025/08/29 16:52:51 using sqlite database: /root/.evcc/evcc.db
/app # evcc dump --cfg
[main  ] INFO 2025/08/29 16:53:14 config file not found, database-only mode
[db    ] INFO 2025/08/29 16:53:14 using sqlite database: /root/.evcc/evcc.db

Home Assistant log output:
[lp-1  ] ERROR 2025/08/29 16:52:49 charger enabled: unexpected status: 404 (Not Found)
[lp-1  ] ERROR 2025/08/29 16:52:57 charge power: unexpected status: 404 (Not Found)
```

### Log details

```text
Output from HomeAssistant when testing the API endpoint:

curl -X GET -H "Authorization: Bearer TYPETOKENHERE" -H "Content-Type: application/json" http://127.0.0.1:8123/api/states/switch.op_ev_lader_smartsocket
{"entity_id":"switch.op_ev_lader_smartsocket","state":"off","attributes":{"friendly_name":"OP_EV_Lader_SmartSocket"},"last_changed":"2025-08-29T14:40:07.337652+00:00","last_reported":"2025-08-29T14:40:07.337652+00:00","last_updated":"2025-08-29T14:40:07.337652+00:00","context":{"id":"01K3V4B0N9XHA3RPMSHPCNB06Z","parent_id":null,"user_id":null}}#
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.5

andig, 2025-08-29:
Needs a trace log to see the actual evcc request

Bloodhoundje, 2025-08-29:
> > [main ] FATAL 2025/08/29 18:01:39 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
> 
> We still need a trace log of the error

I have the tracelog now and also have seem to found the possible issue in the log!:

```
[lp-1 ] DEBUG 2025/08/29 20:48:15 charger status: A
[ha-switch] TRACE 2025/08/29 20:48:15 GET http://192.168.1.108:8123//api/states/switch.op_ev_lader_smartsocket
[ha-switch] TRACE 2025/08/29 20:48:15 404: Not Found
[lp-1 ] ERROR 2025/08/29 20:48:15 charger enabled: unexpected status: 404 (Not Found)
[site ] DEBUG 2025/08/29 20:48:15 ----
[ha-switch] TRACE 2025/08/29 20:48:15 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:15 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:15 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:16 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:16 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:16 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:16 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:16 404: Not Found
[providerauth] TRACE 2025/08/29 20:48:16 publishing 0 auth providers
[ha-switch] TRACE 2025/08/29 20:48:16 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:16 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:16 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:16 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:16 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:16 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:16 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:16 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:17 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:17 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:17 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:17 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:18 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:18 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:19 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:19 404: Not Found
[ha-switch] TRACE 2025/08/29 20:48:22 GET http://192.168.1.108:8123//api/states/
[ha-switch] TRACE 2025/08/29 20:48:22 404: Not Found
[httpd ] TRACE 2025/08/29 20:48:22 GET /api/system/log 
```

The tracelog shows "http://192.168.1.108:8123//api/states/" where there is a double '/' after the port number. This causes an invalid address. This is likely a very small change to get it fixed in the code! :-).

andig, 2025-08-30:
Dann lass doch den unnötigen / mal weg

andig, 2025-08-30:
Closed in 062ceceef64f8f15592049c4416ec5f6481a05ca

closed 2025-08-30 by andig

spuky, 2025-09-17:
@julianwachholz creating a number helper with 300w in HA might be the better solution for now

andig, 2025-09-17:
What same error? Logfile?