evcc-io/evcc #23323: evcc does not set charge power to 0W after vehicle disconnected from OCPP managed wallbox

lurtz, 2025-08-30:
### Describe the bug

I have configured a EM2GO Duo Power to connect via OCPP, because evcc cannot connect via Modbus yet (see #23317). When the car is disconnected from the wallbox, evcc notices that the car was unplugged but continues to assume that it is charged with the last reported value from the wallbox.

[image]

Then evcc gets confused because input power and output power do not match anymore.

> NOTE: evcc complains about incorrect phase wiring as well, but I hope this is not related to this issue or a non issue.
>
> ```
> EM2GO Duo Power 1/2: invalid phase wiring between charge meter and charger
> EM2GO Duo Power 2/2: invalid phase wiring between charge meter and charger
> ```

### Steps to reproduce

1. Configure EM2GO Duo Power to connect to evcc via OCPP
2. Configure evcc for both connectors
3. Charge a car and unplug it
4. Check that evcc still assumes power is put into the car even though it knows that it has been unplugged

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: deadbeef

interval: 30s # control cycle interval

sponsortoken: F00BA4

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
  - type: template
    template: sungrow-hybrid
    id: 1
    host: 192.168.1.91
    port: 502
    usage: grid
    modbus: tcpip
    timeout: 2s
    name: grid1
  - type: template
    template: sungrow-hybrid
    id: 1
    host: 192.168.1.91
    port: 502
    usage: pv
    modbus: tcpip
    timeout: 2s
    name: pv2
  - type: template
    template: sungrow-hybrid
    id: 1
    host: 192.168.1.91
    port: 502
[... 267 lines cut ...]
[lp-3  ] DEBUG 2025/08/30 12:43:42 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2025/08/30 12:43:42 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/08/30 12:43:42 charge power: 0W
[lp-1  ] DEBUG 2025/08/30 12:43:42 charge currents: [0 0 0]A
[site  ] DEBUG 2025/08/30 12:43:42 pv 2 power: 1837W
[site  ] DEBUG 2025/08/30 12:43:42 grid power: -2566W
[site  ] DEBUG 2025/08/30 12:43:42 pv 1 power: 2669W
[site  ] DEBUG 2025/08/30 12:43:42 pv power: 4506W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.207.4

mfuchs1984, 2025-08-31:
Looks like the wallbox stops sending the power measurements when the car is disconnected.

lurtz, 2025-08-31:
> Looks like the wallbox stops sending the power measurements when the car is disconnected.

The wallbox only sends power measurements when a car is connected. I can also see that in Grafana. In the middle of the image you can see the restart of the wallbox before testing the Duo Power.

[image]

lurtz, 2025-08-31:
I have hypothesis what is wrong, but I might be wrong because I am not familiar with the code.

Here power for each phase is reported. when the car is charging:

> [ocpp  ] TRACE 2025/08/31 15:31:55 recv EM2Go1479: [2,"f7a39b5d-60a0-01df-7db4-6185cc2cf6c8","MeterValues",{"connectorId":2,"transactionId":1756646767,"meterValue":[{"timestamp":"2025-08-31T13:31:54.001Z","sampledValue":[{"value":"2.363","unit":"kW","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L1-N","location":"Outlet"},{"value":"2.312","unit":"kW","context":"Sample.Periodic","format":"Raw","measurand":"Power.Active.Import","phase":"L2-N","location":"Outlet"},{"value":"89.998","unit":"Wh","context":"Sample.Periodic","format":"Raw","measurand":"Energy.Active.Import.Register","location":"Outlet"},{"value":"9.950","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L1","location":"Outlet","unit":"A"},{"value":"9.930","context":"Sample.Periodic","format":"Raw","measurand":"Current.Import","phase":"L2","location":"Outlet","unit":"A"},{"value":"237.900","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L1-N","location":"Outlet","unit":"V"},{"value":"233.400","context":"Sample.Periodic","format":"Raw","measurand":"Voltage","phase":"L2-N","location":"Outlet","unit":"V"},{"value":"6","context":"Sample.Periodic","format":"Raw","measurand":"Current.Offered","location":"Outlet","unit":"A"}]}]}]

But when the car is disconnected it is not set for each phase: https://github.com/evcc-io/evcc/blob/997f84ac180974184623f7dc712f18ba9efabdb5/charger/ocpp/connector_core.go#L111-L116

mfuchs1984, 2025-08-31:
Evcc instructs the charger to provide the power, the charger accepts the instruction, but fails to comply when no car is charging. The vast majority of charger comply and sends the power and other measurands regardless of the state of the connection to the car. 

Your charger and one or two others fail to do so and leave out some measurands when no car is connected. Your charger is then only reporting the total energy and the offered current. This should be reported to the manufacturer. 

The other question is if evcc should comment a Workaround for those chargers, e.g. relax the measurand requirements when no car is connected.

Anyways, there is a workaround. You could specify `metervalues: Energy.Active.Import.Register,Current.Offered`. Then, evcc will only use the ones that are always available from your charger. When applying this workaround, you will loose some power measurement precision.

lurtz, 2025-08-31:
> Evcc instructs the charger to provide the power, the charger accepts the instruction, but fails to comply when no car is charging. The vast majority of charger comply and sends the power and other [...]

Thank you, I will do that.

> The other question is if evcc should comment a Workaround for those chargers, e.g. relax the measurand requirements when no car is connected.
>
> Anyways, there is a workaround. You could specify [...]

I will try the workaround tomorrow (configuration already changed). Maybe I will also try to implement a workaround in evcc.

lurtz, 2025-09-01:
With

```
chargers:
  - type: template
    template: em2go
    id: 255
    host: 192.168.1.220
    port: 502
    modbus: tcpip
    name: wallbox4
  - type: template # zweite Wallbox
    template: ocpp
    stationid: EM2Go1479
    name: wallbox5
    connector: 1
    metervalues: Energy.Active.Import.Register,Current.Offered
  - type: template # zweite Wallbox
    template: ocpp
    stationid: EM2Go1479
    name: wallbox6
    connector: 2
    metervalues: Energy.Active.Import.Register,Current.Offered
```

evcc sets indeed the power to zero, after the car has been disconnected. However measurements being less precise is very accurate:

[image]

As can be seen by the battery power evcc overestimates the power consumption of the wallbox by up to 2kW (peak power consumption of the wallbox). I hope this works better, when either the manufacturer fixes the firmware or have another workaround in evcc.

lurtz, 2025-09-02:
I got positive feedback from the manufacturer that they will fix it with a firmware upgrade. But they also said that it will take some time.

lurtz, 2025-09-09:
I believe #23464 actually fixed this one. It looks like in OCPP the wallbox should send a StopTransaction, when charging stops and now evcc clears power measurements of all phases in this case. Once a nightly build is available I will test it.