evcc-io/evcc #23345: Stiebl Eltron Wärmepumpe / Fehlermeldung: power: read failed: modbus: exception '2' (illegal data address), function '4'

basiokan, 2025-08-31:
### Describe the bug

[evcc-20250831-135135-trace.log](https://github.com/user-attachments/files/22066175/evcc-20250831-135135-trace.log)

HI - ich kann zwar die Wärmepumpe hinzufügen , bekomme aber immer die rote meldung:
power: read failed: modbus: exception '2' (illegal data address), function '4'

wie bekomme ich die weg? ich kann die WP auch daher nicht über evcc steuern

[image]

danke euch!

### Steps to reproduce

1.Heizung hinzfügen
2.Hersteller: Stiebel Eltron WPM
3. IP Adresse der WP eingeben
4. auf prüfen > es kommt die Fehlermeldung
5. speichern geht zwar
...

### Configuration details

```yaml
## minimal configuration example

site:
  title: Home # display name for UI
loadpoints:
#  - title: Garage # display name for UI
#    charger: my_charger # charger
#    vehicle: my_car # default vehicle
    
    
    # replace with your real charger
# see https://docs.evcc.io/docs/devices/chargers
#chargers:
#  - name: my_charger
#    type: template
#    template: demo-charger
#    status: C # charging
#    power: 1 # 2 kW charging power
#    enabled: true # optional

# replace with your real vehicle (optional)
# see https://docs.evcc.io/docs/devices/vehicles
#vehicles:
#  - name: my_car
#    type: template
#    template: offline
#    title: blue e-Golf
#    capacity: 0.001 # in kWh

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints

# enter your real grid tariff and feed-in price
# see https://docs.evcc.io/docs/tariffs
#tariffs:
#  currency: EUR
#  grid:
#    type: fixed
#    price: 0.1839 # EUR/kWh
#  feedin:
#    type: fixed
#    price: 0.08 # EUR/kWh
```

### Log details

```text
siehe anhang oben
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.5

Vogelbecker, 2025-09-01:
Hi, 

welche WP hast du genau? 
Ist Modbus im ISG aktiviert und welche Firmware hat dein ISG?

basiokan, 2025-09-01:
> Hi,
> 
> welche WP hast du genau? Ist Modbus im ISG aktiviert und welche Firmware hat dein ISG?

Hi - 
- Wärempumpe: Stiebel WPL 20a
- ISG Version: v12.2.3
- Modus ist aktiviert, siehe Bild

[image]
[image]
[image]

gruß

basiokan, 2025-09-03:
hi @Vogelbecker  - danke schon mal für Deinen Support
- hast du nen Tipp fürs Register auslesen? mir steht ein Macbook zur Verfügung und zugriff via LAN auf das ISG....
google hat mir zu thema modbus register auslesen nur krass komplizierte Wege aufgezeigt :/

andig, 2025-09-03:
brew install mbpoll

Vogelbecker, 2025-09-03:
probiere mal folgendes:
` mbpoll 192.168.50.170 -t 3 -r 518,522 -1
`
das sollte ISTTEMPERATUR und PUFFERISTTEMPERATUR anzeigen

für den Power-Wert könnte es dann so sein, falls er auslesbar ist: 
` mbpoll 192.168.50.170 -t 3 -r 680 -1
`
(könnest statt der 680 auch nochmal 679 und 681 probieren)

Ansonsten nochmal mit ` -v ` aufrufen um besser debuggen zu können

Vogelbecker, 2025-09-03:
Dann bitte mal ein `mbpoll 192.168.50.170 -t 3 -r 518,522 -1 -v` dann ist er etwas gesprächiger

basiokan, 2025-09-03:
selbes :( 
```
 @MacBookAir ~ % mbpoll 192.168.50.170 -t 3 -r 518,522 -1 -v
mbpoll: Illegal data value: -t ! Try -h for help.
 @MacBookAir ~ % mbpoll 192.168.50.170 -t 3 -r 518,522 -1 -v
mbpoll: Illegal data value: -t ! Try -h for help.
```

Vogelbecker, 2025-09-03:
Das sieht doch schon ganz gut aus.
Dein Pufferspeicher hat 25,3° und dein Warmwasser 45,4°
Das heißt diese beiden Register gibt es und sie sind auch auslesbar.

Für mich sieht es aus aus, als würde es das Power-Register bei dir nicht geben.... 

@andig sollten wir ggf https://github.com/evcc-io/evcc/pull/18505 reverten? Das scheint ja eh nur ein Schätzwert zu sein...

freakezoid1604, 2025-09-04:
Ich hab ne WPL-A 07.2 Plus HK 230

Hab nun auch mal nach den Werten geschaut.
Auch bei mir kommen Werte allerdings ab 600 -699 nichts

[321798-44755-9770_ISG Modbus_de_en_fr_it_nl_cs_sk_pl_hu.pdf](https://github.com/user-attachments/files/22130066/321798-44755-9770_ISG.Modbus_de_en_fr_it_nl_cs_sk_pl_hu.pdf)

[image]

[image]

freakezoid1604, 2025-09-07:
[image]

Es gibt eine Inverterleistung. Aber nicht auf Parameter 680

andig, 2025-09-07:
Solche Schnittstellen sind toll. Inverter Aufnahmeleistung und Aufnahmeleistung Inverter, aber mit unterschiedlichen Werten. Kopfschüttel...

Tombra1889, 2025-09-07:
Hab es nochmal überprüft. Es scheint mit 

[image]

zu funktionieren. also 3680

freakezoid1604, 2025-09-07:
könnte hinhauen... ist aber nicht in der Doku
[image]

basiokan, 2025-09-08:
Danke! > das werde ich mal testen

mit dem heutigen Update von evcc ist ja der FIX hier eingespielt worden > nun habe ihc nicht meh den fehler: ower: read failed: modbus: exception '2' (illegal data address), function '4

dafür habe ich nun diesen fehler:

[image]

und die pumpe verbraucht quasi dauerhaft 3kw laut evcc?

[image]

andig, 2025-09-09:
Vorschlag: Leistung fliegt raus, wer das braucht kann einen Shelly nutzen.

Vogelbecker, 2025-09-09:
Oder kann man es recht einfach Konfigurierbar machen? 
Checkbox: Leistung auslesen ja/nein?

moku01, 2025-11-01:
Heyy- Meine WPL-A 07 zeigt in EVCC einen Tagesverbrauch von 9,4 kWh an, in der Servicewelt werden allerdings nur rund 4 kWh für Heizen und WW für die letzten 24h geführt. Außerdem werden für den Oktober 850 kWh als verbrauch angegeben, das ist zu viel. Wenn die Wp läuft werden in EVCC Verbrauchswerte angezeigt die den Gesamtverbrauch des Hauses + WP deutlich übersteigen. Verfälscht das den Gesamt-Sonnenanteil in und damit auch die Ladekosten des Autos ? Wenn ja, kann man den EVCC Energiesensor deaktivieren ?

moku01, 2025-11-02:
> Vllt. falsches Register/ falscher Messwert im Template (thermische Energie statt Primärseite)?
Ich habe die WP über die UI und nicht per YAML eingebunden. Dort konnte ich nichts anders konfigurieren :/

andig, 2025-11-02:
Der Fehler mag trotzdem im Template sein. Muss sich jemand mit Gerät anschauen und ausprobieren.

mp1981, 2025-11-03:
> Heyy- Meine WPL-A 07 zeigt in EVCC einen Tagesverbrauch von 9,4 kWh an, in der Servicewelt werden allerdings nur rund 4 kWh für Heizen und WW für die letzten 24h geführt. Außerdem werden für den [...]

In der Servicewelt wird nur der Verbrauch des Verdichters angezeigt. Alles andere wie Pumpen, Sumpfheizung usw. wird da nicht berücksichtigt. Je nach Konfiguration kann der Unterschied erheblich sein...

moku01, 2025-11-04:
> Am einfachsten wäre wohl, wie bei fast allen WP am Markt, mit einem externen Zähler die tatsächliche Gesamtleistung + elektrische Primärenergie des Heizsystems zu messen und ggf. noch mittels `sgready-relay` die passenden Steuereingänge zu bedienen? Das spezielle Template dann entfernen.
> 
> So eine WP-Heizung ist kein Wunderwerk der Technik. Strom rein, hinten nach Bedarf und Heizkurve Wärme raus.
> 
> Selbst wenn einzelne Spezialmesswerte wie diverse Temperaturen, Drücke und Teilleistungen zur Systemdiagnose interessant sein mögen, für das Energiemanagement sind völlig irrelevant und ggf. sogar irreführend. Die Anzeige der Verdichterleistung ist so relevant wie beim Auto die Anzeige der Lüfterdrehzahl.

Jup, der Stromverbrauch ist mir eigentlich auch egal. Ich möchte nur nicht das die WP die EVCC Werte zur Sonnenstromnutzung / Kosten des Autos verfälschen. Wegen mir kann die Leistung da komplett raus.

blindefliegenhasser, 2026-08-31:
Anfänger hier: Stiebel hat ne TCP/IP Modbus Dokumentation. Da ist das komplette Register (u.a. auch die 1 based info etc.) und alle wichtigen Infos zur Einrichtung notiert. Ist ne PDF, die ich vom TOP Kundenservice bekommen hatte. So geilen und schnelle Service inklusive Rückruf der Fachabteilung hatte ich noch nie. Läuft damit bei mir als Sensoren in HAOS.

basiokan, 2026-08-31:
> Anfänger hier: Stiebel hat ne TCP/IP Modbus Dokumentation. Da ist das komplette Register (u.a. auch die 1 based info etc.) und alle wichtigen Infos zur Einrichtung notiert. Ist ne PDF, die ich vom [...]

Hi dann poste doch mal bitte die PDF
Danke dir

andig, 2026-09-01:
Oder einfach die Erkenntnis teilen ob das Register hier nicht erwartungsgemäß funktionieren müsste…

blindefliegenhasser, 2026-09-01:
[202601_Kompatibilitaetsliste_ISG_STE.pdf](https://github.com/user-attachments/files/31679690/202601_Kompatibilitaetsliste_ISG_STE.pdf)

[321798-47824-0146_ISG Modbus_de_en_fr_it_nl_cs_sk_pl_hu.pdf](https://github.com/user-attachments/files/31679694/321798-47824-0146_ISG.Modbus_de_en_fr_it_nl_cs_sk_pl_hu.pdf)

[kompatibilitaetsliste-isg.pdf](https://github.com/user-attachments/files/31679938/kompatibilitaetsliste-isg.pdf)

Sorry Leute, ich verstehe nur die Hälfte von dem, was in dem Thread geschrieben wurde. War mir auch nicht sicher, ob ich das hochladen darf. Anbei die beiden Dokumente, die mir geholfen haben, die config so zu schreiben, dass ich den Modbus auslesen/ansteuern(?) konnte. Da dort auch viel Grundlegendes (für mich) drin stand, dachte ich mir: "vielleicht stolpert ja einer bei der Recherche über diesen Thread"

Danke für eure Beiträge, die mich auch einen Schritt weiter gebracht haben.

andig, 2026-09-01:
Danke für die Dokumente. Ich habe die Register gegen `321798-47824-0146_ISG Modbus` (Stand 2026) geprüft.

**Adressierung: die -1 in unserem Template stimmt**

§4.1: *"Um das Register 30501 auszulesen, wird die Adresse 501 mit dem Funktionscode 04 angesprochen."*
§5.1: *"Die Adressierung des ISG ist eins-basiert (one-based). Berücksichtigen Sie je nach Konfiguration ggf. ein Versatz um 1."*

Das Template rechnet konsequent `Doku - 1`, und genau diese Register funktionieren im Feld:

| Doku | Objekt | Template |
|---|---|---|
| 522 | ISTTEMPERATUR Warmwasser | 521 |
| 518 | PUFFERISTTEMPERATUR | 517 |
| 5001 | Betriebszustand (1-4) | 5000 |
| 4002 / 4003 | SG READY MODBUS 1 / 2 | 4001 / 4002 |

Damit ist auch die Frage nach der Leistung beantwortet: **3679**, nicht 3680. Genau das hat @MicSchu berichtet.

**3680 gibt es, aber nicht bei jedem Regler**

- Kap. 6 WPMsystem (WPM 4, Reglerkennung 449), S. 12: `3680-3684 INVERTER AUFNAHMELEISTUNG IWS 1..5`, eine Nachkommastelle, 0-200, Datentyp 2, kW
- Kap. 8 LWZ/…Plus H(K)(W)L 230 (nur ISG Connect), S. 24: dito
- Kap. 9 Lüftungsintegralgeräte ohne R290, S. 29: `3680`, aber **zwei** Nachkommastellen, Datentyp 7
- Kap. 7 **WPM 3 / WPM 3i**, S. 14-20: die Liste endet bei 3655, kein 3680

Der `illegal data address` bei WPM 3/3i ist also dokumentationskonform und kein Fehler der Anlage. Deshalb kann `power` im generischen WPM-Template nicht bedingungslos gesetzt werden.

**Der zurückgenommene #18505 hatte zwei Fehler**

1. Adresse 3680 statt 3679, also fehlender 1-Versatz. Bei Anlagen mit nur einem Verdichter wird damit IWS 2 gelesen, das es nicht gibt.
2. `scale: 0.1` statt `scale: 100`. Rohwert x 0,1 = kW, evcc erwartet W.

Zum negativen Wert: laut §4.1 liefert das ISG für nicht verfügbare Objekte den Ersatzwert `32768 (0x8000H)`. Als int16 gelesen sind das -32768, mal 0,1 also -3276,8. Das erklärt die Anzeige aus [diesem Kommentar](https://github.com/evcc-io/evcc/issues/23345#issuecomment-3266850425). Mit `encoding: int16nan` filtert evcc genau diesen Wert.

**Semantik**

3680 ist die Aufnahmeleistung des Inverters von WP 1, Auflösung 0,1 kW. Pumpen, NHZ und Sumpfheizung sind nicht enthalten, wie @mp1981 richtig angemerkt hat. Bei Kaskaden liegen WP 2..5 auf 3681-3684.

PR folgt.

🤖 Generated with [Claude Code](https://claude.com/claude-code)

closed 2026-09-01 by andig