evcc-io/evcc #23383: Batterien Marstek Venus & Alpha ESS werden nicht gesteuert

CommoCruise, 2025-09-02:
### Describe the bug

### Bug 1:
Nach erfolgreicher Konfiguration von:
- Alpha ESS 5kwh Batterie
- Marstek Venus E 5kwh Batterie
- Netzzähler und PV Leistung wird von der Alpha ESS mitgeliefert
via der UI Konfig, werden alle Werte angezeigt, die Batterien werden aber nicht gesteuert(ESS entlädt/lädt selbst und Marstek hat sich nicht entladen(manueller modus)

[image]

### Bug 2:
Konfiguration via Commandline war in den Schritten erfolgreich. Aber die im yaml abgespeicherte Konfiguration hat nicht funktioniert: UI zeigt keine Geräte und auch evcc status via Commandline zeigt nur nicht konfigurierte Geräte.
[evcc.odt](https://github.com/user-attachments/files/22104017/evcc.odt)

[image]

[image]

### Steps to reproduce

Bug1: Konfiguration der seriell angeschlossenen Geräte via UI Konfig
Bug2: Konfiguration der seriell angeschlossenen Geräte via Commandline Konfig

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <placeholder>

interval: 30s # control cycle interval

meters:
- type: template
  template: alpha-ess-smile
  id: 85
  device: /dev/ttyUSB1
  baudrate: 9600
  comset: 8N1
  usage: grid
  modbus: rs485serial
  name: grid1
- type: template
  template: alpha-ess-smile
  id: 85
  device: /dev/ttyUSB1
  baudrate: 9600
  comset: 8N1
  usage: pv
  modbus: rs485serial
  name: pv2
- type: template
  template: alpha-ess-smile
  id: 85
  device: /dev/ttyUSB1
  baudrate: 9600
  comset: 8N1
  usage: battery
  modbus: rs485serial
  name: battery3
- type: template
  template: marstek-venus
  id: 1
  device: /dev/ttyUSB0
  baudrate: 115200
  comset: 8N1
  usage: battery
  modbus: rs485serial
  capacity: 5.12
  name: battery4

chargers:
- type: template
  template: demo-charger
  status: A
  power: 0
  enabled: true
  name: wallbox5

loadpoints:
- title: Garage
  charger: wallbox5
  mode: on

site:
  title: Mein Zuhause
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
    - battery4
```

### Log details

```text
kann ich nachreichen bei Bedarf, da es nicht meine Anlage ist und ich immer nur Abends nach Absprache draufkomme.
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.207.5

andig, 2025-09-02:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

mfuchs1984, 2025-09-04:
Bzgl Punkt 2 sieht es für mich so aus, als hättest du die erstellte yaml nicht in /etc verschoben.

CommoCruise, 2025-09-04:
@mfuchs1984 doch yaml liegt in /etc. Warum sie nicht berücksichtigt wird ist mir ein Rätsel.

CommoCruise, 2025-09-04:
> Bzgl Punkt 1: welche Art der Steuerung der Batterien erwartest du denn von evcc?

Ich erwarte das evcc die Batterien steuert. Diese werden aktuell von evcc nur angezeigt und laufen mit eigener Einstellung.
Mein Ziel ist es das beide Batterien von evcc sinnvoll geladen und entladen werden. Da aktuell sich die Batterien gegenseit entladen/laden.

mfuchs1984, 2025-09-04:
> [@mfuchs1984](https://github.com/mfuchs1984) doch yaml liegt in /etc. Warum sie nicht berücksichtigt wird ist mir ein Rätsel.

Laut Log verwendet evcc 
[main  ] INFO 2025/09/04 20:45:33 using config file: /etc/evcc.yaml
Schau mal in diese Datei.

mfuchs1984, 2025-09-04:
> Da aktuell sich die Batterien gegenseit entladen/laden.

Das musst du, sofern deine Batterien es zulassen, mit der Residualleistung und Reaktionszeiten der Batterien spielen.