evcc-io/evcc #23395: Modbus Error repeated until restart

lka, 2025-09-03:
### Describe the bug

Das Fahrzeug wurde nicht geladen und der gestrige Ladezustand wurde angezeigt. In den Logs tauchen wiederholt die Modbus Fehler auf. Die Synchronisation ist außer Tritt geraten und wurde auch über Stunden nicht korrigiert.
Erst ein Neustart, von mir durchgeführt, konnte Abhilfe schaffen.

### Steps to reproduce

Fand bisher (seit 4 Wochen im Betrieb) einmalig statt.

Ich kann das Verhalten nicht bewusst herbeiführen, also auch nicht mit den Nightly Build prüfen.

### Configuration details

```yaml
## minimal configuration example

site:
  title: Home # display name for UI
  meters:
    grid: my_grid
    pv:
      - my_pv
    battery:
      - my_battery

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints
loadpoints:
  - title: Garage # display name for UI
    charger: my_charger # charger
    vehicle: my_car # default vehicle

# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  # replace with your real grid meter
  - name: my_grid
    type: template
    template: solaredge-hybrid
    usage: grid

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.73 # Homeassistant Modbus Proxy
    port: 502 # Port
    timeout: 10s # optional
  # replace with your real pv system
  - name: my_pv
    type: template
    template: solaredge-hybrid
    usage: pv

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.73 # Homeassistant Modbus Proxy
    port: 502 # Port
    timeout: 10s # optional
  # replace with your real battery
  - name: my_battery
    type: template
    template: solaredge-hybrid
    usage: battery

    # Modbus TCP
    modbus: tcpip
    id: 1
    host: 192.168.178.73 # Homeassistant Modbus Proxy
    port: 502 # Port
    timeout: 10s # optional

# replace with your real charger
# see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: my_charger
    type: template
    template: em2go-home
    host: 192.168.178.111
    # template: demo-charger
    # status: C # charging
    # power: 2000 # 2 kW charging power
    # enabled: true # optional

interval: 30s

# replace with your real vehicle (optional)
# see https://docs.evcc.io/docs/devices/vehicles
#vehicles:
#  - name: my_car
#    type: template
#    template: offline
#    title: blue e-Golf
#    capacity: 50 # in kWh

vehicles:
  - name: my_car
    type: template
    template: bmw
    user: "me"
    password: "Pad"
    title: "iX1 edrive20"
    capacity: 64 # Akkukapazität (kWh), optional
    hcaptcha: „captcha“

# enter your real grid tariff and feed-in price
# see https://docs.evcc.io/docs/tariffs
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.39 # EUR/kWh
```

### Log details

```text
…
[site  ] ERROR 2025/09/03 16:26:38 battery 1 power: read failed: modbus: response transaction id '51622' does not match request '51623'
[site  ] ERROR 2025/09/03 16:26:38 battery 1 soc: read failed: modbus: response transaction id '51623' does not match request '51624'
[site  ] ERROR 2025/09/03 16:26:38 grid power: model 203 block 0 point W: modbus: response transaction id '51624' does not match request '51625'
[site  ] ERROR 2025/09/03 16:27:05 
2025-09-03 16:26:58.632277777 +0200 CEST m=+711845.194920802 !!  15ms read failed: modbus: response transaction id '51625' does not match request '51626'
2025-09-03 16:26:58.662252884 +0200 CEST m=+711845.224895904 !!  51ms read failed: modbus: response transaction id '51628' does not match request '51629'
2025-09-03 16:26:58.752671461 +0200 CEST m=+711845.315314409 !!  44ms read failed: modbus: response transaction id '51631' does not match request '51632'
2025-09-03 16:26:58.821811469 +0200 CEST m=+711845.384454565 !!  83ms read failed: modbus: response transaction id '51634' does not match request '51635'
…
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.207.5

lka, 2025-09-04:
Nein, die Verbindung wurde auch nach mehreren Stunden nicht geschlossen. Erst nachdem ich einen Neustart von evcc durchgeführt habe.

site  ] ERROR 2025/09/03 16:23:36 battery 1 power: read failed: modbus: response transaction id '51363' does not match request '51364'
[site  ] ERROR 2025/09/03 16:23:36 battery 1 soc: read failed: modbus: response transaction id '51364' does not match request '51365'
[site  ] ERROR 2025/09/03 16:23:37 
2025-09-03 16:23:28.651091181 +0200 CEST m=+711635.213734166 !! 152ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51326' does not match request '51327'
2025-09-03 16:23:28.814949 +0200 CEST m=+711635.377591975 !! 169ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51329' does not match request '51330'
2025-09-03 16:23:29.010483463 +0200 CEST m=+711635.573126446 !! 113ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51332' does not match request '51333'
2025-09-03 16:23:29.151420107 +0200 CEST m=+711635.714063120 !!  96ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51335' does not match request '51336'
2025-09-03 16:23:29.319534905 +0200 CEST m=+711635.882177883 !! 137ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51338' does not match request '51339'
2025-09-03 16:23:29.597334723 +0200 CEST m=+711636.159977699 !!  15ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51341' does not match request '51342'
2025-09-03 16:23:29.704624047 +0200 CEST m=+711636.267267038 !!  22ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51343' does not match request '51344'
2025-09-03 16:23:30.013362311 +0200 CEST m=+711636.576005279 !!  15ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51346' does not match request '51347'
2025-09-03 16:23:30.454566245 +0200 CEST m=+711637.017209264 !!  47ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51351' does not match request '51352'
2025-09-03 16:23:30.979886575 +0200 CEST m=+711637.542529559 !!   5ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51352' does not match request '51353'
2025-09-03 16:23:31.953075461 +0200 CEST m=+711638.515718435 !!  15ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51355' does not match request '51356'
2025-09-03 16:23:33.655719441 +0200 CEST m=+711640.218362457 !!  16ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51360' does not match request '51361'
2025-09-03 16:23:35.523256615 +0200 CEST m=+711642.085899606 !!  79ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51362' does not match request '51363'
2025-09-03 16:23:37.053423002 +0200 CEST m=+711643.616066014 !!   5ms add[0]: model 103 block 0 point DCW: modbus: response transaction id '51366' does not match request '51367'

[site  ] ERROR 2025/09/03 16:23:37 pv 1 power: add[0]: model 103 block 0 point DCW: modbus: response transaction id '51366' does not match request '51367'
[site  ] ERROR 2025/09/03 16:23:37 grid power: model 203 block 0 point W: modbus: response transaction id '51365' does not match request '51366'
[site  ] ERROR 2025/09/03 16:24:06 
2025-09-03 16:23:58.633254026 +0200 CEST m=+711665.195896982 !!  15ms read failed: modbus: response transaction id '51367' does not match request '51368'
2025-09-03 16:23:58.659471792 +0200 CEST m=+711665.222114791 !!  87ms read failed: modbus: response transaction id '51370' does not match request '51371'
2025-09-03 16:23:58.78592615 +0200 CEST m=+711665.348569126 !!  63ms read failed: modbus: response transaction id '51374' does not match request '51375'
2025-09-03 16:23:58.888806485 +0200 CEST m=+711665.451449474 !!  56ms read failed: modbus: response transaction id '51377' does not match request '51378'
2025-09-03 16:23:58.993611015 +0200 CEST m=+711665.556254034 !!  49ms read failed: modbus: response transaction id '51380' does not match request '51381'
2025-09-03 16:23:59.104393704 +0200 CEST m=+711665.667036712 !!  68ms read failed: modbus: response transaction id '51383' does not match request '51384'
2025-09-03 16:23:59.334647369 +0200 CEST m=+711665.897290338 !!  84ms read failed: modbus: response transaction id '51387' does not match request '51388'
2025-09-03 16:23:59.712836142 +0200 CEST m=+711666.275479108 !!  15ms read failed: modbus: response transaction id '51390' does not match request '51391'
2025-09-03 16:24:00.089597946 +0200 CEST m=+711666.652240826 !!  11ms read failed: modbus: response transaction id '51393' does not match request '51394'
2025-09-03 16:24:00.444896873 +0200 CEST m=+711667.007539889 !!  15ms read failed: modbus: response transaction id '51396' does not match request '51397'
2025-09-03 16:24:01.461618881 +0200 CEST m=+711668.024261881 !!  15ms read failed: modbus: response transaction id '51399' does not match request '51400'
2025-09-03 16:24:02.806346903 +0200 CEST m=+711669.368989884 !!  15ms read failed: modbus: response transaction id '51402' does not match request '51403'
2025-09-03 16:24:04.454359566 +0200 CEST m=+711671.017002553 !!  15ms read failed: modbus: response transaction id '51405' does not match request '51406'
2025-09-03 16:24:06.837726927 +0200 CEST m=+711673.400369943 !!  15ms read failed: modbus: response transaction id '51407' does not match request '51408'

[... cut ...]
[site  ] INFO 2025/09/03 16:32:28     co2:       âœ—
[site  ] INFO 2025/09/03 16:32:28     solar:     âœ“
[lp-1  ] INFO 2025/09/03 16:32:28 loadpoint 1:
[lp-1  ] INFO 2025/09/03 16:32:28   mode:        off
[lp-1  ] INFO 2025/09/03 16:32:28   charger:     power âœ“ energy âœ“ currents âœ“ phases âœ“ wakeup âœ—
[lp-1  ] INFO 2025/09/03 16:32:28   meters:      charge âœ“
[lp-1  ] INFO 2025/09/03 16:32:28     charge:    power âœ“ energy âœ“ currents âœ“
[lp-1  ] INFO 2025/09/03 16:32:28 vehicle updated: unknown -> iX1 edrive20

lka, 2025-09-04:
Sorry länger ist das Log nicht zu sehen….
Das sind allerdings nicht Stunden, sondern ca. 9 Minuten.

andig, 2025-09-04:
Verwendest du den Modbus Proxy oder greifen externe Automatisierungen auch noch auf das Gerät zu?

andig, 2025-09-04:
Und gibts davon auch ein Log? Nutzt ja nix an der Stelle zu suchen wo das Problem _nciht_ passiert?

spuky, 2025-09-04:
Wenn ich das in der config oben richtig verstehe verwendet @lka den Homeassistant modbus proxy...
also 

Gerät <-> HA modbus proxy <-> evcc

und nicht den evcc modbus proxy... und der hat immer mit einer transaction id geantwortet die -1 von dem war was evcc gefragt hat...

und so oft wie die Connection geschlossen wurde in dem Zeitraum hat Evcc beim Fehler die Verbindung auch neu Aufgebaut...

spuky, 2025-09-04:
Also an sich redet in dem Fall Evcc direkt mit den Geräten und Modbus abfragen oder schreiben von anderen Clients wird immer an die Geräte durchgereicht...

gerät <->evcc<->anderer Client z.b. HA

lka, 2025-09-04:
Nachdem ich die Adressen umkonfiguriert und den Proxy abgeschaltet habe, bekomme ich diese Fehlermeldungen:

2025-09-04 15:21:30.333504694 +0200 CEST m=+54.948318398 !! 10000ms read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:21:40 battery 1 power: read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:21:50 battery 1 soc: read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:22:10 
2025-09-04 15:22:00.336401963 +0200 CEST m=+84.951215683 !! 10000ms read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:22:10 battery 1 power: read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:22:20 battery 1 soc: read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:22:40 
2025-09-04 15:22:30.32373975 +0200 CEST m=+114.938553463 !! 10001ms read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:22:40 battery 1 power: read failed: dial tcp 192.167.178.110:1502: i/o timeout
[site  ] ERROR 2025/09/04 15:22:50 battery 1 soc: read failed: dial tcp 192.167.178.110:1502: i/o timeout

closed 2025-09-17 by github-actions[bot]