evcc-io/evcc #23401: Wolf CHA Support

tepebox, 2025-09-03:
Hi,
I use a Wolf CHA 7 heat pump and would like to know if there's any plan to support the Wolf SmartSet Cloud Solution? I know that there's ism7mqtt as a Home Assistant Add-On and allows MQTT communication but this is only good as an alternative to SmartSet which is i.e. required for remote assist from my installation partner.
Wolf just announced that they will support EEBus very soon, will that then be supported? I have not seen support for EEBus enabled heat pumps...
Thanks,
Thomas

andig, 2025-09-03:
We cannot support anything „very soon“. Not sure what the real request is here?

tepebox, 2025-09-04:
Die vorhandenen evcc plugins unterstützen die Wolf CHA nicht, ich wüsste nicht, wie ich sie anbinden soll... Das sind Split- bzw. Sole-Anlagen und keine Monoblock Luft-Wasser, die momentan wohl die gebräuchlichsten Modelle sind... Oder verstehe ich da etwas falsch?

andig, 2025-09-05:
Die Plugins wissen nichts von "Wolf CHA" sondern stellen technische Möglichkeiten zur Verfügung. Nutzen/parametrieren kannst und musst du sie selbst. Uns liegt keine Information oder Schnittstellendoku zu dem Gerät vor.