evcc-io/evcc #23427: Integration of the DTU Gateway

steffengoethling, 2025-09-05:
Hello,

thanks for your great work on evcc. I just came across the project and was able to configure my home PV system + EV in no time at all.

One question: Are there already any efforts to integrate the DTU Gateway?

https://github.com/ohAnd/dtuGateway

https://github.com/ohAnd/dtuGateway?tab=readme-ov-file#api-reference

This would make it possible to support Hoymiles inverters with Wi-Fi.

Best regards,
Steffen

steffengoethling, 2025-09-05:
I was expecting a configuration for the DTU Gateway. Apparently, you can enable OpenDTU support and then I need to select the OpenDTU configuration in evcc accordingly?

spuky, 2025-09-05:
yes just select Hoymiles OpenDTU and give it the IP of the Gateway...

[image]

spuky, 2025-09-05:
just test if http://{{ gateway ip}}/api/livedata/status returns json... else we could do an implentation using the json api

RenatusRo, 2025-09-05:
no need to compile. just pick up the template
[hoymiles-dtugateway.yaml](https://github.com/evcc-io/evcc/pull/23430/files#diff-249853bff85a26fe27737adf7b046be6ac210e72eac00754e4442a823dab4085)
and run evcc with startup option
--template hoymiles-dtugateway.yaml
to test it

https://docs.evcc.io/docs/reference/cli/evcc

closed 2025-09-05 by andig

spuky, 2025-09-05:
or use tomorrows nightly and report back if something is not working ;-)

steffengoethling, 2025-09-05:
Ah cool, will try it tomorrow.

Thanks a lot!

steffengoethling, 2025-09-06:
Hi @all,

looks good! I configured the DTU Gateway as "Balkonkraftwerk". 
Thank you again!!!

Steffen

[image]
[image]