evcc-io/evcc #23447: Wrong SOC when multiple batteries configurated

mopro1985, 2025-09-07:
### Describe the bug

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
Batteries configurated over UI
```

### Log details

```text
time="2025-09-07T15:53:57+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2025-09-07T15:53:57+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2025-09-07T15:54:07+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2025-09-07T15:54:07+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2025-09-07T15:54:07+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2025-09-07T15:54:07+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2025-09-07T15:54:17+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
time="2025-09-07T15:54:17+02:00" level=debug msg="write [{ EMS_REQ_POWER_PV None <nil> } { EMS_REQ_POWER_ADD None <nil> }]"
time="2025-09-07T15:54:17+02:00" level=debug msg="write [{ EMS_REQ_POWER_BAT None <nil> }]"
time="2025-09-07T15:54:17+02:00" level=debug msg="write [{ EMS_REQ_BAT_SOC None <nil> }]"
time="2025-09-07T15:54:27+02:00" level=debug msg="write [{ EMS_REQ_POWER_GRID None <nil> }]"
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

mfuchs1984, 2025-09-07:
Did you configure the capacites?

closed 2025-09-07 by andig

mfuchs1984, 2025-09-07:
Denke schon, klick einfach mal drauf und schau was es an (ggf. erweiterten) Einstellungen zur Batterie gibt.